prusa3d/Prusa-Firmware-Buddy #3945: [FEATURE REQUEST] MK4 Display LED strip custom setings

lightingcze, 2024-04-23:
**Printer type** - MK4
**Optional upgrades** - custom settings for under display led light

It would be nice to have a options to set own colors and states (flashing,static) and intesity of led light.

And i see more possibilites when can be used this light . like status when heating or filament changing :-) etc.

DrPurrple, 2024-04-26:
I would love that as well. It’s the first thought I had when I saw the LED bar. Please let us customise it.

lightingcze, 2024-05-09:
In this document its written that its possible already at this moment.
https://help.prusa3d.com/article/status-led-explained-mk4-xl_418975

But here its written a opposite.
https://help.prusa3d.com/article/buddy-firmware-specific-g-code-commands_633112#m-codes
(find the M150 and M151 g-gode cmd)

So what is correct?

reboot81, 2025-01-13:
Please, this needs to be sorted out.

danopernis, 2025-01-14:
Hi,

I looked into this a bit and it seems https://help.prusa3d.com/article/status-led-explained-mk4-xl_418975 is wrong when referring to Marlin documentation. The G-codes are [entirely reimplemented](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/8978fe1b166806a4ef6ecc5bff210cbbf183fc06/src/marlin_stubs/M150.cpp#L55-L69) as https://help.prusa3d.com/article/buddy-firmware-specific-g-code-commands_633112#m-codes suggest. However, the implementation [doesn't actually do anything](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/8978fe1b166806a4ef6ecc5bff210cbbf183fc06/src/gui/led_animations/printer_animation_state.hpp#L37-L39).

There are already several open issues related to LEDs and I feel like they will benefit from being aggregated into a single, consistent and comprehensible package.

Marlin's original M150 seems to let users fully control the colors, while Buddy firmware wants to let users assign animation style and colors to individual printer states. Maybe we need both. If users are given ability to redefine those, it should also be probably stored to EEPROM to survive printer restart and controllable from GUI and PrusaConnect. All of this is doable, but takes a plenty of time.

We are going to dicsuss this with the product deparment.

lightingcze, 2025-01-27:
I would like to have a new tab in PrusaConnect web page with all controllable functions.That would be great. 
Im LD and realy like to control the lights.
If is any option how to help you with that ,let me know.
Ondra.

closed 2025-04-08 by github-actions[bot]