prusa3d/Prusa-Firmware-Buddy #3946: [FEATURE REQUEST] Sticky temperature adjustment for multitool XL

TojikCZ, 2024-04-23:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - [XL]
**Optional upgrades** - [2T or 5T]

**Is your feature request related to an existing problem? Please describe.**
  Say I want to adjust the temperature of tool 3 after i start the print. The temperature gets reset every toolchange making the temperature adjustment pointless.

**Describe the expected functionality**
  The tool remembers the temperature I have set until a new one is set to it. Or we disable the temperature adjustment on multitool prints.

OP: https://forum.prusa3d.com/forum/bugs-errors/prusa-xl-nozzle-temperature-changes-do-not-stick/#post-708726

wafflecart, 2024-04-23:
Think this is the same as #3669. Maybe add your findings to that ticket.

TojikCZ, 2024-04-23:
I think that "manually set temperature resets on toolchange during print" is not the same as "For example, when preheating for a filament change on tool 3, tool 1 and tool 2 will also heat up."
Please go away

wafflecart, 2024-04-23:
Ok my bad sorry I read the issue wrong and didn’t see the “during print” part. but don’t need to be so rude and say to me “go away”. You’re a prusa employee and should not be addressing people like that there’s politer ways dude.

TojikCZ, 2024-04-23:
How do you say go away without being rude?

Say, “ I am in a terrible mood and just want some space and be left alone for a while. I do not want to hurt anyone's feelings, but I just am not up for company.”

Okay, here we go: I am in a terrible mood and just want this out of the "to-do" pile. Please stop moving it back in there. I do not want to hurt anyone's feelings, but I am not up for debating the legitimacy of this issue, I think I checked well enough.

Sorry for giving you the short version.

wafflecart, 2024-04-23:
Yes that’s worse 🤣. I’ll leave you be and delete my original comment so people don’t get confused and think it is related to that other issue. Hope you have a good day!

TojikCZ, 2024-04-23:
Yes sorry, was fed up and not in the mood for being corrected. Thanks for the effort, it's great you genuinely try to weed out duplicates. Thank you for that

TojikCZ, 2024-04-23:
Yes sorry, was fed up and not in the mood for being corrected. Thanks for the effort, it's great you genuinely try to weed out duplicates. Thank you for that
I'd like to delete all of these if that's fine with you, but they stay if you say no

TojikCZ, 2024-04-24:
Duplicate of #3343

closed 2024-04-24 by TojikCZ