prusa3d/Prusa-Firmware-Buddy #3947: [BUG] Failing to boot (BSOD) after installing MMU3 on MK4

pars3cproton, 2024-04-23:
**Printer type** - MK4 + MMU3

**Printer firmware version** - 6.0 RC1, RC2, Final release

**Optional upgrades** - MMU3

**Describe the bug**
After installing the MMU3, everytime the printer is turned on it tries to boot, it fails once, tries to boot again and then shows a BSOD message (the same message no matter if it's 6.0 RC1/RC2/Final release.

The only possible fix is "touching the cables" of the MMU3 while it boots, as shown in the video attached (you can also see the message in the video).
https://drive.google.com/file/d/10S5PZIgaxnuu6wRmS4JdfCftR7sWLvyO/view?usp=drive_link

With support we thought it was one of the two mmu3 boards fault, I tried replacing them but got the same error.
Then I bought another printer to mmu3 cable thinking that was the problem, but it's still showing the same behavior.

Support already has a dump file, but feel free to ask for a new one.

As of now the printer is usable if I boot it touching the cable behind the mmu3.

I tried everything, clearing MMU3 eeprom, factory resetting the printer but nothing seems to work.

m-cas, 2024-04-24:
my money is on bad solder (or cracked printed circuit trace) on the board you are pressing and the rest of the MMU unit. By pressing you are bending the material enough to restore the contact ... looks like a production defect, I'd ask for a complete new unit.

pars3cproton, 2024-04-24:
> my money is on bad solder (or cracked printed circuit trace) on the board you are pressing and the rest of the MMU unit. By pressing you are bending the material enough to restore the contact ... looks like a production defect, I'd ask for a complete new unit.

I already got a replacement for both the sheep board and the pd board (big one and small one of the mmu3), replaced and doing the same.  I don't need to push it hard, I just need to lightly touch it. I also bought a new mmu3 to printer cable, at this point all the electronics are replaced but still getting the same behavior.

I don't know much about boards or circuits or firmwares, but my money is on the actual printer buddy board (I have version 27)

m-cas, 2024-04-24:
this is clearly mechanical in nature. Something you are bending with your fingers enable the power to reach the MMU (leds on). You could try to push in different places (eg don’t touch the connectors, just the board; push something else, etc etc) to try to find out where is it making a difference

pars3cproton, 2024-04-24:
> this is clearly mechanical in nature. Something you are bending with your fingers enable the power to reach the MMU (leds on). You could try to push in different places (eg don’t touch the connectors, just the board; push something else, etc etc) to try to find out where is it making a difference
First of all thanks for your answers! :)
I tried touching just the cable getting inside the pd board and it boots with that too, pushing the cable inside the connector - not touching the board. If I reboot tho it doesn't boot anymore until I keep the cable pressed inside again. It's weird tho because the cable clicks and it looks in place (I don't feel any jiggle keeping it pushed or not) - also the cable is new.
I don't really know what to do at this point

m-cas, 2024-04-24:
hypothesis: bad manufactured cables and you have two of the same bad batch.
 You could try this:
 - remove the cable from the machine. Hold the plastic connector firmly and push the cables inside. You should see, by looking in the little holes or on the side locking tabs if they move. From what you described the seem not to be making contact until you press. I had cables that looked good but were not completely pushed in giving the same effect.
 - bed very slightly the pins inside the receiving connector and see if it works better (by changing the angle it may make contact)

Any chance you have a picture of the cable itself and the socket where it goes in?

pars3cproton, 2024-04-24:
Sure, here you go!
You can find pics of both old and new cable + the pd board
https://drive.google.com/drive/folders/1YdJ2uTDLopkkKX7Hkqpco7Au8qQre-0z

m-cas, 2024-04-24:
I’d try VERY GENTLY with a pin or a very little pointy object to bend the female contacts on the receiving board a little out and see if this makes a difference; not so much that the connector doesn’t fit anymore but just enough to be sure they touch well-From my phone while mobile, please excuse any errors.

https://drive.google.com/drive/folders/1YdJ2uTDLopkkKX7Hkqpco7Au8qQre-0z

ClaGre, 2024-04-24:
Since touching sometimes it works and sometimes it doesn't, couldn't it be a “simple” cold solder that doesn't make contact properly?

m-cas, 2024-04-24:
that is a valid possibility. I’d try with the pins first (lowest effort). Then, if you‘re handy with a solder, refresh the soldering on the connector. My bet is that one of the two will fix it...

pars3cproton, 2024-04-24:
Bending the pin slightly didn’t work. i can try resoldering the pin on the
pd board, it’s just weird to me that 2 pd board of 2 different batches show
the same behavior…

m-cas, 2024-04-24:
oh well, let’s go for another classic: take a metal brush or anything mildly abrasive and try to scrub the contacts of the cable and the board clean from a possible oxydation … if this doesn’t fix it, try to push the cable really well into the socket … I’d not try with the solder before having asked prusa support though…

Another possibility is to see if you cn verify the contacts with a tester both on the cable side and on the board side and see if you find any “wobbling when pressed” … try the power lines first (easier to find and from the look of the problem probably the culprit).

m-cas, 2024-04-24:
> I have 2 of every electronic component, so I can experiment :D i broke the
> housing of the pin on the sheep board (big one) and probably you were
> right, those pins are not making good contact with the pd board, since I
> can just bend the pd board lightly and it works, but if I let go of my hand
> it doesn’t.
> I might just end up soldering the two boards together, but I’ll try your
> new options first!

at least you’ve found the issue :D direct soldering is a little heavy handed though. With all double and a broken connector housing, I’d try to resolder the connector, put some good glue on the broken plastic and try again if this works…

m-cas, 2024-04-26:
were you able to solve the issue?

pars3cproton, 2024-04-26:
Not really, but it’s better. Keep fiddling with it now put it in a position
where it boots half the time without touching anything 😅

HaWiWe, 2024-05-01:
I have the same BSOD problem when the MMU3 is connected. The fix for me is to leave the USB cable plugged into the MMU (it does not have to be connected to a computer or anything, which logically shouldn't do anything). Video of it can be found here: https://youtu.be/CG1UNtzrMik
Support is also stumped as to what the reason is.

pars3cproton, 2024-05-01:
What kind of sorcery is that?? It does work as you said, if I keep the
cable plugged in I have 100% boot success!

HaWiWe, 2024-05-01:
Glad to be of help. :) I agree with above poster who said it might be something with soldering, conductive whatever, but noone knows what exactly. As long as that works I am okay with it. :P

m-cas, 2024-05-01:
… if I were to bet on it, the problem seems to be related to the power lines. The soldering (bad assembly batch) or the bad connection (bad connector production batch) prevent the power to reach the mmu (thus no leds on on the MMU) unless you ‘press’. With the usb cable you deliver power through an alternative path and all parts are now happy.I could also bet that the hw engineers at prusa are already looking at this :)-From my phone while mobile, please excuse any errors.

pars3cproton, 2024-05-01:
It would make sense, but the usb is just plugged in and not connected to
anything, as the other user said. And I was able to pinpoint the fault to
either the cable or the pd board plug

pars3cproton, 2024-05-01:
This is a very funny bug btw!

HaWiWe, 2024-05-01:
My friend said that that was the reason that I can't be an influencer: I also find the bug to be really funny and interesting. He said to become an influencer I would have had to make a 2-hour rant about it, namedropping other companies that do it a lot better. :P

m-cas, 2024-05-01:
LOL true!In my case is the ‘I need to find out what it is and fix it’ spirit that gets tickled by this kind of errors. After having looked at the video I wonder if there is a ground problem that the antenna (the usb cable) helps to sway a little so that it works again.  It would be nice to hear from a prusa engineer!-From my phone while mobile, please excuse any errors.

pars3cproton, 2024-05-01:
One thing that is tickling me is that maybe the bootloader expect the MMU
to react faster than it could and the usb plugged in might put a small
delay on it, giving it time to properly boot. i don’t even know if it makes
sense lol

m-cas, 2024-05-01:
The only thing inside a usb cable is at most a couple of resistors to tell the system that powering the lines is ok… and 20cm of cable in this case…-From my phone while mobile, please excuse any errors.

F1rst-Lay3r, 2024-05-04:
I have a similar issue with 6.0.0 and a MMU3 on the MK4. Difference is that I think mine is not hardware related. When I revert to 6.0.0 RC2, the BSOD is gone. Re flashing to 6.0.0 and the BSOD is back. Sometimes I'm able to boot without a BSOD. 

Also with 6.0.0 there are some random beeps now and then, that are also gone when I revert to 6.0.0 RC2. 

For now I keep running on 6.0.0 RC2, without issues.

thadogg, 2024-05-07:
Hello,
Exact same problem here. The USB cable workaround worked perfectly. This is definitely a critical bug/defect...

F1rst-Lay3r, 2024-05-20:
Ever since I reverted back to firmware 6.0.0 RC2, I was able to boot fine without any BSOD. Up untill today that is, today I could hardly boot at all. So I tried the 'unconnected' USB cable workaround and for now it's working.

fnordsh, 2024-05-27:
I seem to have the same or a similar issue - for me a reliable workaround is to press the MMU3's reset button until the bootup of the MK4 is finished. After that, everything works fine.
[image]

fnordsh, 2024-05-27:
Just a thought: Is there maybe a floating input somewhere that can, depending on the read value, delay or otherwise influence the MMU3 boot process?

m-cas, 2024-05-27:
… or a ground imbalance or something like this. Definitely something that an HW engineer at Prusa shouldn‘t have problems at finding out what is is give how easy it is to reproduce and the three workarounds known to bypass it.-From my phone while mobile, please excuse any errors.

sascape, 2024-05-28:
I've had this problem since assembly over a month ago. I thought I had assembled it incorrectly but I noticed that when I hold the MMU at a different angle in my hand during boot up I have no problems. Once it is booted up the printer and MMU work perfectly!

As soon as my current print finishes, I will try the cable trick. It would be very funny if this fix worked.

pars3cproton, 2024-05-28:
> assembled the MMU incorrectly but I noticed that when I hold the MMU at a

The different angle thing is due to the mmu to printer cable, you’ll notice
that even just touching that slightly where the plug is (mmu side) will
make it boot. And yes, the cable trick works, I’m bsod free since someone
here mentioned it!

sascape, 2024-05-28:
Well, the cable seems to be in the right position right now (weirdly, I had the boot bug just a few hours ago). The printer boots up correctly every time I cut the power or reset. 
I don't want to push my luck and I will wait for the moment the bug reoccurs.
Right now I need reliability. I'm happy to test around with the cables in 1-2 weeks when I have finished my time-sensitive projects.

tlachmann, 2024-06-03:
Hi,
having the same issue as descrided here, https://forum.prusa3d.com/forum/original-prusa-i3-mmu3-hardware-firmware-and-software-help/pd-board-issue-overcurrent-overvoltage/

After some more investigation, I assume that this is caused by the inrush current while the capacitors are charging. 
The LDO circut on the PD-board consumes also an inrush current at the same time. 
I´m not sure if it is possible to softstart the MMU 24 Volts Rail or to changing the time bias of the overcurrent detection.
I assume that allowing the higher inrush current for additional 50µSec could help.

Greets Thorsten

tlachmann, 2024-06-04:
yesterday I did some measurements.. my opinion, that issue is annoying... 

But first: having a micro-USB cable plugged into the MMU, does not have quick-fix the behavior on my side, it still shows MMU Overcurrent

Now the annoying part, the measurements... 

- On my side, the Overcurrent issue is persistant, so if i want to boot the MK4 with mmu in factory default setup, no chance, always the Overcurrent notice.

- Once I prepared for the current measurement, cut the three GND wires from MK4 to MMU approx 5 cm to the PD-Board and put a 2mOhms Measurement shunt in beetween which has the effect, that the Issue is gone, no longer a overcurrent failure, of course , the shunt has only 0.002 Ohms...

My assumption was then, that combining the three GND wires coming from MK4 to connect the single shunt has "fixed" the isssue. 
But I was also wrong, after bridging the 2mOhms shunt, the issue was back again, persistantly, removing the bridge, results that the issue was gone...
So for me it seems, that even the 0,002 Ohms resistor limits the inrush current on the 24Volts sufficiently... 

But I did also some measurements of the inrush current in the working environment. 

I measured a peak current **up to 52A** at the moment when the 24V is switched on. 
the time **period above 10A is approx 1.8 to 2mSec**. Then the current reduces quickly below 0.5A

See attachd screeners.

[image]
[image]

m-cas, 2024-06-04:
Good work! So your hypothesis is that the usb cable plugged-in leading nowhere is fixing the problem (for a few at least) because it acts like a little capacitor changing the inrush?What about the ‘bend things a little’ that seem to work for others? Is that consistent with your observations?Now on the permanent solution: board changes (long time to fix) or an adapter containing some passive components (faster)?-From my phone while mobile, please excuse any errors.

PD-Board-current.-.1.jpeg (view on web)
PD-Board-current.-.2.jpeg (view on web)

tlachmann, 2024-06-04:
I don't know which influence the USB cable from scientific side has. 
in the schematic of PCB version 0.3 the ID pin of the microUSB cable isn't connected, normaly the ID Pin controls switching from USB Slave to OTG. If it would be connected to MCU and maybe the cable has that pin grounded, then I could imagine.
For me the USB cable is more or less a esotherik thing.
[image]

But even the fact, that the difference of an 0.002 Ohms Resistor (some cables has higher resistance) fixes the behavior on my side, is also not explainable, at least not at that tiny value.

On the other hand, I´m sure that a resistor between 1 and 1.4 Ohms on the 24 Volts Input of the PD-Board could act as a current limiter. On the other side, this value shuldn´t also not have a influence to the Stepper functionality.

Greets Thorsten

HaWiWe, 2024-06-04:
> For me the USB cable is more or less a esotherik thing.

I tried for a loooot of boots. So no, it is definitely not an esoteric thing but something that does work and does fix it for most of the people who have that bug even though there is no logical explanation for it. It don't find the bug too bad, rather curious. :D

pars3cproton, 2024-06-04:
But in all of this, did prusa aknowledge it or not?

fnordsh, 2024-06-04:
> Hi, having the same issue as descrided here, https://forum.prusa3d.com/forum/original-prusa-i3-mmu3-hardware-firmware-and-software-help/pd-board-issue-overcurrent-overvoltage/

Are you sure this is the same issue? I never got the overcurrent error, but I do get the BSOD on boot when I don't have the USB cable connected to the MMU3. Holding the MMU's reset button also helps.

"Esoteric" is a nice way to put it. ;-) In my experience, weird semi-deterministic bugs like this are often related to floating inputs. I'm sorry that I don't have much time to investigate further right now, but I would first check if all MCU input pins are initialized properly, with pullups except where they are explicitly not needed.

Next thing to try would be adding a few seconds of delay in the MMU's boot process. (Based on the observation that holding the MMU's reset until the MK4 has finished booting works reliably for me.) Not sure if I would classify that as fix or rather as a workaround, though. ;-)

HaWiWe, 2024-06-04:
Yes, for every new MMU update I have a short conversation with the support, telling them it's still there and him confirming that they are still looking for the reason.

tlachmann, 2024-06-05:
If the open USB cable would add capacitiy, then it also need to be charged on boot, which would worse the issue.

electrically that usb cable represents a antenna and maybe it acts like that and disturbes the MCU on boot which adds a delay.

tschaerni, 2024-06-05:
> If the open USB cable would add capacitiy, then it also need to be charged on boot, which would worse the issue.
> 
> electrically that usb cable represents a antenna and maybe it acts like that and disturbes the MCU on boot which adds a delay.

A crap, I meant inductance, not capacitance, that would lower the in rush current.

> Yes, for every new MMU update I have a short conversation with the support, telling them it's still there and him confirming that they are still looking for the reason.

I'm also gonna get a replacement board in a few days, it'll probably be unresolved, but they asked me to send my board back so they can investigate it.

tlachmann, 2024-06-05:
> Are you sure this is the same issue? I never got the overcurrent error, but I do get the BSOD on boot when I don't have the USB cable connected to the MMU3. Holding the MMU's reset button also [...]

regardless of the message BSOD/Overcurrent notice, the symptoms looks identical, so my assumption is that both has the same cause.

Subtle3D, 2024-06-05:
Had the MMU3 overcurrent issue happen on 2 MK4s. 

What led to it:
It worked fine on during install and firmware updates. MMU prints would work but on starting a print it kept saying there was filament detected and needed to unload before being able to assign tools. It would ask for filament type, heat up and unload (nothing) then go to the filament selection page. Up to this point the filament sensor reading always reported off. 

In the settings menu, disabling the filament sensor also disabled the MMU. Reenabling the sensor, then reenabling the MMU caused an instant reboot into MMU overcurrent.

Multiple reboots, unplugging replugging MMU, changing MMU sheep board, changing PD board, booting with each plug for steppers and Pinda sensor unplugged one at a time, caused no change. Booting without the MMU plugged in allowed the MK4 to boot, then plugging in the MMU and enabling it in setting again caused reboot to MMU overcurrent.

"Fix"
The USB cable plugged in "trick" stated above, allowed everything to boot normally and the MMU works for both of the printers (so weird).
Don't know how or if the filament sensor issue plays into this but the filament issue that led to this no longer exists on these printers either.

If the current spike on boot is non-damaging, then an interim fix until a proper hardware (if that's the issue) fix might be a firmware change to disable or increase the overcurrent detection limit for the first few seconds during MMU booting.

Subtle3D, 2024-06-05:
Addition to my previous post.

I have several other MK4's with MMU3's that also had the filament detection issue and disabling/enabling the filament sensor and MMU in the settings page stopped the detection issue and did Not cause an MMU overcurrent. (Definition of insanity? Same actions different results :)

tlachmann, 2024-06-05:
Today I had a chance to measure current when the MMU overcurrent error appers, and compared to my previous non-error measurements, I can´t measure an overcurrent.
in my previous measurement an peak overcurrent upt to 25A (for a very little time) happens, now the printer detected an overcurrent, while only 10,10A was reached (yellow line 20,20mV/2mOhms Shunt = 10,10A) , and switched off the 24 V Supply when it reached only 12-13V (blue line)

I think now it seems it is a wrong positive interpretation in the software.

[image]

tlachmann, 2024-06-05:
I spend some time to take a brief view into the code, and found that normally the MMU is "soft" powered up with a pulsed power-up procedure. 
But I didn´t see anything that looks like a pulsed power-up procedure.. So it seems that the issue is really software based and the pulsing didn´t happen.

Source: 
https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/b1b77f69574daa16b57e7361bc469ea14daa57e4/src/mmu2/mmu2_power.cpp#L20

```
using namespace buddy::hw;

// The code below pulse-charges the MMU capacitors, as the current inrush
// would due to an inferior HW design cause overcurrent on the xBuddy board.
// In case overcurrent would still be triggered, increase the us_total
// value to pre-charge longer.
static constexpr uint32_t us_high = 5;
static constexpr uint32_t us_low = 70;
static constexpr uint32_t us_total = 15000;
```

```
  if (!config.can_power_up_mmu_without_pulses()) {
      CriticalSection critical_section;

      for (uint32_t i = 0; i < us_total; i += (us_high + us_low)) {
          {
              buddy::DisableInterrupts disable_interrupts;
              MMUEnable.write(Pin::State::high);
              delay_us_precise<us_high>();
              MMUEnable.write(Pin::State::low);
          }
          delay_us(us_low);
      }
  }

  MMUEnable.write(Pin::State::high);
```

tlachmann, 2024-06-06:
I hope Prusa also recognize that behavior. For today released 6.0.2. FW the changelog doesn`t show anything.

I also investigate regading the pulsed prodcedure, and after changing to a larger shunt, I could observe some artifacts that lokks like the pulses, before it was hidden by disturbances.
But of course the artifacts are so minor, that it doesn`t take account to reduce inrush current.

you can see the artifacts by the pulsed procedure in the small slope of the blue line before it rises vertical

[image]
[image]

tschaerni, 2024-06-11:
I few days ago I received a new xBuddy board and a new MMU3 board (and a new power panic cable, as I have an earlier revision with the 4 screw gearbox cover).

After replacing them, it seems to work without issue so far (crossing fingers).

Maybe the Issue is mitigated with newer versions of the xBuddy board?

Could be helpful to gather information about the revisions of the printers/xBuddy boards?

GvnCampbell, 2024-06-11:
I contacted support as well with the same issue.  We did a bunch of tests.  If I unplugged the cable from the mmu3 it still happened.  Once I fully removed the cable from mmu3 and buddy board it would boot up fine.  So they think it was the cable and shipped me a new one that comes tomorrow.

I did share them links to this issue though in case.  We will see if the new cable works.

tschaerni, 2024-06-11:
> I contacted support as well with the same issue. We did a bunch of tests. If I unplugged the cable from the mmu3 it still happened. Once I fully removed the cable from mmu3 and buddy board it would [...]

Hm... I didn't had that issue, it worked fine with the cable connected to the xBuddy board but not the MMU3, it even worked fine with the PD Board connected to the cable, but disconnected from the MMU3 board.

But yeah, maybe you just have a faulty cable, even tho I can't really imagine the cable being the culprit, unless there is a short in the cable somewhere.

GvnCampbell, 2024-06-13:
> Hm... I didn't had that issue, it worked fine with the cable connected to the xBuddy board but not the MMU3, it even worked fine with the PD Board connected to the cable, but disconnected from the [...]

Well I can confirm that its not the cable.  I got the new cable.  Replaced it and same issue. 

I can keep hitting the reset button and it eventually boots up but its still very annoying.

Back to support I go.

m-cas, 2024-06-13:
So far, the only success reported has been from people with new boards sent to them from support (I am not including the *workarounds* like pressing reset, adding a cable to nowhere, and bending the board/cable on the emu while booting). New cables have been sent multiple times already and apparently w/o changing the outcomes. Two things would be really interesting:
- a word from Prusa
- hearing from those that got new boards that work, if they can spot any difference from the previous ones

HaWiWe, 2024-08-13:
Still not working. Got me boards for the MMU, no luck. Support is still trying to find the reason.

Subtle3D, 2024-08-13:
I've still had it happen occasionally on a few of mine. The cable to the MMU3 workaround is sporadic but removing the usb drive from the screen before turning on the printer has worked every time so far.

GvnCampbell, 2024-08-13:
The only way I can get around it at this point is to turn off the machine and the raspberrypi for about 2 minutes.  Then the machine boots up fine and then I turn on the pi.  Its like I gotta give it time for everything to discharge.

Subtle3D, 2024-08-18:
> I've still had it happen occasionally on a few of mine. The cable to the MMU3 workaround is sporadic but removing the usb drive from the screen before turning on the printer has worked every time so far.

Well, I spoke too soon. I now have one I can't get past the overcurrent screen. The last 4 with the USB removed must have been a fluke.

Subtle3D, 2024-08-18:
> I don't know which influence the USB cable from scientific side has. in the schematic of PCB version 0.3 the ID pin of the microUSB cable isn't connected, normaly the ID Pin controls switching from USB Slave to OTG. If it would be connected to MCU and maybe the cable has that pin grounded, then I could imagine. For me the USB cable is more or less a esotherik thing. <img alt="image" width="623" src="https://private-user-images.githubusercontent.com/47011492/336559540-53074baf-430a-424c-8082-823164220e25.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM5ODg0NTcsIm5iZiI6MTcyMzk4ODE1NywicGF0aCI6Ii80NzAxMTQ5Mi8zMzY1NTk1NDAtNTMwNzRiYWYtNDMwYS00MjRjLTgwODItODIzMTY0MjIwZTI1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODE4VDEzMzU1N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBhMGZmMmI0MjMxMWU0Y2MyOWY0NjVhMGIyYzJlMDNjZjEzZGY2NDZiMzYwODkyMmZkMjg2NWRiNDkwZjMzYmMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.GrRGeqsos3OkFrIorOmUnbFfXcx_aZG32jvLx64IXNo">
> 
> But even the fact, that the difference of an 0.002 Ohms Resistor (some cables has higher resistance) fixes the behavior on my side, is also not explainable, at least not at that tiny value.
> 
> On the other hand, I´m sure that a resistor between 1 and 1.4 Ohms on the 24 Volts Input of the PD-Board could act as a current limiter. On the other side, this value shuldn´t also not have a influence to the Stepper functionality.
> 
> Greets Thorsten

I've now put a 1 Ohm 1 Watt resistor on each of the 2 red 24V wires to the MMU3 and it's worked. We'll see if that holds.

ntruchsess, 2024-09-18:
> I've now put a 1 Ohm 1 Watt resistor on each of the 2 red 24V wires to the MMU3 and it's worked. We'll see if that holds.

I had the same issue (false overcurrent error) with my MK4 + MMU3 and adding 0,5 Ohm Resistance to the 24V Supply (2 times 1Ohm added to each red power cable) solved it instantly. Prusa should modify the power-delivery board that sits in between the cable and the mmu-controller-board and add such current-limiting resistor with the next hardware iteration.

[118 more replies]

closed 2025-04-15 by Prusa-Support