prusa3d/Prusa-Firmware-Buddy #3949: [BUG] Firmware 6.0.0 for the XL isn't recognized

rcmaniac25, 2024-04-24:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - XL Single Tool

**Printer firmware version** - 6.0.0 RC3

**Original or Custom firmware** - Original

**USB drive or USB/Octoprint**
USB flash drive

**Describe the bug**
I saw the official release of Firmware 6.0.0 was released for the XL. I downloaded the bbf to the USB drive and plugged it in. I restarted the XL and it did not tell me there was an update. It ignores the latest BBF.

**How to reproduce**
1. Have an XL (single tool?) with 6.0.0 RC3 installed
2. Put the latest, non-RC, 6.0.0 firmware BBF onto a USB drive.
3. Plug in to XL, and restart the printer.

**Expected behavior**
The printer says there's a new firmware and asks if I want to install it.

rcmaniac25, 2024-04-24:
Hmm... SHA1 is different between them, so there's something different.

F1rst-Lay3r, 2024-04-25:
Same issue. Could install latest firmware by using the "Alternative method" -older firmware release (firmware downgrade). But that's not how it's supposed to work.

https://help.prusa3d.com/article/how-to-update-firmware-mk4-xl_453086

rcmaniac25, 2024-04-25:
Derp, I kept holding the reset button. No wonder that didn't work.

wafflecart, 2024-04-27:
In to report the same, I couldn't update from RC3 to 6_0_0 using USB drive, also "alternative method" in link above doesn't seem to work for me. I tried:

- Making sure no other .bbf files on usb a part from v6.0.0-xl
- Tried different usb drive
- Erased and reformatted usb
- Putting in usb and pressing reset
- Putting in usb and turning off and on at switch
- Tried alternative method of pressing reset and knob immediately after
- Tried alternative method of pressing both at same time

No beuno!

What **DID** work, is flashing it via Prusa Connect:
- Put latest .bbf file on usb drive
- Plug into printer
- Go to prusa connect
- Go to "printer files"
- Click flash

If I was to guess there's some bug in the checking of the firmware and if it's new via USB

Prusa-Support, 2024-06-14:
Thanks for bringing this to our attention.

This is usually the case of a temporary glitch and hardly reproducible.
What really seems to help in most cases is a combination of the following actions.

- Format the [USB drive](https://help.prusa3d.com/article/sd-cards-and-usb-drives_112291) or try a different one.
  - max 32 GB
  - filesystem FAT32
  - even consider trying one type USB 1.0
- Temporary [firmware downgrading](https://help.prusa3d.com/article/how-to-update-firmware-mk4-xl_453086#alternative-method-older-firmware-release-firmware-downgrade).
- [Hard reset via menu](https://help.prusa3d.com/article/factory-reset-mk4-xl_421206) - this requires the BBF firmware file on the USB flash drive.
- Enforce the firmware re-installation via the alternative method ([same method for downgrading](https://help.prusa3d.com/article/how-to-update-firmware-mk4-xl_453086#alternative-method-older-firmware-release-firmware-downgrade)).
- [Hard reset via bootloader](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3629#issuecomment-1998628052) as an alternative method to enforce the hard reset and the firmware installation at the same time.

Other than this, there is a work in progress on the bootloader which should improve it in several ways over the next few firmware releases and may correct potentially even partially related issues like this one.

Michele Moramarco
Prusa Research

rcmaniac25, 2024-06-14:
> Enforce the firmware re-installation via the alternative method ([same method for downgrading](https://help.prusa3d.com/article/how-to-update-firmware-mk4-xl_453086#alternative-method-older-firmware-release-firmware-downgrade)).

We tried this but it didn't work. Regardless, there are newer firmwares out now.

closed 2024-06-14 by rcmaniac25