prusa3d/Prusa-Firmware-Buddy #3950: [BUG] Heatbed overcurrent error at random occasions

shledge, 2024-04-25:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK4

**Printer firmware version** - 6.0

**Describe the bug**
Sometimes, usually once every few months of printing, my mk4 will reboot at the start of a print, showing a "heatbed overcurrent error" red screen.

Resetting the printer won't get rid of it, but I can instantly make it work again by manually setting the hotend temperature first in the menu, then starting the print. The issue then goes away for another few months.

I've checked all connections, nothing is loose and no signs of damage/overheating due to overcurrent, so I suspect this to be firmware related or perhaps an issue with the xbuddy board.

**How to reproduce**
Very difficult to reproduce as it happens so seldomly.

**Expected behavior**
Print to just start as normally, with hotend / bed heating up for probing.

danopernis, 2024-04-26:
Yeah, the part about resetting worries me a bit. Can you try hard power cycle next time this happens?

aeonik, 2024-08-06:
@zpeters Support hasn't gotten back to me yet, but I also suspect hardware issues. 

Currently, activating my hotend will reboot my printer every single time with this heatbed overcurrent error message. I am unable to print at all.

I have tested all the power cables with a multimeter, and I can run power to the hotend directly. All wiring and components seem to be fine. Heatbed works fine, even from the xLCD controls.

I suspect the controller is erroneously reporting this error message when the hotend is activated, or perhaps there is some sort of transient electrical/grounding issue, though none of my probing has found such electrical faults yet. Maybe interference? 

I triggered this issue on my printer while cleaning the hotend nozzle and heat block with Isopropyl alcohol (while it was heating up), so this indicates to me that it's must be something physical, and something related to the hotend.

I plan on probing the circuitry of the heatbed current detection sensors once I can find some information on their location, this way I can at least confirm if the sensor misreporting. If the sensor is telling the truth, I should be able to trace the current spike back to source, since I can reproduce this 100% of the time on demand.

I just need help finding the right places to probe.

closed 2024-10-15 by github-actions[bot]