prusa3d/Prusa-Firmware-Buddy #3955: Needless Filament Unload Procedure, when MMU Print is cancelled in the Warm Up Phase

JWartenberg, 2024-04-27:
Printer: MKS 3.5 + MMU3
Printer FW: 6.0.0
MMU FW 3.0.3

Scenario:

- Starting a MMU Print
- Cancel the print in the heating phase, before filament is loaded for the first time

Printer proceeds the filament unload procedure (heating up, ramming, ...) although no filament is loaded.
The procedure is needless in this case.

Best regards
Jochen

firetech, 2024-06-02:
Was just about to report this, glad someone else did it already. One would think that the printer would realize that no filament is loaded, since both FINDA and FSensor are showing OFF on the unloading screen...

Mercur13, 2024-06-03:
have the same issue here MK3.5 Fw: 6.0.1

firetech, 2024-06-03:
There is a workaround, albeit slightly annoying.

If you reset the printer (reset button) _instead of_ stopping the print in this stage, you will not need to unload (if you reset _after_ stopping the print, the printer will think it has filament loaded even after reboot).

danopernis, 2024-07-08:
This is scheduled for next minor (not patch) version of the firmware.

danopernis, 2024-08-22:
New version of the firmware 6.1.2 should contain the fix, can you test that?

JWartenberg, 2024-08-23:
Not before mid of next week

JWartenberg, 2024-09-01:
Tested
Issue is fixed

Sorry for the delay

Regards
Jochen

PR #4031 (2024-06-06): [BFW-5652] mmu3: skip unloading filament on early abort
checks if filament is loaded before queuing M702

fixes #3955
commit 024103c3 (2024-06-06): mmu3: skip unloading filament on early abort

checks if filament is loaded before queuing M702

fixes #3955
closed 2024-09-02 by danopernis