prusa3d/Prusa-Firmware-Buddy #3956: [BUG] MMU3 MK4 - Sequential printing causes a tool change after initial purge, fails, then prints mid-air

Shushuda, 2024-04-27:
**Printer type** - MK4

**Printer firmware version** - 6.0.0 stable

**Original or Custom firmware** - Original

**Optional upgrades** - MMU3

**USB drive or USB/Octoprint**
  USB drive

**Describe the bug**
  Sequential printing while using at least 2 filaments generates faulty Gcode. The printer loads one filament, does the initial purge line, then attempts to switch the filament to the other one. It seems to not perform the ramming procedure correctly - MMU3 attempts to pull the filament out while it's still in the hotend. It looks like the hotend performs the single-colour unload procedure while the MMU3 instantly tries to unload the filament without waiting for the unload to finish - both procedures seem to start at the same time instead of waiting for one another. When the hotend finally unloads, the MMU3 pulls the filament to then do the cut procedure (due to repeated fails) and load the next filament. Since the purge tower is not compatible with sequential printing, the printer then proceeds to print mid-air. The filament will eventually start flowing, but it will take a while.

**How to reproduce**
1. Enable sequential printing: Print settings - Output options - Complete individual objects.
2. Use at least 2 filaments in the print.
3. Slice.
4. The preview will show the bug.
5. Additional unload issues will occur if the print job is actually started.

**Expected behavior**
  There should be no filament change after the initial purge line.

**G-code**
  
[inverted+frames_0.2mm_PLA,PLA_13m.bgcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/15137428/inverted%2Bframes_0.2mm_PLA.PLA_13m.bgcode.zip)

Shushuda, 2024-04-27:
Furthermore, there seem to be similar issues with unload when the wipe tower is disabled - in general, no matter if the sequential printing is on or off. I've started a print without a wipe tower and on filament change the MMU3 tried to pull the filament before the hotend released it.

Shushuda, 2024-07-03:
Writing a comment to refresh as the issue still occurs.

DRracer, 2024-07-03:
I'm afraid this issue is a combination of a slicer issue and an unsupported scenario.

Please note, it's the slicer's responsibility to generate G-codes to perform ramming and freeing of the filament tip from the nozzle. If the purge tower is missing, the slicer probably doesn't generate G-codes to free the filament tip from the nozzle and therefore the MMU has a very hard (read "almost impossible") time pulling the filament out.

In your G-code, ramming, freeing and priming sequences are completely missing:
```
T0
M104 S220
M900 K0.05
M572 S0.036
M142 S36
M107
;LAYER_CHANGE
;Z:0.2
;HEIGHT:0.2
M486 S1
G1 E-.7 F2100
; Filament-specific end gcode
; <- here - purge tower ramming, filament freeing is missing
T2
M104 S230
M900 K0.05
M572 S0.036
M142 S36   
; <- here - filament priming and color cleaning is missing
```

closed 2025-02-10 by github-actions[bot]