prusa3d/Prusa-Firmware-Buddy #3957: [BUG] MMU3 MK4 - MMU tries to unload filament at the same time as ramming

Shushuda, 2024-04-27:
**Printer type** - MK4

**Printer firmware version** - 6.0.0 stable

**Original or Custom firmware** - Original

**Optional upgrades** - MMU3

**USB drive or USB/Octoprint**
  USB drive

**Describe the bug**
  When NOT using the purge tower, the printer performs the old-style ramming on filament change. But MMU3 does NOT wait for this hotend unload to finish. It tries to unload the filament back into the unit at the same time as the hotend starts ramming. This results in failed loads, pulleys grinding the filament down and the filament being slightly pulled out during ramming which ruins the tip.

**How to reproduce**
1. Disable the purge tower.
2. Use at least 2 filaments in the print.
3. Slice, print.
4. On filament change the printer will start the old-style ramming and MMU3 will start the unload into the unit without waiting for ramming to finish.

**Expected behavior**
  MMU3 should wait for the hotend unload to finish before it pulls the filament back into the unit.

Shushuda, 2024-07-03:
Writing a comment to refresh as the issue still occurs.

DRracer, 2024-07-03:
The same as in issue #3956 applies here as well. MMU prints without a purge tower are not supported. It is not a FW bug, FW relies on the fact, that the slicer freed filament tip from the nozzle. Closing...

closed 2024-07-03 by DRracer