prusa3d/Prusa-Firmware-Buddy #3959: [ENHANCEMENT] Visual feedback upon choosing "Set Ready"

ChrisHeerschap, 2024-04-29:
Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - MK3.5, MK4, XL, Mini

**Optional upgrades** - None

**Is your suggested improvement related to an existing problem? Please describe.**
  Please provide some form of visual feedback when choosing the "Set Ready" menu option. Currently there is only a beep if the sounds are enabled, but no immediate indication that the command was taken.

Selecting "Set Ready" has an immediate beep response if sounds are enabled, but no visual indication whatsoever that it has been submitted. I ran a test on my MK4 where I clicked "Set Ready" and it took 10 seconds for the printer to show as ready in the PrusaConnect page, but after 50 seconds the "Set Ready" option was not greyed out in the printer menu. The menu timed out at 50 seconds and returned to the main screen, at which point I could return to that menu and finally see the option greyed out. No other visual feedback is given.

I ran the same test on my Mini, MK3.5 and XL with the same results.

**Describe the expected functionality**
Have a pop-up message that says the command has been submitted and it should show up in PrusaConnect soon. Or, something to that effect.

CZDanol, 2024-05-03:
Hey, I've made a proposal solving this some time ago under internal ticket BFW-4938. I'm hoping to get some word from the UX guys eventually.

murk-sy, 2024-05-03:
Related issue but more about safety due to automated start / remote commands
- https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3315

closed 2025-11-19 by github-actions[bot]