prusa3d/Prusa-Firmware-Buddy #3961: MK4 Fw 6.0.0 Blue screeen fault at power up BSOD

VH-fixer, 2024-04-29:
MK4  FW 6.0.0 Production release. USB drive, 16GB.  

When powering up the printer, if faults showing a blue screen.   I will upload the dump file from one of these. 
Reset once or twice may be needed to boot successfully.   It doesn't always happen.  But about 25% of the time.

Once booted, there are no more blue screen faults and can run for days.  

There were no hardware changes.  Just updated to FW 6.0.0 and this started happening. 

This happened without the MMU installed and still can happen after installing the MMU3.

danopernis, 2024-04-30:
Hello and thanks for the report. Crash report would be much appreciated, or at least a photo of the actual blue screen. Sorry for the inconvenience.

VH-fixer, 2024-04-30:
Just did a lot of power on/off cycles trying a few variations and may have found the root cause.
There is a Beagle V2 camera plugged into the buddy board USB C port.   The camera has two usb connections.  One to the buddy board and one for power to the camera.  The one for camera power runs off an external USB power plug.  It is not connected to the printer.  The blue screen tends to happen when the power USB is not plugged in for the Beagle V2 camera.   When the camera does have power at printer boot up, the blue screen does not happen.

Without the camera being powered, a reset will clear the blue screen and allow a normal boot up.   Could this be overcome with a boot sequence or timing?  

The camera must have power at printer startup or be unplugged from the Buddy board to prevent the problem. 

[image]

Prusa-Support, 2024-05-09:
Thanks for reporting and we are taking notes.
However, the use of the camera through the type-C port as well as powering any extra third-party device through the printer is not tested and not approved. In fact, PSU & board may not supply adequate power to boot up and function correctly.
The free type-C USB port is available for custom uses - outside of the regular use described by the manual -  but its use has to be considered experimental, at your risk.

Our developers would gladly give this a look in the future, compatibly with our development plans, but please be advised that this at the moment remains an unsupported use of the printer. 

Michele Moramarco
Prusa Research

closed 2024-07-16 by github-actions[bot]