prusa3d/Prusa-Firmware-Buddy #3963: M593 on MINIIS with firmware version 6

koppensb, 2024-04-30:
Is the firmware version 6 for the MINIIS accepting the M593 gcode? 
On version 5 I printed the ringing-tower according to the marlin documentation. 
In the onscreen print settings I could see the input shaper values changing.
Now on version 6 there is no change in the values, and it seems as if M593 doesn't get accepted.

danopernis, 2024-04-30:
Looking at the code, M593 should work on current firmware. There is a documentation at https://help.prusa3d.com/article/buddy-firmware-specific-g-code-commands_633112 Without any arguments it should print current settings on serial line. Settings only get written to EEPROM after you explicitly write them with `M593 W`

koppensb, 2024-04-30:
I will have a look into it with octoprint and come back on this.

koppensb, 2024-05-02:
It seems as if the input shaper settings only show the eeprom value not the volatile settings as version 5 did. If I do a M593 F 32 and a M593 for getting actual values the output is right. On the onscreen settings menu I get the eeprom values. I am not sure if this indented. For my understanding the onscreen information should reflect the current running config especially in the calibrations menu.

danopernis, 2024-05-02:
It is debatable whether we should show "eeprom" or "volatile" value. Both make sense. I guess technically we can show both, but I am not UX expert and the menu can get quite overcrowded.

Also worth noting, IIRC the values in menu are not invalidated on change so you might need to hop out of the menu to see changes.

Do you have other questions, or can we close this?

koppensb, 2024-05-03:
Thanks for your answers. The Issue can be closed.

closed 2024-05-03 by danopernis