prusa3d/Prusa-Firmware-Buddy #3964: MMU3 MANUAL CONTROL WHEN PRINTER PAUSES FOR FAILURE MID-PRINT

legobrick2x4, 2024-04-30:
Just got the mmu3 for my mk3.9
Love it but it isn't without problems. Whenever there is a load failure mid print the menu has only the option to continue which retries it again. I want an option to enter a diagnostic state of sorts where I can see things like the finda sensor values (one issue I've had is a sensor wiggled loose mid print and I couldn't diagnose it) manually select the active filament, tell it to preform a load to mmu/nozzle action and even unload from extruder or mmu. And I would absolutely love the option to extrude filament too. 
(one issue I has was mostly unknown. I believe a piece of filament broke in the nozzle tube. I solved it by heating to 290c, going into settings and moving the e axis to ram whatever was stuck out.) that single simple act could have saved a 13 hour print. 

All these are ways I could fix a problem that is either annoying to fix as is, or impossible without canceling the print. 

And as soon as you exit that state it would be important to re-home at minimum the X and Y axis as I've already caused an x axis layer shift while fixing an issue. 

The goal with this feature would ultimately be able to be able for the mmu3 to recover from a problem without needing to abort the entire print. Because not every issue is can be solved by cutting the end of the filament and trying again.

canman72, 2024-05-28:
I wholly support this feature request.... I have yet to have a single MMU3 print work flawlessly and it is very frustrating trying to recover when the only option is "continue".  Suggest at lead having multiple options upon failure.... Key ones that I would like to see are:  1. Unload from MMU  2.  Reload from MMU  3. Cut filament.

TyzzyT, 2024-05-30:
I would love to see this feature request being implemented into my MK4. I'm planning on doing a multi color print of 35 hours with 663 of tool changes. Since the MMU3 is very stable, I believe it will be possible to do this. But... last time I did MMU some filament broke and got stuck. The only option was to retry, which wasn't the solution.
It would be great if we can solve it with some help from the printer: unload from MMU, reload to MMU, unload from extruder.

Many thanks in advance!

canman72, 2024-07-30:
Hey Prussia... any news on this?

cozmo14047, 2024-09-03:
any news yet this would be really helpful ?

MusedLife, 2024-09-13:
This would save everyone a lot of time. Thanks for all the great work on this firmware!

closed 2025-11-24 by github-actions[bot]