prusa3d/Prusa-Firmware-Buddy #3966: [BUG] No communication between MK3.5 and MMU3 after Firmwareupdate

Presswerk65, 2024-05-01:
**Printer type** - MK3.5+MMU3(upgraded from MMU2S)

**Printer firmware version** - 6.0.0

**Original or Custom firmware** - Original

**Describe the bug**
After the printer firmware upgrade to 6.0.0, the MK3.5 and MMU3 stop communicating. All LEDs on the MMU3 are on, the printer shows no MMU connected (icon on display inactive, no information in the settings menu) - even via usb the MMU3 doesn't get recognized by the computer. As soon I flash MK3.5_firmware_5.2.2.bbf to the 
printer again, the communication between MK3.5 and MMU3 starts again and I get the error, the firmware on MMU3 doesn't match
the printer firmware on the MK3.5. The MMU3 get recognized by the Computer again and I'm able to flash MMU3_FW3.0.2+878.hex. Now, everything works fine again - I tried to unplug the MMU3 while the printer upgrades his firmware, but the behavior is the same: As soon the printer is on 6.0.0, the MMU3 dies.

**How to reproduce**
Just flash 6.0.0 to the printer

**Expected behavior**
MMU3 still works after printer firmware update

RedTara, 2024-05-05:
I have exactly the same issue.

cozmo14047, 2024-05-09:
> I have exactly the same issue.

After you flash the printer firmware you have to flash the new mmu firmware for it to work,

Presswerk65, 2024-05-09:
As mentioned, at the moment the printer is on the new firmware, the communication with the MMU3 is no longer possible - also not via the USB port - it's also the same behaviour when I flash the MMU in advance

RedTara, 2024-05-09:
Thanks for your reply.Not sure if the work around I found will work for you, but here is how I got up and running.With both mk3.5 and mmu3 running firmware 5.2.2, use prusa slicer to update MMU3 via USB lead to MK4_MK3.9 firmware MMU3_FW3.0.3+896.hex.
Now update MK3.5 to MK3.5_firmware 6.0.0 and hopefully like me you will be able to then enable your MMU3 via the MK3.5 setting screen.
Hope this works

Presswerk65, 2024-05-10:
Unfortunately, this doesn't work ether - but I think the root of the problem is when the printer flashes the MMU after his update. I will try to update the MMU manually, then physically disconnect the MMU so the printer can't update the MMU and the flash the printer

Presswerk65, 2024-05-11:
Okay, even when I physically disconnect the MMU3 while the printer updates and updated the MMU3 in advance, the behaviour keeps the same: No communication with the MMU after the printer is on the new firmware

RedTara, 2024-05-11:
Noted. I didn't need to disconnect either using the upgrade route I was advised to use. My mk3.5 and MMU3 continue to talk and work together without any further issues. 
Yahoo Mail: Search, organise, conquer

closed 2024-07-22 by github-actions[bot]