prusa3d/Prusa-Firmware-Buddy #3967: [FEATURE REQUEST] Delayed Filament Runout Detection to Minimize Waste and Optimize Filament Use

LuckyResistor, 2024-05-02:
**Printer Type:** Prusa XL - This issue is particularly relevant for this model due to its specific design.
**Optional Upgrades:** None

# Problem Description

The Prusa XL features a lengthy filament transport tube that extends from the side entry to the printer head. This design results in approximately one meter of filament remaining in the tube when it runs out and/or spools are joined—a section of filament that goes unused. This results in filament waste, which is neither economical nor environmentally friendly. Moreover, waiting for the filament to reach the in-head filament sensor often leads to jams, making removal difficult and risking print failure.

# Proposed Solution: Filament Runout Delay Feature

Implementing a "Filament Runout Delay" feature in the firmware would address this issue by allowing the printer to continue operating for a preset distance after the filament sensor detects the end of the filament. This feature would significantly reduce waste and enhance the printer's functionality.

## User Interface Changes

1. **Menu Setting for Runout Delay:** Introduce a new menu item to set a "Filament Runout Delay" as a distance in centimeters.
2. **Enable/Disable Toggle:** Provide a toggle option to activate or deactivate the runout delay, preserving the user's distance settings.

## Operational Details

- Upon detection of filament absence by the sensor at the tube's entry, the printer initializes a delay counter to the user-defined length.
- This delay counter, stored temporarily in RAM, decrements corresponding to the extruder’s forward movements.
- The filament runout process triggers when the counter reaches zero, replicating the functionality as if no delay were set.

## Handling Special Cases

- **Print Completion:** If a print finishes before the counter zeroes out, the printer treats the filament as completely run out upon the next start.
- **Manual Intervention:** Manually initiating a filament change resets the delay counter.
- **Toggle Mid-Print:** Disabling the delay feature mid-print triggers an immediate runout detection if the filament has already passed the entry point.

## Additional Feature: Layer Change Optimization

- **Tolerance Setting:** Implementing a tolerance margin (e.g., 5cm) could synchronize the filament runout warning to coincide with a layer change, minimizing potential disruptions in print quality.

murk-sy, 2024-05-03:
Duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2564

danopernis, 2024-05-03:
@murk-sy thanks for keeping track of the issues
@LuckyResistor let's keep discussion in #2564 and close this in order to better manage issues

closed 2024-05-03 by danopernis

LuckyResistor, 2024-05-03:
@danopernis Sure, I missed that duplicate because of the title. Could you please improve #2564 so it can be better found and reflect the state of the discussion there?

- The authors main focus was to optimise the point where filament run-out triggers on the MINI+.
- My concern is to reduce filament waste on PrusaXL.
- At the moment, the title of the issue is misleading as it only reflects the authors original intent, and makes it hard to find.

murk-sy, 2024-05-03:
Improved title, thanks for feedback