prusa3d/Prusa-Firmware-Buddy #3968: Menu, Info, Odometer, amount of filament used in Kilometers is way off.

VH-fixer, 2024-05-02:
Printer MK4, FW 6.0.0  MMU enabled. 
See the attached picture
This odometer shows I have used over 4 MILLION kilometers of filament.  
 
[image]

My printer overall has only moved a few hundred kilometers.
I think someone forgot to devide mm by 1,000,000 to get to KM.

thijstriemstra, 2024-05-19:
it's issues like this that really doubt my trust in the firmware dev at prusa: rushed, incomplete and buggy.

VH-fixer, 2024-07-23:
In 6.0.4 I did a factory reset to clear out all statistical data.  After recalibrating and configuring the MK4 I ran a new part.
Meters of filament are being reported correctly.    
I am wondering if this number was data on the card from the factory.  That needs to be checked.    It is the first factory reset I have done.   Will keep an eye on it.

closed 2024-07-23 by VH-fixer

danopernis, 2024-07-23:
Basically the same as https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4095 I will copy my answer here as well:

> Clearly there is something wrong with the statistics. For now, this invalid value is already committed to EEPROM so there is nothing to do about that other than performing factory reset 😟

> There are some improvements related to EEPROM consistency scheduled for next release which should hopefully prevent this from happening again.

As to your question about filament number reported not changing, it is basically so large that any addition in order of meters is basically invisible.

No, the data was not there from the factory, but during the print, somehow an invalid number was stored there. Once such mistake is committed, it can't be reverted.