prusa3d/Prusa-Firmware-Buddy #3969: intermittent Wifi with 5.2.2 and 6.0.0

OopAck, 2024-05-02:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - Mk3.5

**Printer firmware version** - 6.0.0

**Original or Custom firmware** - 6.0.0

**Optional upgrades** - Wifi intermittently not being picked up 

**USB drive or USB/Octoprint**
Prusa USB Key Storage

**Describe the bug**
 When powering on the Wifi is not being picked up
Have power off and back on a couple or few times to get it to pick it up  

**How to reproduce**
  At boot

**Expected behavior**
  Wifi Icon to appear

**G-code**
  n/a

**Crash dump file**
  n/a

**Video**
no video

murk-sy, 2024-05-03:
Wifi issues are difficult to diagnose, so you need to provide some more information. It's possible it's a known issue.

Some details that would probably be useful:
1) What access point/router do you use?
2) Are there multiple access points?
3) Which frequencies is the access point running on? (just 2.5 Ghz, 2.4+5, etc)
4) Which security do you use? (WPA2, etc)
5) After connecting, does it stay connected?

OopAck, 2024-05-03:
What access point/router do you use? CAX80 Nighthawk
Are there multiple access points? no
Which frequencies is the access point running on? (just 2.5 Ghz, 2.4+5, etc) 2.4Ghz
Which security do you use? (WPA2, etc) WPA2-PSK[AES] + WPA3-Personal
After connecting, does it stay connected? yes

murk-sy, 2024-05-03:
Since you're running WPA2 and 3, it might be a duplicate of this issue:
- https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3864

If it is, I recommend closing this issue and posting your info there too, will make it known that it's a more common issue.

OopAck, 2024-05-04:
Bug-5393 fixed it

closed 2024-05-04 by OopAck