prusa3d/Prusa-Firmware-Buddy #3971: [BUG] <Enter comprehensive title>

GregFax, 2024-05-03:
**Printer**
Mk3.5
Original FW 6.0.0
Prusa Mk3S to Mk3.5 Upgrade Kit (Standard)
USB Drive

**Problem**
If an object on the print bed extends beyond about X = 180mm (or so), then Auto Mesh Levelling will fail.
After AutoHome, the printhead will travel to XMAX placing the PINDA probe beyond the printbed.
Moving to the first Mesh probe point, the first probe will try to drive the nozzle through the printbed, generating a Failure.

**Mimic**
Place any object on the furthest right limit (XMAX), and slice with PrusaSlicer 2.7.4.
Save bGCode and Print.

**Expected behavior**
  It should have printed.
I consider the PINDA X Offset might be involved? Or not contemplated by the new Mesh algorithm?

**G-code**
  PS generates BGCode (not supported by GitHub), so file attached as .ZIP
[fuse3 Mk3.5_0.4n_0.2mm_PLA_MK3.5_5h21m.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/15195478/fuse3.Mk3.5_0.4n_0.2mm_PLA_MK3.5_5h21m.zip)

CZDanol, 2024-05-03:
Internal ticket: BFW-5476

danopernis, 2024-06-27:
This actually looks like duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3851 which should was fixed in v6.0.3

@GregFax can you confirm that?

GregFax, 2024-06-28:
Issue more insidious than that!
Bu all now resolved with the excellent help of Prusa Support.
Not Firmware, Hardware. Although the available error messages weren't much help.....
For the record, a cable was fouling 1 mm on the left hand extremity - this was wrecking calibration, resulting in the printer "missing" the right hand edge.

closed 2024-06-28 by danopernis