prusa3d/Prusa-Firmware-Buddy #3972: [FEATURE REQUEST] Scheduled Stealth Mode for Overnight Printing

LuckyResistor, 2024-05-03:
**Printer Type**: All Models Equipped with a Real-Time Clock

## Problem Description

Stealth Mode is essential for reducing printer noise during overnight operations, particularly when long-duration prints are scheduled. Currently, transitioning between Normal and Stealth Modes requires manual intervention. This becomes inconvenient when prints run unsupervised overnight. Users must either preemptively set the printer to Stealth Mode for the entire duration—compromising print speed during daytime—or allow the printer to operate noisily at night. An automated solution would alleviate the need for manual mode switching during unsupervised operations.

## Proposed Solution

### User Interface Enhancements

- Introduce a new UI option to set start and end times for Stealth Mode activation.
- Add a toggle to enable or disable scheduled Stealth Mode.

### Operational Workflow

- The printer should automatically enter Stealth Mode at the configured start time.
- At the designated end time, the printer should revert to Normal Mode.

### Implementation Details

- Implement a background check every minute to ascertain if the current time falls within the user-defined range (e.g., 22:00 to 07:00).
- If the current time is within the range, the printer should operate in Stealth Mode.
- The printer should monitor its current operational mode and switch modes only if it differs from the scheduled mode.

## Usage Scenario

For example, if Stealth Mode is scheduled from 22:00 to 07:00 and a print job starts at 12:00 and runs for 24 hours:
- The printer operates in Normal Mode from 12:00 to 22:00.
- It switches to Stealth Mode from 22:00 to 07:00.
- It returns to Normal Mode from 07:00 until the job completes at 12:00.
This automatic switching eliminates the need for manual intervention, ensuring that the printer operates efficiently while minimizing noise during nighttime hours.

Racemade, 2024-05-12:
This would be fantastic! On langer "wake up calls" to activate the stealth Mode :)

CZDanol, 2024-05-24:
Hey, we are considering something like this :) Internal ticket: BFW-5317

aderusha, 2024-06-16:
What'd be really neat would be to surface Stealth mode into the API so Home Assistant could control it.  That allows for basically any trigger condition, you could use time as suggested or maybe "whenever my calendar shows I'm in a meeting" or "whenever my PC microphone is active" or "if the TV is turned on" or "the lights in the shop are turned off" etc etc.  Exposing stealth control to an outside system makes the whole thing a lot more flexible and responsive.