prusa3d/Prusa-Firmware-Buddy #3978: [BUG] Bluescreen on Prusa XL & USB-Stick not reading

Stippy19, 2024-05-04:
**Printer type** - XL (2 toolheads)
**Printer firmware version** - 6.0.0 final release
**Original or Custom firmware** - Original
**Optional upgrades** - none
**USB drive or USB/Octoprint** USB flash drive (stick provided with the printer)
**Describe the bug** After powering up the Prusa XL with the stick being inserted already, a loud noise and then constant short beeping occured, paired with the bluescreen, see below. Now the stick is either not detected at all or detected, starts loading a file, then starts showing the preview and then freezes
**How to reproduce** Unsure.
**Crash dump file**  Printer said on the next start that it recognized a crash and wanted to save a crash dump. Unfortunately then froze again and only a file with no content (0KB) was created. I assume the USB stick has been dismounted / stopped being recognized while writing the file. So no file available.
[image]

Stippy19, 2024-05-04:
I am not really sure why, but gently hitting on the back of the printer where the XLBuddy board is connected doeslead to the USB stick being recognized again. Totally unsure, what that is, as I would have thought if, then the bigger board where the LCD is connected to should be the cultript... 
I therefore think, this is indeed not really a firmware bug. Sorry for the confusion

tsamisa, 2024-05-08:
Actually i have the same issue. It resolved when i downgraded to 5.1.2. Hitting in the back, taking the usb out and reinserting it didnt resolve the problem. This is similar when to a situation that arose when i first upgraded to 5.1. Ive been printing non stop with 5.1.2 and it works every time. So i think its a firmware issue. Maybe it is due to a hardware aspect in conjuction with the latest firware (6.0) but the matter of fact is that it doesnt happen to 5.1.2.

thijstriemstra, 2024-05-19:
there are comments about prusa xl usb issues that explain EVM will cause issues with USB since prusa didn't properly use twistedpair cabling for USB, causing it to fail (without error messages). Slapping the printer might cause enough static to bring back usb back to life temporarily. All in all, very concerned about prusa's design choices once again.

Stippy19, 2024-05-19:
I opened the back of the printer and pulled the board out of the card edge connector and put it back in. Then closed the back again. Since then I did not have any issue anymore (fingers crossed!). But @thijstriemstra that would explain that "sudden" issue coming up and going away again by basically "not doing anything".

thijstriemstra, 2024-05-19:
see #3983

closed 2024-07-30 by github-actions[bot]