prusa3d/Prusa-Firmware-Buddy #3982: [FEATURE REQUEST] ADD Support for both E3D REVO heater Cores models (40W and hi-flow 60W)

antimix, 2024-05-05:
**Printer type** - [MK3.5]
**Optional upgrades** - REVO 60W

**Is your feature request related to an existing problem? Please describe.**
The Initial Setup fails with message:

> **Attention, the test has failed. Check below the expected printer setup and adjust it accordingly:**
> Hotend type: **E3D Revo**
> 

As you may know E3D sells the REVO MK3 hotend kit in two versions: **40W** for "slow" MK3S and **60W** for hi-speed/flow intended for usage if updated to MK3.5.  
Currently on the FW 6.0.0, the heater is defined as STOCK, or E3D Revo (as if it were a unique model)
The definition should be divided accordingly to the reality, were Revo produces two distinct models and each one has its own PIDs and features. Probably during initial setup, and the FW  uses the 40W PIDs values on a 60W heater and it fails.

**Describe the expected functionality**
FW definitions should be changed to support the two models. Some code samples to clarify how the FW should be below.

in **hotend_type.hpp**
```
/// Shared for all printers.
/// !!! Never change order, never remove items - this is used in config store
enum class HotendType : uint8_t {
    stock = 0,

    /// Stock Prusa hotend with sillicone sock
    stock_with_sock = 1,

    /// E3D Revo (MK3.5 only)
    e3d_revo40w = 2,
    e3d_revo60w = 3,
    _cnt,
};

/// other code here in real file

static constexpr EnumArray<HotendType, const char *, HotendType::_cnt> hotend_type_names {
    { HotendType::stock, N_("Stock") },
    { HotendType::stock_with_sock, N_("With sock") },
    { HotendType::e3d_revo40w, N_("E3D Revo 40W") },
    { HotendType::e3d_revo60w, N_("E3D Revo 60W") },
};

/// Some hotend types are only supported by some printers, but the enum is the same for all -> hence this filtering array
static constexpr EnumArray<HotendType, bool, HotendType::_cnt> hotend_type_supported {
    { HotendType::stock, true },
    { HotendType::stock_with_sock, PRINTER_IS_PRUSA_MK4 || PRINTER_IS_PRUSA_MK3_5 || PRINTER_IS_PRUSA_iX },
    { HotendType::e3d_revo40w, PRINTER_IS_PRUSA_MK3_5 },
    { HotendType::e3d_revo60w, PRINTER_IS_PRUSA_MK3_5 },
};

```
in **selftest_mk3.5.cpp**

```
static constexpr HeaterConfig_t Config_HeaterNozzle[] = {
    {
        .partname = "Nozzle",
        .type = heater_type_t::Nozzle,
        .getTemp = []() { return thermalManager.temp_hotend[0].celsius; },
        .setTargetTemp = [](int target_temp) { marlin_server::set_temp_to_display(target_temp, 0); thermalManager.setTargetHotend(target_temp, 0); },
        .refKp = Temperature::temp_hotend[0].pid.Kp,
        .refKi = Temperature::temp_hotend[0].pid.Ki,
        .refKd = Temperature::temp_hotend[0].pid.Kd,
        .heatbreak_fan_fnc = Fans::heat_break,
        .print_fan_fnc = Fans::print,
        .heat_time_ms = 60000,
        .start_temp = 40,
        .undercool_temp = 37,
        .target_temp = 290,
        .heat_min_temp = 180,
        .heat_max_temp = 230,
        .heatbreak_min_temp = 10,
        .heatbreak_max_temp = 45,
        .heater_load_stable_ms = 200,
        .heater_full_load_min_W = 20,
        .heater_full_load_max_W = 50,
        .pwm_100percent_equivalent_value = 127,
        .min_pwm_to_measure = 26,
        .hotend_type_temp_offsets = EnumArray<HotendType, int8_t, HotendType::_cnt> {
            { HotendType::stock, 0 },
            { HotendType::stock_with_sock, -25 },
            { HotendType::e3d_revo40w, 40 },
            { HotendType::e3d_revo60w, 40 },
        },
    }
};
```

in **general_response.hpp**

```
/** list of all button types */
enum class Response : uint8_t {
    _none = 0, // none must be zero because of empty initialization of array
    Abort, // when used in selftest, handled automatically in PartHandler
    Abort_invalidate_test, // when used in selftest, ignored by PartHandler, must be handled in test. This behavior allows to run additional code before abort
    ABS,
    Adjust,
    All,
    ASA,
    Back,
    Cancel,
    Change,
    Continue,
    Cooldown,
    Disable,
    Done,
    Filament,
    Filament_removed,
    Finish,
    FLEX,
    FS_disable,
    NozzleType_Normal,
    NozzleType_HighFlow,
    HIPS,
    Ignore,
    Left,
    Load,
    MMU_disable,
    Never,
    Next,
    No,
    NotNow,
    NozzleDiameter_04,
    NozzleDiameter_06,
    Ok,
    Pause,
    PC,
    PETG,
    PLA,
    PP,
    Print,
    HotendType_Stock,
    HotendType_StockWithSock,
    HotendType_E3DRevo40w,
    HotendType_E3DRevo60w,
    Purge_more,
    PVB,
    Quit,
    Reheat,
    Replace,
    Remove,
    Restart,
    Resume,
    Retry,
    Right,
    Skip,
    Slowly,
    SpoolJoin,
    Stop,
    Unload,
    Yes,
    Heatup,
    PA,
    Postpone5Days,
    PRINT,
    _count,
    _last = _count - 1,
};

```

in **selftest_hotend_specify.cpp**

```
LoopResult CSelftestPart_HotendSpecify::stateAskHotend() {
    // Revisit this when new hotends are added
    static_assert(hotend_type_count == 3);

    switch (rStateMachine.GetButtonPressed()) {

    case Response::HotendType_Stock:
    case Response::No: // If only stock & sock is available, only yes/no do you have sock dialog is shown
        rResult.hotend_type = HotendType::stock;
        return LoopResult::RunNext;

    case Response::HotendType_StockWithSock:
    case Response::Yes: // If only stock & sock is available, only yes/no do you have sock dialog is shown
        rResult.hotend_type = HotendType::stock_with_sock;
        return LoopResult::RunNext;

    case Response::HotendType_E3DRevo40w:
        rResult.hotend_type = HotendType::e3d_revo40w;
        return LoopResult::RunNext;

    case Response::HotendType_E3DRevo60w:
        rResult.hotend_type = HotendType::e3d_revo60w;
        return LoopResult::RunNext;

    default:
        return LoopResult::RunCurrent;
    }
}
```

excerpts from **general_response.cpp**

```
Prusa-Firmware-Buddy-MK3.5\src\common\general_response.cpp:
         R(Print),
         R(HotendType_Stock),
         R(HotendType_E3DRevo40w),
         R(HotendType_E3DRevo60w),
         R(HotendType_StockWithSock),
         R(Purge_more),

excerpts from general_response.hpp
```

```
Prusa-Firmware-Buddy-MK3.5\src\common\general_response.hpp:
        HotendType_Stock,
        HotendType_StockWithSock,
        HotendType_E3DRevo40w,
        HotendType_E3DRevo60w,
        Purge_more,
        PVB,
```

and so on. Remember to update also the PID definition table for the 60W.

Patrick-Schmitt, 2024-05-22:
Just expressing my interest here as well. Please support both Revo Heater (40W and 60W). Thank you!

Prusa-Support, 2024-06-18:
The use of third-party components is not directly supported however, as speculated on https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3764, our developers are already making their considerations about this.

Michele Moramarco
Prusa Research

ripnyt-ripnyt, 2024-06-23:
adding supporting voice for this.

dukestah, 2024-07-14:
I would highly appreciate it too.
Revo 6 Quickchange worked fine on MK3S but after upgrading to 3.5 not anymore, I'm very disappointed.

appleimperio, 2024-07-27:
Please add support for the 60W version.

bjlled, 2024-08-19:
Please add support for the 60W heater.

ChrisKeeleyGitHub, 2024-08-28:
@danopernis Would it be prudent to re-assign this feature request to another user? It looks like @JakoobCZ 's github account has been inactive since February of this year.

dukestah, 2024-08-29:
I can confirm that the Revo 40 W 24 V version works fine on the MK3.5. I've set the heater option to Revo, just to be on the safe side...

antimix, 2024-08-29:
Now I have just purchased the new REVO High Flow nozzle, so having a 60W cartridge it would work better. 
But unfortunately for now I will have to use the 40W heater since the 60W is not supported.

lovemetwotimes, 2024-08-30:
I only have a 60-watt heater. Only after I bought it did I realize that this option was not supported. After installation, the printer does not pass the hotend test and displays a warning before each print. The prints come out perfect.

antimix, 2024-08-30:
I got the same error, but I did not want to risk since a bad handled 60W heater can cause severe risks. So I dismounted the 60W and installed again the 40W. I am waiting PRUSA to support it.

Prusa-Support, 2024-09-07:
Thanks for the patience but this is a third-party component not directly supported by us on this printer model.
We would need E3D to contact us and set in place a collaboration for the specific task, otherwise, the possibilities for our developers would be extremely limited.

I believe that escalating this request through E3D company would be the best course of action. Thanks for understanding.

Michele Moramarco
Prusa Research

jfreed-dev, 2024-09-10:
> Thanks for the patience but this is a third-party component not directly supported by us on this printer model. We would need E3D to contact us and set in place a collaboration for the specific [...]

Has anyone opened a request with E3D? I am in the same boat as @antimix  and @lovemetwotimes

antimix, 2024-09-11:
Today I have done an official request to E3D to collaborate with PRUSA in order to develop the 60W support in the PRUSA 3.5 Firmware via Ticket# 126214

Let's hope they will get in contact with PRUSA developers soon to solve the issue.

Regards

antimix, 2024-09-18:
Here is the last response from E3D support:

> _Hi there,
>  
> Thanks for getting back to us!
>  
> I have already passed on your request to the team, as well as our engineers for further consideration. We haven't reached a decision regarding this yet. However, I would suggest that you keep an eye on our socials for any updates regarding this matter._

bjlled, 2024-09-18:
Is this project full open source such that anyone could write it and do a
PR??? Cuz I’ll sure take a crack at it I think….

antimix, 2024-09-24:
Hello, I got the official answer from E3D regarding the future develop of the 60W heater on MK3.5 below:

 _

> Ayoub Khouyi (E3D Online)
> 
> Sep 24, 2024, 08:45 GMT+1
> Hi there,
>  
> Thank you for your patience!
>  
> After checking with the team, we regret to inform you that **there is no confirmed deadline for developing** the aforementioned firmware as all of our engineers are busy with current projects. **However, they confirmed that it is on their priority list**. 
>  
> Once they finish developing the firmware, in cooperation with Prusa, we will update all our clients through our social media and emails.

_

So now we have just to wait for them  or someone else capable of fork and develop this change.

xyz3ddesign, 2024-10-31:
Had a Mk3s running Revo since Revo came out - zero issues, really great system imo.
Upgraded to a Mk3.5s without researching (like a fool) any issues , found myself here reading the above sadness.
E3D and Prusa have been working together for as long as I can remember, very surprised we at this impasse.

options:
wait for Prusa / some hero to develop the required changes 
OR
swap out my Revo system and upgrade to Nextruder and purchase all the nozzle sizes again but in nextruder 

a Mk3.5s sitting around being useless feels wrong. 😞

antimix, 2024-10-31:
@ xyz3ddesign  I was in your same situation, and dismounted the 60W REVO heater and installed the 40W REVO heather version that is supported. Now I am using my MK3.5 That was for me better than spend more and convert it to a MK3.9. I liked REVO and it was working well. If you don't own a 40W Revo heather, then you can decide to purchase it (not the whole set, just the heather) and reuse the MK3.5 with REVO and wait the support of the 60W. Remember than the heater should be the PRUSA version.

xyz3ddesign, 2024-10-31:
@antimix , that's solid advice and makes the most financial sense. suppose ill go get a 40W heater so.... 🤢 
sure who likes high flow anyway?

Ill be attending Formnext in November , ill bring up the 60W lack of support with Prusa and E3D direct

antimix, 2024-10-31:
@ xyz3ddesign regarding the High Flow with my MK4S I am learning that **with high flow it's not all sunshine and roses**.
I though to use the 60W on the MK3.5 for high flow, but then I discovered that it goes with the new REVO HF Nozzle (Yes it exists 😊), so I purchased the new HF REVO 0.4 nozzle, but it stays in the box for now, waiting for the 60W be operative.
However on the MK3.5 for now I want to keep some stability, so I did not changed anything for now and I want to use the standard 0.4 REVO nozzle.

IconicLight, 2024-12-01:
Another upvote for supporting the 60W heater on a 3.5. Was planning to upgrade to the 3.5 with the Black Friday sales, but reading around, apparently, there is no support for the 60W heater which I'm running on my MK3S+ currently. Really don't feel like buying another heater when this one is only a few months old.

JohnnyricoMC, 2024-12-23:
> Thanks for the patience but this is a third-party component not directly supported by us on this printer model.

That's a bit a cop-out. For the mk3(s+) family Revo-specific firmware builds are published (including builds for the HF60 heater core) and the mk3.5 is marketed as an upgrade for said family. 

You'd expect all mk3(s+) hotend configurations that got official support (e3dmv6, revo6, revo6-hf60) to be supported for the mk3.5. (Revo high flow nozzles are even for sale in the PR webshop) But both PR as well as e3D appear to be dragging their feet?

danopernis, 2025-01-02:
Adding reference to internal ticket.

I would really love to have the support for REVO for those of you who are upgrading from MK3, but some cooperation from REVO itself would be required. For example, I was talking to our testing department and we in fact only have one 40W REVO nozzle available at the moment. Releasing untested firmware, especially when it involves the heater sounds like a potential fire hazard to me.

Not sure how the 8-bit firmware REVO support actually works. @gudnimg @3d-gussner @DRracer do you know?

3d-gussner, 2025-01-03:
@danopernis E3D and some community members tested the Thermal Model on the modified MK3S and provided some data to get a [baseline](https://github.com/prusa3d/Prusa-Firmware/issues/4105) for the REVO and REVO HF.

**E3D is responsible for the REVO firmware and support.**

Prusa "ONLY" compiles and publishes the firmware for these two versions.

The per-release firmware is not only sent to Prusa Test team but also to E3D so they can test and update their support team about upcoming changes with the new release.

There are also few other E3D REVO changes in the firmware compared to the stock MK3S
- Show the nozzle size on [LCD screen when IDLE](https://github.com/prusa3d/Prusa-Firmware/pull/4563)
- Nozzle change menu doesn't heat up to 280°C and warns the user when above [40°C](https://github.com/prusa3d/Prusa-Firmware/pull/4600)

My concerns about the 60W heater is that it might be at or above the limit of the electronics and monitoring circuit on the Buddy board.

To be able to monitor the heaters more efficient the buddy board has some current measurements implemented. We can see the current usage in the Info - Sensor screen.

On the MK3S we "only" monitor the voltage.

This current measurement circuit probably works only in a defined range. The stock printer uses a 40W heater cartridge, so the current of any "working" 40W heater cartridge including their tolerances will be within that circuit range.
A faulty or shorten heater cartridge will draw a lot of current and this circuit may have a hardware / electronic safety feature which cuts off the power.

Another concern are the connectors and cables used. The 24V heater connection uses two cables and two pins in total 4 for VCC and GND from the buddy board to the love board.

I am sure that the double pins and double cable and cable awg are way above the expected current draw to ensure the safety.

But can these also handle 60W cartridge currents and what happens when the cartridge is in the higher tolerance range?

bjlled, 2025-01-03:
I am pretty sure the Loveboard is irrelevant to the conversation, since
you’re using a Revo/Revo Heater core you aren’t using the nextruder.   We
are talking about exclusively the mk3.5.

Ezrae, 2025-06-06:
Just adding another vote for support for the 60W heater core!

CZDanol, 2025-06-06:
Loveboard is not irrelevant, the MK3.5 toolhead still needs to use the same connector, because that's the one that's on the mobo

patrickTUD, 2025-07-19:
> I have a 40w Revo Six. I have selected that nozzle in the hardware selection but it still fails even though it gets to the highest tempurature. Is this what happens for everyone even with the 40W.

No. I use my M3.5s with a 40w Revo Six. No issues.

bjlled, 2025-07-19:
Per the Prusa site itself, the Loveboard does NOTapply to the MK3.5.  the 3.9 literally is the addition of the Loveboard and Nextruder.   MK4 adds new motors.  You aren’t using a Revo if you’re using the nextruder.I really would love to see this get some traction; I’m about ready to switch to klipper on my MK3S and give up on giving Prusa $$$ for this upgrade.Sent from my iPhone

bjlled, 2025-09-19:
Definitely don’t want this issue going stale.  Sent from my iPhone

Not stale.

antimix, 2026-05-10:
Hi guys, we are now in 2026 and many things are changed. In addition to the MK3.5S+MMU I have now an MK4S and a CORE One+. There is not focus any more to have the REVO 60W working on the MK3.5 that has been basically replaced by the big brothers MK4S and Core One+. I was using the MK3.5 for single and multicolour print with the MMU3S, but now all the prints are done on the MK4S and CORE One. Moreover also my multi-colour print are now done by the new SnapMaker U1 multi tool, and the MK3.5S is just getting powder. 
The reality is that the soon arriving INDX will make the MK3.5 even more obsolete, because there will be zero interest from E3D to develop anything on an old product, and PRUSA is really putting all their energies in the release of INDX PRUSA Edition. 

Note here the learned lesson: once the PRUSA INDX will be released and supported by the PRUSA FW, don't assume that if, in a future, Bondtech will release a special version of the hotend (e.g. 100W High Temp 500° nozzle) it will be automatically supported in the PRUSA products. What PRUSA release as _PRUSA EDITION_ will be supported by PRUSA, but whatever will be release by Bondtech by their own, don't assume or pretend it will be automatically supported by PRUSA. 
Exactly how in the past was happening when a skilled maker  installed a different extruder on the MK3. He wrote his own customised FW and released as open source with no intervention of PRUSA.

Said that, the good news is that MK3.5 FW is open source, so at this point, if there is anyone in the community (and for sure, there is) that is able to fix and support the 60W heater, this is the moment to do, before the printer will be too obsolete.

Best Regards