prusa3d/Prusa-Firmware-Buddy #3983: [BUG] <5T XL Printer stops during color prints without error>

corywagner, 2024-05-05:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [Prusa XL 5 toolhead (upgraded from 2T]

**Printer firmware version** - [6.0]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [None]

**USB drive or USB/Octoprint** - [USB]
  Please let us know if you print via USB flash drive or USB/Octoprint [USB]

**Describe the bug**
  
Prusa stops printing sometime during print. It only seems to happen with long, muti-color prints (I have a 5T XL).  Stops  randomly. Inducing power panic usually allows it to resume (for a while).  There are no errors and the printer log does not indicate a failure.  If I induce power panic, it will resume right where it left off.  Sometimes it will complete the print, sometimes it will fail later.

Here is what I tried:

- Replaced thumb drive with 2 different "high reliability" drives.  Prints still inevitably failed on each.

- Ensured that drive was formated for FAT32 and uncorrupted

- Bought a Triplight UPS in case power fluctuations were causing it.  Again, problem remains.

- Used a USB extension cable 

- Rigged a heat sink to the USB stick

- Completely reset and re-calibrated the printer

- Upgraded to 6.0 firmware

-  Disabled PrusaConnect and PrusaLink and printed directly from USB drive

- Used different sliced models, so I don't think it is a gcode issue

**How to reproduce**
 Unfortunately, I can't reproduce it at-will.  It happens during long color prints.  It does not see to happen with single color.

**Expected behavior**
  I expect it to complete the print, or at least throw some error so we can troubleshoot.

**G-code**
  Tried to attach bgcode, but would not let me.

**Crash dump file**
  Please send the crash dump file to Prusa by emailing it to reports@prusa3d.com. Sharing this file is important and helps us investigate the bug.
  Do not share the file publicly, as the crash dump contains a raw snapshot of the printer's memory and may include unencrypted sensitive information.

Unfortunately, there is no crash dump file to send because the printer does not know it crashed**Video**
 
 Please attach a video. It usually helps to solve the problem.
No video because I am never in the room when iit fails.  Pic attached for what it is worth.[
[image]
[image]
](url)

Mr3Deee, 2024-05-05:
I have seen the same problem - BUT I didn't even need to do a multi color print for this. I had a simple PETG single color print which casued this. It happened many times (about every 30 minutes). In the end it wasn't even able to continue the print and it stopped.

My solution was to disable PrusaConnect. My feeling is/was that this was due to concurrency where multiple clients tried to access the same USB stick at the same time.

Someone else was raising the comment that there might be hardware problems. As I was assmbling the 3D printer, I know that the cabeling was a standard ribbon cable and as such it might be suceptibe to EMV. The stepper motors might therefore be feeding stray signals into the wire which will cause the USB bus to become unusable (and time out) for a short time. However I hope that this is not it. One thing to try is to use 6.0.0, turn Prusa connect and Prusa Link on buit the new "silent modes" off and see if it still happens.
IF this would be USB stick related, it MIGHT be a good idea to use an empty and / or a slow (or fast) stick. If there is enough power coming out of the port you might want to try using an external SSD drive(?) If it gets even worse (faster errors) it might be a good indication for HW problems.

GeasyCreasy, 2024-05-06:
I have the same issue. There is a large thread on Prusa forums that describes the same problem. (https://forum.prusa3d.com/forum/original-prusa-xl-tool-changer-hardware-firmware-and-software-help/usb-error-mid-print/paged/4/) 

The issue is maddening! Upvote to get this problem addressed quickly please!

ddrl46, 2024-05-07:
I have also been running into this issue very consistently while attempting to print a fairly short single-material print. Every time the printer halts with a "USB drive or file error, reconnect the drive". Sometimes the print is recoverable from this point by reconnecting the USB stick, however quite often the USB stick is not recognized anymore until the printer is reset via the front panel.

[image]

Attempting to disable phase-stepping as well as prusa connect / networking all together did not resolve the issue. I've tried multiple types of USB stick, all with the same result.

As a last ditch effort I downgraded to 6.0.0-alpha, which is was previously running without issue, and after doing so I was able to print the same g-code I was previously having issues with multiple times without any issue. Attached is the g-code in case it helps.

[Quarter Baseplate HBM Ladekast v1_0.4n_0.2mm_PETG_XLIS_1h56m.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/15239250/Quarter.Baseplate.HBM.Ladekast.v1_0.4n_0.2mm_PETG_XLIS_1h56m.zip)

tsamisa, 2024-05-08:
I have the same issue. I downgraded to 5.1.2 to be able to print

Blizzke, 2024-05-14:
I have the same issue with 6.0

4 color / 4 hour print stopped at 33% (halfway in a layer) without any message or error.
Knob was still active so I "paused". Status went to "pausing" right away,  but it took almost 2 minutes before the printer actually responded by lowering the build-plate and parking the head. 

Had a pretty huge blob of filament which I cut away as best as possible and then resumed. 
The print finished without any further issues.

goodridm, 2024-05-14:
I had this happen today at the 48% point of a 3 day print. The printer just stopped at what seemed to be the end of a layer before color change. I noticed the nozzle temp was dropping and Prusa Connect said it was cooling to 70C from 250C. I set it back to 250C and it seemed to heat back up, but no printing resumed. I pressed Pause on the printer and it just seemed to stick at "Pausing...". I couldn't resume form Prusa Connect. The display responded in sound to my clicking, dial moving, but nothing happened. I found your post, pulled the plug and plugged back in. The print job resumed! However, I now have a slight layer shift.

The model was 20MB 3mf file, 11.83x11.97x12.07, 1722592 facets, two colors, 1499 tool changes

madnotangry, 2024-05-14:
I have had this happen on several different prints. The XL would pause during a print with toolhead changes or single color prints alike, sit for about a minute or two, and then park itself without an error or notification at all. It would let me resume it from the LCD, but I couldn't connect to the printer with Prusa Link or Prusa Connect unless I cycled power to the printer. I tried to alleviate the problem with a Hard Reset, reflashed the 6.0.0 firmware, and had the problem continue. 

I downgraded to the 6 alpha firmware and have had no problems whatsoever.

TheShuttlesworth, 2024-05-17:
Currently experiencing this issue. I have so far had two stoppages on a 4 hour total print. Multi-material (2), sent through Prusaslicer via Prusa Connect, running 6.0.0 firmware, Prusaslicer 2.7.4. It seems to happen on tool changes, right before it picks up to go to the back. I have cut power and it has started back. Trying the pause, pull USB stick, etc. has not worked. Firmware 6.0.1 just released, but I don't want to lose this print. I will report back if it comes up again in 6.0.1.

Torx999, 2024-05-19:
Have the same issue with 6.0.1

XBurgy, 2024-05-28:
Hello, I have the same problem, specifically a USB drive reading error during printing on the Prusa XL 5T. The issues started after updating to FW 6.0 (before this update, the error never occurred!) and they persist even after updating the FW to version 6.0.1. Sometimes it is possible to resume printing using the continue option, but sometimes the print is completely lost. Based on personal experience, I would therefore guess that there is a bug in the printer's firmware since version 6.0.

corywagner, 2024-05-28:
This is a consistant and ruinous issue, please address it. If you have any doubts, please check this reddit thread:
https://forum.prusa3d.com/forum/original-prusa-xl-tool-changer-hardware-firmware-and-software-help/usb-error-mid-print/paged/5/#post-712173

nhustak, 2024-05-28:
I'm on an 5 head as well and see this issue on non-multi-material prints as well.

Mr3Deee, 2024-05-28:
The issue is still present. For me the rate can be reduced by clearing the USB drive and turn off PrusaConnect. However - even then it is still happening. (Also 5 head)

I would still think that it must be a polling issue where PrusaConnect, PrusaLink and the printer itself are using the drive at the same time and running into a timeout. (It _could be maybe that due to_ cabelling a higher error rate would occur, fostering the problem) 

So my guess for a fix would be: Increase fault tolerance for read timeouts by increase timeout values. Additionally it might help to decrease the priority of PrusaLink/PrisaConnect on the USB bus. Even better would be to download the entire G-code blobb before the printing starts. I assume that there must be enough memory to do so.

Mr3Deee, 2024-05-29:
Okay - I am now in a totally broken state. No matter what I do (clean up the USB drive, use a new one with the same files, ..) - it claims that it cannot read the USB drive. Great. 10hrs print time, ~300g filament behind me and it's simply broken. 

Sorry - but this SW bug needs to be fixed. I will see that I can go back to 6.0.0. alpha. But this is really sad.

corywagner, 2024-05-29:
Beware, I went back to 6.0 Alpha and it didn't fix the problem.

Mr3Deee, 2024-05-29:
Okay. Thanks for letting me know!
I did think a bit more about the problem and here are some more ideas for whom ever fixes this bug:

1. I did yesterday try to connect the printer several times from a very spotty network connection from the beach. I would think that this might have triggered a remote action (reading a file, making the printer wait, etc.) which was then never released and this caused then in turn a stall (e.g. not closing a file, hence blocking the device in some way)
2. It was said that turning off and back on the printer would make the printer to continue printing. As the printer is going in that case through a full power / reboot / reset cycle, the only way to explain that the printing continues is that the server is triggering that. This has the smell that the server is actually interrupting the printer session and also able to resume it as the printer has lost all internal state at that time. So - the remote operation MUST be involved in all of this.
3. It is somehow conceivable that an underlying library / kernel change around I/O / Mutex / event handling is triggering this.

I am not sure if there is anything I could supply from my side which allows you to analyze this further - but please let me / us know if there is anything we can help with! (And yes, I am also a programmer, I do know what I am talking about even though I didn't program 3d printers, but I have done low level, OS level, server side, ... programming).

sauer-m, 2024-05-31:
Same here! DualHead- and SingleHead-Prints on 6.0.1. 
Please get this fixed!

danopernis, 2024-05-31:
Hi! We are already aware of the issue, however we are still unable to track it down :disappointed:

corywagner, 2024-05-31:
What can we do to help?  I get the phenomenon at least once a day.

CZDanol, 2024-05-31:
I've possibly found a reproducer now, so I'm trying to diagnose and fix. We might contact you later when I manage to fix it to confirm that it is indeed the issue you were having :)

Teque5, 2024-06-04:
Same issue firmware v6.0.1 PrusaXL w/5T print; screen didn't show USB icon but could detect when detached (showed no usb error).

Was able to save print via "power panic". Will downgrade to v5.x in the meantime.

peterjhl, 2024-06-07:
Struggling with this problem, right now I'm trashing 2 out of 3 prints. Tried everything! Pretty frustrated right now.

TheShuttlesworth, 2024-06-07:
Update from my post above. I ended up updating to 6.0.1 firmware. Also, I noticed that my problem started after I connected a co-worker to the printer via PrusaConnect. I looked into this because I lost my connection to the printer. So I set up the "Teams" on PrusaConnect per the forums recommendations. After I did this, I reconnected to the printer through the team on PrusaConnect and I have not had an issue since. Multiple prints of various times, materials, heads, etc.

This may not help anyone, but I wanted to share. I do see that there is another firmware update (6.0.2) and I plan on updating within the next week since I am stable.

tutulino, 2024-06-11:
Same problem with XL 5T after update to 6.0.0, usb file or drive error message at mid print

Juan-VV, 2024-06-16:
This is driving me crazy, I have wasted about 200 hours of print time and many kgs of filament on this, and it making me consider not using the printer any more until there is a solution. Contacting Prusa support tomorrow.

Unsure how to help but please let me know if I can do anything let me know. Of course this happens randomly... finished 2 long prints, then today it happened again at 85% of a 12 hour print. I could not recover. 

5T although most of the time only printing with 2 heads.

I cannot tell when the problem started, it was about 2 months ago but one month ago gotten worse. Tried many random suggestions online...
- Reconnecting all cables to screen, front and back.
- Reflashing firmware.
- New USBs, smaller, slower, etc., reformat drive, etc.
- Turning off wifi, prusalink prusaconnect.

When the failure happens most often I cannot recover. Sometimes, unplugging and plugging does the trick. Occasionally, unplugging printer and restarting gets the print  to start but this is not always the case. 

Now I am waiting for some ferrite coils to put around LCD screen cable on the back, apparently that helped someone in facebook group...

peterjhl, 2024-06-23:
I have put a ferrite choke on the LCD cable, I suspect that it is "noise" that disturbs the signal. Have been printing for 7 days now and all prints have gone without problems. Just to check if it's luck and it has no effect anyway, does anyone else want to try the same "trick"?
[image]
[image]

corywagner, 2024-06-23:
I'll try anything to get this to work.  Do you have a link to the choke?  Also, what are we seeing in the second picture?

Blizzke, 2024-06-23:
> I'll try anything to get this to work. Do you have a link to the choke? Also, what are we seeing in the second picture?

Given that it is earlier timestamp wise I suspect that would be the "before" :-) 
I'm sure I have some of these around somewhere, time to check the junk drawer.

peterjhl, 2024-06-23:
> I'll try anything to get this to work. Do you have a link to the choke? Also, what are we seeing in the second picture?

 I deleted picture, I posted the wrong picture, forgot to delete it again, sorry, but it was a picture before fitting the ferrite choke.   It is similar to this one.  [https://uk.rs-online.com/web/p/ferrite-sleeves/1232490](url)

Mr3Deee, 2024-07-22:
Just checking in here: 

- Did anyone try yet the 6.0.3 or 6.0.4 version to see if the problem went away?
- Or did the ferrite sleeve do the trick?

For myself - as said earlier, disabling PrusaConnect and clearing everything off the USB stick did do the trick - for now. But I'd like to activate PrusaConnect again (and I do not like the idea of constantly scrubbing the USB drive).

tsamisa, 2024-07-22:
> Just checking in here:
>
>     * Did anyone try yet the 6.0.3 or 6.0.4 version to see if the problem went away?
>
>     * Or did the ferrite sleeve do the trick?
>
>
> For myself - as said earlier, disabling [...]

i asked in the forum under the announcement for the new firmware if it solves the issue but again i got no response. I also ask them every time a new fw 6. version comes to at least tell us where the problem lies , so as not to go around installing ferrite chockes, fans etc with no reason but that Jan Krativcill or something that announces the updates doesn't seem to bother to answer.

TheShuttlesworth, 2024-07-22:
> i asked in the forum under the announcement for the new firmware if it solves the issue but again i got no response. I also ask them every time a new fw 6. version comes to at least tell us where [...]

I had the same results as @Mr3Deee . Once I disconnected PrusaConnect (after setting up a secondary operator account) and reconnected to the machine with a "Team", everything worked and has worked since. I am currently running the latest firmware with no issues. 

I don't have a lot of experience with this platform, but I believe that setting up a PrusaConnect connection and disrupting it by changing settings caused the issue. You could try scrubbing the printer of all external connections to PrusaConnect or other wireless systems, reconnect and start from there. That's essentially what I did. 

Also, I have seen some hesitant to update to the latest firmware. It has had no issues for me doing so and if you did and have issues its not too much trouble to revert back. Just update and see what happens.

corywagner, 2024-07-22:
I installed the ferrit sleeve and thought it solved the problem.  I ran the printer non-stop for 5 days without an issue.  I was ready to declare victory, but on day 6, it stopped with the same behavior as before.  I then induced a power panic and it resumed for about 2 hours and then stopped again.  I installed the newest firmware, but that made no difference.
I am now exploreing the theory that this is resulting from overheating on the buddy-board.  I am printing a mode to attach a fan to the buddy board enclosure.  More to follow.

corywagner, 2024-07-22:
Sorry to be dense TheShuttlesworth, but what do you mean by "Once I disconnected PrusaConnect (after setting up a secondary operator account) and reconnected to the machine with a "Team", everything worked and has worked since. I am currently running the latest firmware with no issues."?

corywagner, 2024-07-22:
Shuttlesworth, thank you for the explanation. Since I don't work in a multi-user environment, I'm wondering if this is pertanat to my situation.  Nevertheless, it does not hurt to try.  I'll do anything to get  this thing working reliably.  Thanks again.

Mr3Deee, 2024-07-23:
Thanks for all the feedback!

My thinking was that the constant polling of the USB drive from multiple "clients" (PrusaLink & PrusaConnect & Printer) + potential read error and repeats (due to cabling) - which might even be accelerated by the temperature of the printer - was causing this.

The problem with the PrusaConnect is that it is not only showing the progress, but also available jobs with images and so on. Depending on the browser and your local memory on your computer you might have then background activity from your computer to the server to the printer and the result would be constant polling - and maybe even re-loading of printer image assets (which will likely need to be computed from gcode, hence read).

As such the required throughput from the USB drive can get quite steep. And then there is the cabeling. Dependent on how the EMV between the cables adds interference and USB errors, the read process has even to be repeated (until a timeout is hit) - and on top comes the heat up of the USB stick which then also reduces the speed the stick can operate on (and who knows - maybe even the error rate goes up).

Alternatively this can all be totally wrong and it is really only a firmware issue.

Juan-VV, 2024-07-26:
Hi all - to share my experience which might be anecdotal as it is a single user's. My problem was quite bad, couldn't complete 4 hour prints, most of the time couldn't recover, wasted A LOT of plastic. This was a month driving me crazy....
- Tried many USB drives, ferrite cores, etc.  Followed this thread and forum and facebook, and tried most things people suggested.
- At this moment I was on 6.0.2.
- Then reached to Prusa support. We tried:
-  Exchanging screen with my MK4. Didn't help. 
- Went back to 6.0.0 with a full memory wipe, using procedure Prusa support shared.
- This helped for a few days... and then problem came back. 
- I upgraded to 6.0.3, and it worked much better. A couple USB errors which were recovered from.
- But printer worked well for a few weeks.
- Upgraded to 6.0.4, it hasn't happened since then.

Notes:
- Using same USB drives as always. Now using Prusa's included one!
- Ferrite cores were removed.
- Back cover of MCU was removed as I was worried about heat. Want to install fan at some moment but been lazy...
- Support from Prusa support was excellent, got a replacement filament sensor I needed, and with this problem they helped me and shared openly they acknowledge the problem, are working through it, and I agree given the nature of it this takes time to solve.

Any questions let me know....

m10d, 2024-07-30:
It sure would be nice if prusa would add more detail about what root-cause analysis suggests is at fault here.  From all the various forums (github, prusa forum, reddit) it sure seems this is EMI related due to the long unshielded ribbon cable housing usb D+/D-, SPI, and i2c.  I don't have any experience with USB, but spi and i2c are (much) more commonly employed on one board or board-to-board with very short cables only.  Let alone running adjacent to motors.

Prusa, would you please consider:

- weighing in on what issue should track this, so we're not looking at mulitple, in multiple repos
- provide a summary of your testing, in case some very easy workarounds might bear some fruit:
    - separating cable traces, wrapping/shielding w/ ground on single end
    - ferrite / choke (what spec, containing which lines of the ribbon)
    - outright replacing the ribbon with something else.  Note:  [Pinnouts incl' xBuddy and LCD boards](https://www.prusa3d.com/downloads/Electronics_drawings/MK4_electronics_schematics.zip), termination of the lcd cable with [TE Micro-MaTch](https://www.digikey.com/en/products/detail/te-connectivity-amp-connectors/9-215083-4/4486392) (via [forum](https://forum.prusa3d.com/forum/user-mods-enclosures-nozzles/mk4-display-cable-to-short-for-the-prusa-original-enclosure/#post-657125))

TheShuttlesworth, 2024-08-14:
Update for everyone. I finally had the failure come back last night. I left a 2 extruder print go over night and came in this morning to the same error we have all seen. The print stopped at 82% and the total print time was around 24 hours. 

After finding it in this state I did the following. 
 - Pausing the print and restarting did not work, since it never got out of the pausing sequence. I killed power and turned it back on and the print started right where it left off. 
 - Shortly after, less than 5 minutes, it stopped again. It did so in the middle of a layer, so layer change, tool head change, etc. did not set this off for me. 
 - I killed power a second time and upon restart I wanted to test if it was indeed a PrusaConnect or PrusaLink issue. So after the print had started up, I went through the menu and turned PrusaConnect and PrusaLink off in the setting. The print has been running without issue since. 
 - The print has about ~28 min left, and I can let everyone know if this comes up again. 

In summary, I double down on my stance that this is a PrusaConnect or PrusaLink issue. For clarity, I have not modified the printer in any way. Only updating with fully released firmware when Prusa makes it available. My theories are either they are polling too often to the device for updates on progress or state, or there is an interrupt on the processor to make this communication and this is enough of a hiccup for the machine to freeze. 

Hope this helps move things along so we can no longer have this issue with our prints.

corywagner, 2024-08-15:
I thought I solved it by putting chokes on both end of the USB cable, but alas, no. I did not get a mysterious stop for  couple of days .  Just as I got my hopes up, the problem came back...then it came back again, and again all within 3 hours. There is no pattern that I can discern. Luckily, a power cycel restarts the print from where I left off (thank you power panic), so I can't say I'm wasting material, but I am losing a whole lot of time.  I should not have to watch my printer every minute in case it stops.  I wish they'd get a fix for this!

thotland, 2024-08-23:
I installed chokes, reverted to older firmware (5.1.2) and it still pauses for no reason or sometimes on USB coms fault. Upgraded to day to 6.1.2 and USB error 30 minutes into a 24 hour print. This is ridiculous! And I'm ready to try whatever it takes to add t the community here to solve this problem. 

I run Prusa XL 5-head

CZDanol, 2024-08-24:
@thotland on the 6.1.2, does it allow you to resume standardly without the need to do power panic?

TheShuttlesworth, 2024-08-24:
> @thotland on the 6.1.2, does it allow you to resume standardly without the need to do power panic?

In my experience the pause resume cycle gets hung in a please wait screen after you select pause.

Droneracer101, 2024-08-25:
I had this issue and hoped that upgrading to the new 6.1.2 firmware but it is continuing to happen. Just did. If I power cycle it it does resume. Pausing the print does not work.  It never completes the pause. I will try disabling PrusaConnect on my next long print.

CZDanol, 2024-08-26:
@TheShuttlesworth that's a different issue then and not the USB one - USB error should actually result in the print getting paused.

Droneracer101, 2024-08-26:
> @TheShuttlesworth that's a different issue then and not the USB one - USB error should actually result in the print getting paused.

Sorry I worded this poorly.  The print does pause, it stops printing and it just sits there with no error on the screen.  What I meant was I try choosing pause on the screen and as other commented, it doesn't seems to do anything or ever give me the option to resume.  Power cycling the printer does force it to resume.  So for now I have Prusa Link/Connect disabled on the printer and have the printer connected to a Alexa manageable surge strip to make it easier to power cycle.  I'm still on my first print after trying this.  but 13 hours in and it hasn't stopped yet.  Way to early to call though.

CZDanol, 2024-09-04:
@Torx999 did the USB icon appear in the screen header after inserting the USB?

If yes, then you might have hit an issue that we have an internal ticket BFW-5775 for. We're working on fixing that.

Torx999, 2024-09-04:
For now I can't say if there was an USB icon. But next time I will check it.

Mr3Deee, 2024-09-12:
I have heard that the new 6.2 update should address this. It was said that there might be an alpha version (?). I will not be able to test this for the next 2 weeks though.

CZDanol, 2024-09-13:
Yes, we're hoping the 6.2 release would fix the USB file errors. And yes, currently, it's only released as alpha.

thotland, 2024-09-13:
I have been working with online chat tech support. They had me do a couple things that did not resolve the issue: new USB drive- reformatted FAT32, and a hard reset- installing the latest firmware (6.1.3 at the time). Again, there was no improvement, still pausing with no error (and sometimes getting the USB error). 
They next had me pull the Display cable out and inspect it. There were a couple kinks that I flattened out. I then carefully reinstalled the cable, making sure that each section of the cable lay flat on next, never twisting or alternating their order. 

After reassembly, I've printed for 38 hours with nor pauses or errors. This is not definitive yet, but is a marked improvement from pausing every hour or so.

m10d, 2024-09-13:
@CZDanol i have been very confused at the (lacking) level of detail prusa has been willing to provide here. Especially as FOSS, whatever *eventually* truly fixes the issue will be out in the open, right? Why not give those (brief) breadcrumbs now?

To the best of my knowledge, no detail has been shared here on what type of issue prusa even believes from its RCA - is there a likely ~bug is the way the protocol is used "directly", or are these fixes adding more robust handling around rfi edge cases (as users have speculated given the much longer than common usb & (i2c?) Ribbon cable paths), or other?

We, as users, are happy to test (and indeed expected to, as its our implicit contract of how we support FOSS projects) - but please be respectful of the users' engagement by providing a bit of detail on what's going on.

Keep in mind, several releases have been suggested to users for this issue (which, also worth noting, has been very expensive in failed prints for some users). We do NOT ask for no bugs, or demand the "first try" works - we just want better transparency on what has changed that merits re-testing.

Thank you.

CZDanol, 2024-09-13:
Uff, okay, story time.

It sometimes happens that the USB drive stops communicating properly with the printer, likely due to electrical noise. Over the time, we've implemented better recovery sequences (let's say somewhere around 6.0 and 6.1). These seem to be working well.

But, there's also this thing called media prefetch, which is basically a big buffer that loads the printing gcode from the USB drive and feeds it to the system. This ensures that we always have enough gcodes in the memory so that we can continue printing whlie we're fetching more data from the USB on the background. This is also important for the USB recovery, because during that, the drive can be unresponsive for a second or two.

The main problem was, the original media prefetch was implemented... not ideally. There were possible race conditions, and if the USB recovery triggered in the wrong moment, it could get stuck. So, we've completely rewritten the damn thing. That's what happened in 6.2.0. This is supposed to be a proper fix.

FlowerDIY, 2024-09-16:
I was seeing this issue once every 16 hour print just before a tool change with no error on screen. The issue is always recovered by inducing power panic. 

The issue was improved by adding an MCU fan, but it returned after a few more prints.

I will try the Alpha firmware tonight on a long print of a facehugger to see how it fares.

WeaselWasher, 2024-09-18:
> Uff, okay, story time.
> 
> It sometimes happens that the USB drive stops communicating properly with the printer, likely due to electrical noise. Over the time, we've implemented better recovery sequences (let's say somewhere around 6.0 and 6.1). These seem to be working well.
> 
> But, there's also this thing called media prefetch, which is basically a big buffer that loads the printing gcode from the USB drive and feeds it to the system. This ensures that we always have enough gcodes in the RAM so that we can continue printing while we're fetching more data from the USB on the background. This is also important for the USB recovery, because during that, the drive can be unresponsive for a second or two.
> 
> The main problem was, the original media prefetch was implemented... not ideally. There were possible race conditions, and if the USB recovery triggered in the wrong moment, it could get stuck. So, we've completely rewritten the damn thing. That's what happened in 6.2.0. This is supposed to be a proper fix.

Nice. Thank you for this. 
I work with drives at work on a regular basis, so we know what noise can do to communication. 
These are the cables we use: (different headers)

[image]

I was thinking about pulling the trigger and building my own shielded cable and running it outside of the extrusions, grounding the shield as close to the main control board as I can. 

Also - I HATED stuffing that delicate cable into those extrusions. Even with care, I managed to pinch one.
Perhaps we need a retrofit and separate cable run. Sometimes software can only go so far. 

Not only does electrical noise cause an issue reading from the USB but also potentially for writing to it too - since the wifi is in the back of the printer. It's a double opportunity to corrupt the file.

CZDanol, 2024-09-19:
There are CRC checks in the USB protocol, so you can't be getting write
errors from any errors on the link layer. At worst, the usb packet is
discarded and the printer has to resend it. Same when reading. Plus there
are crc checks on higher layers as well (binary gcode).

I'd suggest first trying out the 6.2.0. We are really trying to fix the
issue for everyone without requiring any printer modifications.

WeaselWasher, 2024-09-19:
That's great. I tried 6.2.3 and it seems really prevalent right now. I can't get to 10% without it faulting. It's weird though.. I was printing for a couple weeks without issue. 
This print is just one tool too. I've done more complex ones. I'm not sure why this suddenly popped up.
I'll try the alpha tomorrow.
I've had some weird glitches.. like all the orange on my LCD went purple. [image]

I think I might do a factory reset along with the firmware

CZDanol, 2024-09-19:
@WeaselWasher 6.2.3 does not exist. Perhaps you updated to 6.1.3?

WeaselWasher, 2024-09-19:
> Perhaps you updated to 6.1.3 instead of 6.2.0-alpha1?

Indeed. I mistyped. I updated to 6.1.3 last night. I will try 6.2 today

WeaselWasher, 2024-09-19:
I did five things since last night 

1 - left the unit powered off all night 
2 - flashed the 6.2 alpha - I did not do a factory reset
3 - updated prusa slicer (newest update notification popped when I turned it on this morning)
4 - formatted a different USB I had around (16 gig fat 32)
5 - started a fresh platter in PS (last time I re-sliced, I did it from a saved 3mf), loaded the STL, simplified it in PS (the STL from blender has a couple million faces - I did not simplify it last time) - set it up again, re-sliced it and put it on the fresh usb 

This print has been going for 2.5h and so far so good. I still have connect and link disabled. 
I was getting errors in 0-10% it's at 15% now

I'll update if the print finishes (tomorrow) or if it doesn't.. sooner. 

I would have liked to take smaller steps but I'm getting impatient to get my project done. I could do some tests after and try and narrow down the culprit. After I get a few more prints done. (Though, the fact that I had probably been powered up for a month without a long power off - will take time to duplicate 😉)

[11 more replies]

closed 2024-11-30 by Prusa-Support