prusa3d/Prusa-Firmware-Buddy #3988: [FEATURE REQUEST] Custom filament presets for the mk4

Samuelmill50, 2024-05-08:
**Printer type** - MK4

**Is your feature request related to an existing problem? Please describe.**
when loading filament on the mk4, there's a list of possible filaments; however, when printing with something that needs a special temperature, I just have to go with the option that is the closest. Also, it means that when starting a print, the printer will come up with a filament warning.

**Describe the expected functionality**
it would be good if there were an option on the list of filaments to enter a custom temperature for that material or an option to disable that warning since I couldn't find a way to disable it.

murk-sy, 2024-05-08:
Duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3915

Samuelmill50, 2024-05-08:
> Duplicate of #3915

Sorry I didn't see that feature request

closed 2024-05-21 by danopernis

Samuelmill50, 2024-05-21:
Thanks for closing, sorry I didnt realise it was a duplicate.

Thanks,

Samuel Miller