prusa3d/Prusa-Firmware-Buddy #3995: [BUG] <Enter comprehensive title>

Racemade, 2024-05-12:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [XL]

**Printer firmware version** - [6.0 final]

**Original or Custom firmware** - [Original]

**USB drive or USB/Octoprint**
 USB Drive

**Describe the bug**
 I started an Oberlicht Print in stealth Mode. All Fine and when the Print was ready I turned off the printer.

Today I started a new Print and initial the stealth Mode was ON (so the Parameter "stealth Mode ON" from the last Print was already on.) This is Not the Problem. The Problem is, when the Print started I turned OFF the stealth Mode. The ICON in the Display disappear, but the acclerations does Not reset and stay in stealth Mode.

**How to reproduce**
  Finish a Print in stealth Mode. Turn off the Printer with stealth Mode Off. Turn the Printer on and start a new Print. 
Heile printing Turn Off the stealth Mode.

**Expected behavior**
  
When turning Off the stealth Mode, normal acclerations should be used.

MaximilianBuegler, 2024-05-12:
I am encountering the same issue

CZDanol, 2024-05-13:
Hey, thanks, we've already heard about this and will look into this asap.

CZDanol, 2024-05-14:
Hello guys, we've (hopefully) created a fix to the issue. Should get out in the next release :)

CZDanol, 2024-05-21:
Okay, this should be fixed, I see noone complaining. Closing the issue. Please let us know if the problem persists.

closed 2024-05-21 by CZDanol