prusa3d/Prusa-Firmware-Buddy #3998: [BUG] Enabling `Stealth Mode` causes a loud screeching noise during X/Y homing

muesli, 2024-05-13:
**Printer type** - XL

**Printer firmware version** - 6.0.0

**Original or Custom firmware** - Original

**Optional upgrades** - None

**USB drive or USB/Octoprint**
  No print involved in this report.

**Describe the bug**
  Enabling `Stealth Mode` reproducible causes a loud audible resonance during X/Y homing. With it disabled, the homing is reasonably quiet. Once the homing is finished and a print starts however, `Stealth Mode` seems to be effective and is a lot quieter than the regular mode.

**How to reproduce**
  After upgrading to 6.0.0 I've calibrated and enabled `Phase Stepping` as well as `Stealth Mode`. During X/Y homing the printer is producing a very loud noise, almost screeching. This can either be triggered manually from the menu or as part of a print job.

**Expected behavior**
  Homing with `Stealth Mode` enabled should not cause loud noises.

**G-code**
  No G-Code involved.

**Video**
  Here's an example recording of the issue:

https://github.com/prusa3d/Prusa-Firmware-Buddy/assets/146378/d1ec0750-6261-4a18-8897-18a9910dfc7e

Racemade, 2024-05-13:
+1
Happens 3 times - same noise.

CZDanol, 2024-05-14:
Hey, thanks for reporting. We will look into it.

Internal ticket number: BFW-5496

CZDanol, 2024-05-16:
Hello, we've hopefully addressed the issue. The fix should get out in the following releases.

chujian, 2024-05-18:
This situation still exists in firmware 6.0.1

CZDanol, 2024-05-18:
Yep, sorry, it was too late to put it into 6.0.1. Should be in the next one :)

danopernis, 2024-06-07:
This should be fixed since v6.0.2 via commit https://github.com/prusa3d/Prusa-Firmware-Buddy/commit/2dbfb4dee266fd8d884ee6f5265b93e5ea3980aa

Can you please confirm that the issue is fixed?

chujian, 2024-06-11:
The problem has been solved and the sound is normal.

CZDanol, 2024-06-11:
> The problem has been solved and the sound is normal.

Thank you, closing :)

closed 2024-06-11 by CZDanol

gpaulit, 2024-12-17:
This is still happening randomely to my XL (Its only a week old, assembled 5Tool one purchased directly from Prusa) The home at the start of a print will be very loud and grindy sounding when stealth mode is enabled. And then it goes back to normal right after.

Firmware is 6.1.3+7898 on the XL, files were prepared and sent to the printer via Prusa Link from Prusa Slicer running 2.8.1+MacOS-arm64 if this is required.

CZDanol, 2024-12-17:
@gpaulit please re-test on 6.2.0-alpha2 and possibly record a video