prusa3d/Prusa-Firmware-Buddy #3999: Wifi Based file transfer problem.

VH-fixer, 2024-05-13:
MK4, MMU3, Original fw 6.0.0  USB drive.  Verbatim 16GB formated to FAT32.  
Windows ll   PrusaLink or transfereing from the slicer via wifi both have similar problems. 

Monitoring of the printer via PrusaLink or Prusaconnect has no problems.   Deleting filles or starting a print in PrusaLink no problems. 
I have even put in a new Wifi card to make sure there was not a memory problem on the wifi card.  No change in the problem. 

I have had no errors reading  the USB drive.  No problems loading the USB from a PC.  No problem when the file has been loaded from the PC to the USB and printed.  

My system was doing wifi transfers well before switching to V6.0.0
I have been doing a lot of testing with 2 files that have been proven to do PrusaLink transfer successfully.  One is under 4mb and one is 11mb.  

Boot the printer, make sure it is connected to the internet.   Using Prusalink to drag-n-Drop, File that is 3.8mb shows 100% immediately.  Sometimes it faults.  Just shows 100% for more than a minute and then errors.
[image]

Boot the printer, make sure it is connected to the internet.   Using Prusalink to drag-n-drop the same 3.8mb file.  It finishes successfully in a few seconds.  
The LCD switches to the print mode for the file.   Here is an example of the LCD display.
[image]
If I try to upload a 2nd file at this time, without changing the selection on the LCD, the upload will fail, often leaving a complete temp file on the USB.   This can HANG UP THE PRINTER COMPLETELY.    Reset or reboot needed.
If I try to Exit this screen when the 2nd file is uploading, the printer will hang up immediately. 

The only way I have been able to transfer a 2nd file is to exit to the main screen after the first file transfer.  Select the print button, parse through the files to read the USB and then exit back to the main screen.   

WiFi transfer from PrusaSlicer have similar problems.  It can take a long time and then errors with this message about the USB is full.
[image]

The USB only has a few files and is 95% unused.   

After many of these transfers errors a temp file is left on the USB drive that is about the same size as the file transfered.   When I compare these files they are identical except near the end where there are differences.  I am including the file and a resulting 1.tmp file from a failed transfer.
[transfer-files.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/15297421/transfer-files.zip)

 I expect the Wifi file transfer to be reliable at all times.  I have done what I can to validate the hardware and show the Wifi card itself is good.     I believe this is V6.0.0 software related since wifi transfer was working much better before this release.

I have an MMU3 and cannot go back to a previous revision.

VH-fixer, 2024-05-14:
I am adding another example of the failed file transfer.  This one 7.5mb.  Transfer from Prusa Slicer 2.7.4 showed USB full error as shown in the message above.    I am sending the master file and the tmp file left on the USB after the failure.   Most of the tmp that exists is correct, then the end is messed up compared to the master.  
The tmp file was dated AHEAD of the time shown on the printer.  I think the tmp file created is dated based on GMT vs. the shown clock time on the MK4e.   Would that cause a check problem?   I am in Texas USA with the -6 hours from GMT set on the printer I believe. 
[bug-files2.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/15301449/bug-files2.zip)

VH-fixer, 2024-05-18:
More information.   If I send the file from the PrusaSlicer to upload and run,  it usually works.   If the MK4 has just been rebooted or idle for a while.    But almost never successful when it is just uploading the file.

VH-fixer, 2024-05-18:
Appears 1/2 fixed.  
Just loaded 6.0.1 for the MK4.  Tested uploading files.  Using PrusaLink, I did 7 uploads of various file sizes.  All successful, one after another.   Never got 2 in a row without failure on 6.0.0.   I did a file compare of all 7 against the masters once on the thumb drive.  All match 100%.
Even though this is not specifically listed as a fix, it appears wifi file transfers have been improved.

PrusaSlicer 2.7.4  testing on 6.0.1 is not as reliable.   Uploaded 1 file.  Good, then trying to upload a 2nd file, upload only, hung the 3D printer as noted above.  
After a reboot of the printer, tried again, got the USB full error.    Is this still a firmware problem or related to the slicer only?   I don't know.

danopernis, 2024-05-31:
@VH-fixer thanks for all the info, it is much appreciated. We will try to reproduce and fix this.

VH-fixer, 2024-06-04:
I tried to enable Prusaconnect.  It was enabled before 6.0.0, but I disabled it for Fw 6.0.0.  Using FW 6.0.1 I enabled it.  It seemed to work initially.  The problem is when uploading a file, Upload to printer only, the printer readies the uploaded file for printing.  Same as is done with Prusa Link.  The Printer often becomes unresponsive and hangs when I try to exit out of the selected file and go back to the main menu.  Similar to what I described above.   Even rebooting, this happens.  It appears PrusaConnect is pushing the file to be readied.   I get in this loop, reboot, after 30-60 seconds, file is readied again, I try to exit, unresponsive and reboot.    After enough frustration I disabled PrusaConnect on the MK4 again. 

After working with 6.0.1 for a fiew weeks,  the problems remain.  I think I was just smarter working around the problems.  Wifi transfer from PrusaLink, PrusaConnect or PrusaSlicer is not reliable.   They all exibit the same problems which points to a core problem in the Firmware or I have a bad Buddy Board.  Everything else works.  Even the monitoring of the printer over WiFi.   All tests passed and calibrations are good.  Wifi Transfer in all cases worked much better before FW6.0.0.  

Additional Details, Buddy Board version is 27.  My MK4 is inside an Original Prusa Enclosure.   My router for the Wifi is within 10 meters of the MK4.   WiFi response is great.

VH-fixer, 2024-06-22:
I have changed the USB drive.   I went back to out of the box setup.   Luckily I had an 8GB, USB2 drive in my collection I acquired over the years.  Quick reformat to FAT32, 4096 Allocation size.   File transfer reliability has improved markedly.  I have had file transfer improvement before, after formatting a new drive, but this is a much better improvement.  Even loading another file while a part is loaded has not caused a lockup.   Currently on firmware 6.0.2.

The 16GB  Verbatim drive was a USB3.
Does USB3 casue the problem? Greater than 8 GB drive size?   A combination of both?   Or will it come back as the drive is used?  I will see in the next week or two, but currently hopeful.

If it is USB3 or greater than 8gb related, then this still needs a long term fix.   Getting good quality, USB2,  8gb drives is getting hard.   There are many cheap ones similar to what I got with the printer,  that soon had formatting problems.

VH-fixer, 2024-06-23:
Have done many file transfers now using the USB2 drive.  Very confident it has solved the issues.    Will not risk using USB3 drives until I see a specific fix in the firmware for the problems above with USB3.

closed 2024-09-02 by github-actions[bot]