prusa3d/Prusa-Firmware-Buddy #4000: [BUG] Cold Pull Wizard shows the wrong URL and QR code

neuron-whisperer, 2024-05-14:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK4

**Printer firmware version** - 6.0.0

**Original or Custom firmware** - Original

**Optional upgrades** - None

**USB drive or USB/Octoprint** - USB drive

**Describe the bug**

The new Cold Pull Wizard displays the following information:

"Welcome to the Cold Pull Wizard. Prepare a 30cm piece of PLA filament and follow the instructions. For more details, visit: prusa.io/13702"

The Wizard also displays a QR code that, I believe, points to the same URL.

[image]

However, the provided links do not show any instructions - they merely forward to https://www.prusa3d.com/ . 

**How to reproduce**

Initiate the Cold Pull Wizard. Follow the displayed URL or QR code.

**Expected behavior**

The pages indicated by the URL and the QR code should provide information about the cold pull technique.

**G-code** - n/a

**Crash dump file** - n/a

**Video** - n/a

neuron-whisperer, 2024-05-14:
The correct URL appears to be:

https://help.prusa3d.com/article/cold-pull-mk3-9-mk4-13702_445071

danopernis, 2024-05-16:
This was already fixed once in https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3962 but the fix somehow didn't persist :confused:

CZDanol, 2024-05-17:
Didn't persist? Both issues are on the same FW version

danopernis, 2024-05-31:
https://prusa.io/13702 is once again working :crossed_fingers:

closed 2024-05-31 by danopernis