prusa3d/Prusa-Firmware-Buddy #4001: [BUG] MK3.5 X-Axis Crash when starting a new print

Kikkawaryu, 2024-05-14:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3.5

**Printer firmware version** - 6.0.0

**Original or Custom firmware** - Original

**Optional upgrades** - None

**USB drive or USB/Octoprint**
Both

**Describe the bug**
  After a print is finished the print head will move all the way to the right and moves the bed forward to remove the part, on the mk3s it would move the extruder to the left 
  
  If you leave the head to the right and start a new print, it will home the print head to the right, the entire printer crashes 
  I turn printer off, manually move the head to the left and start the print where it will auto home with no issues, levels and prints complete

Its almost like the printer does not know the extruder is all the way to the right and needs to bring it left before it can autohome

**How to reproduce**
Print a file have the printer finish with the extruder to the right, then start a new print right after 

**Expected behavior**
I would expect the print to finish, you remove the printed part start a new print but when the print starts it would auto home correctly remembering the head is to the left or have the extruder move all the way to the left which is X 0 so its ready for the next print

The printer passes all self tests with no issues, recently had a new X Axis printed with new bearings but the issue occurred both before and after the new printed pieces 

It happens regardless of the gcode used, all files are sliced in Prusaslicer using the stock MK3.5 profiles 

Printer is mainly in stealth mode however has been tested in normal mode presenting the same issue

Kikkawaryu, 2024-07-11:
The issue still persists on 6.0.3 firmware

the print ends with the extruder to the right if you then start a print right away even from prusa connect where it knows the extruder is at the right, it will still fail to home and crash

if you either manually move the extruder by hand or use connect to move the extruder to the left it will home and print with no issues.

danopernis, 2024-07-12:
I just tried following your steps to reproduce the issue and wasn't able to. Doesn't seem like firmware issue to me, especially since nobody else is reporting it as far as I know.

I suggest closing this issue, reaching to [customer support](https://help.prusa3d.com/article/customer-support_2287) and troubleshoot with them.

Kikkawaryu, 2024-07-12:
I did reach out to support previously with it and they suggested it could be firmware based issue to raise here but I'll re-engage support thank you!

Kikkawaryu, 2024-07-17:
Support had to me change the position of G28 to be from stock Start GCode

G90 ; use absolute coordinates
M83 ; extruder relative mode

M140 S[first_layer_bed_temperature] ; set bed temp
M104 T0 S170 ; set extruder temp for bed leveling
M109 T0 R170 ; wait for temp
M190 S[first_layer_bed_temperature] ; wait for bed temp

G28 ; home all

which was changed to 

G90 ; use absolute coordinates
M83 ; extruder relative mode

G28 ; home all

M140 S[first_layer_bed_temperature] ; set bed temp
M104 T0 S170 ; set extruder temp for bed leveling
M109 T0 R170 ; wait for temp
M190 S[first_layer_bed_temperature] ; wait for bed temp

However this still crashed every 1 in 4 prints, I put the G28 back its in normal place in the Start Gcode I looked at the end Gcode instead

The stock is 
G1 X241 Y201 F3600 ; park 

which presents the Extruder to the right

instead I changed the end to be

G1 X-1 Y201 F3600 ; park

Which now puts the extruder at the far left, so its in prime place to Home correctly and start the print

Saved the profile and have printed 6 items back to back with zero crashes, printer was powered off a few times in between to ensure nothing was by chance and it all works!

Closing but leaving comments in case anyone else has a similar issue

closed 2024-07-17 by Kikkawaryu