prusa3d/Prusa-Firmware-Buddy #4004: Mk3.5 Bed Levelling not properly detecting right side

GalacticGreg, 2024-05-16:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - Mk 3.5

**Printer firmware version** - 6.0.0

**Original or Custom firmware** - Original

**Optional upgrades** - None

**USB drive or USB/Octoprint**
  USB Drive

**Describe the bug**
The height of the right side of the bed is not detected properly resulting in the last 10cm dragging the nozzle on the bed. As a workaround I have set the bed levelling to 70um on the right, but it is still so far off that it makes a mess on that side.

**How to reproduce**
  Print something with a large surface.

**Expected behavior**
  The print should fairly uniform across the surface of the bed.

**G-code**
 

**Crash dump file**

**Video**
 I have a couple of pictures; one showing the squish on the right side, even after compensating with 70um, and ; a photo of the SuperPinda sensor on the right side. I thought maybe this could be the problem as the Pinda sensor is slightly off the bed on that last row of probes.
[image]
[image]

MarkusReichart, 2024-05-18:
I also noticed this behaviour on my MK3. 5 and had to correct the right side with at least 30.

Patrick-Schmitt, 2024-05-22:
I experienced a similar issue with my MK3.5, however, my cables(or zip ties) on the left side of the extruder cable bundle hit VERY SLIGHTLY the printer frame. The calibration routine did not catch that as an error and I passed the xyz tests.

But than during first layer calibration I saw that the very right edge of my printbed was printing very low (actualy touching the printbed wit the nozzle) and very thin plastic was on that side.

Only after long troubleshooting I figured out that I had to be super careful with cable routing and zip tie position - so really nothing is in the path of the extruder rails.

This COULD be the issue you are experiencing as well - but it is for sure not 100%. Just a tipp.

Nevertheless, I think it would be beneficial if the calibration routine would be a little bit more strict to catch such issues. Maybe it is not possible to do it more strictly.

magistersmax, 2024-05-25:
See bug report #3851 , I think this is related to what I previously reported.  There's been a number of us discussing this issue since the original mis-configuration of mesh bed leveling for the MK3.5 was initially fixed.

GalacticGreg, 2024-05-29:
Hi @magistersmax, it does look like this would be the same problem.

danopernis, 2024-05-31:
This looks to me like a duplicate of #4004 It may not be obvious at the first glance, because originally the referenced issue was about centering of the maximal volume print on the bed, but thanks to all the commenters it was revealed that it is in fact related to the probe offset. Thanks for the report, I am closing this in favor of #4004

closed 2024-05-31 by danopernis

ripnyt-ripnyt, 2024-08-27:
@danopernis just noting that it appears you self referenced this as duplicate of #3851

danopernis, 2024-08-27:
@ripnyt-ripnyt indeed. Thanks for the correction.