prusa3d/Prusa-Firmware-Buddy #4009: [ENHANCEMENT] MK4 MMU3 Auto Unload at the end of the Print

huzargarden, 2024-05-21:
I believe it would be beneficial to have an automatic filament unload feature when only one colour is used at the end of a print for MK4 and MMU3. Previously, with MK3s and MMU2, there was a single colour preset, but with the Original Prusa MK4 preset, the filament does not unload at the end of the print. I think it would be a great improvement to either create the MK4 MMU3 Single Color preset or add an auto-unload feature to the firmware.

This is from the perspective of the 6.0.0 firmware

unclej84, 2024-05-21:
Do you mean unload from MMU at the front or removing filament from the hotend back to the MMU? Last is done on my MK4 after every single color print automatically - then you can simply rewind it back to the spool as it is not in grip by the MMU.

huzargarden, 2024-05-21:
Hey, I've updated the firmware to 6.0.1, and now it automatically unloads the filament. In version 6.0.0, the filament stays in the nozzle, and I had to unload it manually after each print.

Gyrxiur, 2024-05-24:
I hate it, if I want to print multiple things of same color one after another, it wastes my time.

BugsBunny1403, 2024-05-25:
It's right, it is function we would select or not. Or for exemple, wait timer to unload after 5 min if printer have no response.

unclej84, 2024-05-30:
I agree, an option to turn this on or off (also accessible during running print) would be great - unloading and loading is not only a waste of time but also of material as it purges both times a bit

dompe-9, 2024-07-19:
On my MK3.5 + MMU3 in 6.0.3, on single color print the filament stay in the nozzle at the end of the print... I have to choose the type of filament on the screen to lauch the unload.
Do you add something in the end gcode on prusaslicer ?

closed 2024-09-25 by github-actions[bot]

CZDanol, 2024-09-25:
Duplicate of #3444.