prusa3d/Prusa-Firmware-Buddy #4010: [BUG] MK4 Stealth mode not working

BigAl66, 2024-05-22:
Printer type - MK4
Printer firmware version - 6.0.1

Describe the bug
When I switch the MK4 to stealth mode nothing changes. My old MK3 was dead silent when I go to stealth mode. The steppers on the MK4 are loud an running a bit "rough" all the time. I think the TMC-Drivers are not switched to silent mode at all. Only the max. speed seems to be limited...

To Reproduce
Print normal or with input shaper

Expected behavior
The stepper should be absolute silent when switch to stealth mode (like it was on the MK3)

bfaliszek, 2024-05-22:
My MK4 is also just as loud with and without Stealth mode

mika1801, 2024-06-09:
Same here, mk4 is terribly loud. Stealth mode changes nothing. I wouldn’t have any problem with longer printing times if only this printer wouldn’t be so loud. I bought it, because it was advertised as most silent 3D printer out there. Well, not mine…

d-999, 2024-06-17:
Unfortunately the stealth mode on the MK4 is fake. It doesn't really do anything except reduce the speed and acceleration of the printer... Essentially it just prints slower. Nothing you couldn't do before by changing the slicer settings.
It's a shame, cause it was a really awesome feature and the MK4 has the same stepper motor drivers as the MK3S, so there's no reason one couldn't implement (real) Stealth mode. Like with the MK3S, some features like step loss detection etc. wouldn't work in steath mode. But personally I feel like that's a small price to pay for almost absolute silence.

Glaedr304, 2025-04-27:
does commenting reopen an issue? this is a problem for me as well. I bought a mk4 with some of the expectations that it would have the features of the mk3 as well, with the understanding that it is of course a new version and things change. but between stealth mode not being dead silent, and no collision detection, I am beginning to regret buying this printer.

CZDanol, 2025-04-28:
Reopening

closed 2025-07-07 by github-actions[bot]