prusa3d/Prusa-Firmware-Buddy #4012: Heat warning and shutdown Prusa XL firmware 6.0.1+14848 [BUG] <Enter comprehensive title>

crashkg, 2024-05-25:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

Prusa XL

6.0.1+14848

Original

**Optional upgrades** - [e.g. Filament Runout Sensor]

Prusa connect

Got an extruder heat warning and then the machine shut down and restarted 

Sent a command to preheat the bed and to preheat the nozzle, then sent a command to start the print a couple minutes later

I expected the machine to start the print

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  Please send the crash dump file to Prusa by emailing it to reports@prusa3d.com. Sharing this file is important and helps us investigate the bug.

  Do not share the file publicly, as the crash dump contains a raw snapshot of the printer's memory and may include unencrypted sensitive information.

**Video**
  Please attach a video. It usually helps to solve the problem.

CZDanol, 2024-06-11:
Please contact our support for this.

Prusa-Support, 2024-07-20:
This issue is too generic and very likely related to an actual fault that occurred to the machine.
It would mean that the behavior is "correct" from the point of view of the firmware, hence that the problem can't find a solution in this repository.

Although the problem is probably on the hardware' side, feel free to share further details and evidence about the problem, and useful information for its reproducibility **but** please, _not without consulting with our Customer Support first_ and giving them the chance to solve the problem together with you.

Until then, it seems safe to close this issue as non-firmware related.

Michele Moramarco
Prusa Research

closed 2024-07-20 by Prusa-Support