prusa3d/Prusa-Firmware-Buddy #4013: [FEATURE REQUEST] Option to disable "tools mapping" screen when submitting an MMU printjob via prusa-link

lsr1, 2024-05-26:
I did search for a feature request like this, but was not successful. Apologies if this is a duplicate.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - [MK4]
**Optional upgrades** - [e.g. MMU3]

**Is your feature request related to an existing problem? Please describe.**
_Please note: I am intentionally describing the situation for my MK4 with MMU3_
When submitting a printjob from prusa-slicer via prusa-link to my MK4 with MMU3, I have to get up, walk to the other room, and confirm on the panel of the printer on the "tools mapping" screen that the tools / filaments are correct.

This is annoying, since all the settings are already provided via slicer (and I have my filaments/tools configured there).

**Describe the expected functionality**
My suggestion would be to make the "tools mapping" screen optional, to allow to start a print totally remotely. 

Link to the screen: https://cdn.help.prusa3d.com/wp-content/uploads/2024/03/19c3ae06c83625aa60137b2ccbe41b68_painted-768x576.jpeg (also attached)
[image]

lcemoore, 2024-05-27:
I am 100% onboard with this suggestion.  Having to go to my printer every time I send it a print, completely voids the benefit of prusalink and takes away the fun of using the printer.  Please provide a way to disable this screen for those of use who will never use it.

lcemoore, 2024-06-01:
Addition item with this tool,
I'm still in favor of a disable option for this tool but I've learned a little more and just to get it out of the way, the documentation for this tool is way to scant.  I think most of my complaints and many of others I read is because we were never told that you had to program the filament types in using the load filament advanced screen.  Once you do this the management tool doesn't reappear unless you select a conflicting filament type for that lane in the slicer. 
Now that I know that, I can avoid the screen most of the time, but here is the thing, If I open the load filament advanced screen from the menu when no print job is running, the firmware doesn't show the current firmament settings and it should.  As it exists, I have no way of knowing what filament types are programmed in before I send a job to the printer.

jltx1, 2024-06-01:
100% agree.   First, the original suggestion is still valid.  I should be able to submit remotely and override the query screen if I trust I have it correct.

Even if you use advanced load, you can't ever get the status on the machine that it believes it is in to confirm the filament settings so you have no idea if you will be successful with a remote print, so again, you need the override option.

For safety(?), the override should be ephemeral, a choice at each submit with a confirmation, not a permanent preference state.

currently this is NOT typical Prusa fire-and-forget easy printing

lsr1, 2024-06-03:
> Addition item with this tool, I'm still in favor of a disable option for this tool but I've learned a little more and just to get it out of the way, the documentation for this tool is way to scant. [...]

Holy ***! I never new this was even an option to not have this screen come up. **Especially** since - as you mentioned - there is no hint WHAT SO EVER that the printer is aware _what_ filament it has loaded (the fact that you have to use the "advanced" filament load is also something I would have never figured out). Who designed this!??!

seifer69er, 2024-06-14:
Agree, we should be able to bypass this if we choose to!, otherwise our prints will sit there waiting for confirmation

mj8w, 2024-07-01:
I too am finding this really annoying and not in line with what I expect from Prusa tools. Please address this. 
1.) Allow to bypass this screen. We already have a way to set the filament, in prusa slicer, we shouldn't need to do it twice.
2.) Provide some instruction on how Prusa expected this to be used. I have no idea what problem this is supposed to solve, and the screen has no meaning to me.

davejhilton, 2024-07-04:
Even better would be to have a way to actually do the filament remapping from within Prusa Link (if wanted) before starting the print. You could ideally, of course, just skip the remapping, and start the print as-is, if no remapping is needed. But it would be great if there was a way to remap the filaments when starting the print remotely. Without this option, I rarely end up using PrusaLink/Connect to start prints, because I almost always need to see the remapping screen (on the printer) to confirm that I've got the correct filaments loaded in the correct order—and often remap things before printing.

My use-case is that I often don't remember which order I had the colors set up in the slicer when creating the gcode... since I'm often trying to re-print models I sliced a while ago. If PrusaLink/Connect could show the same remapping screen that the printer itself does, it would also serve to show which color swatches are expected for each filament slot, which would be great.

Having the remapping screen as a feature in PrusaLink/Connect (in addition to the originally-requested option to skip or dismiss the mapping screen, and continue the print) would make PrusaLink/Connect MUCH more usable with the MMU installed, because it would allow us to reliably start prints remotely!

NeverMissAFlag, 2024-08-13:
I really have nothing to add here other than my support for a way to remotely bypass this screen. I assumed the screen would allow me to remap the colors before printing which would be great but it needs a remote override so I don't have to visit the printer in the other room to start.

SnowmanPacha, 2024-08-14:
+1 !! Same issue encountered. This filament mapping menu effectively makes remote operations via WiFi pointless: What is the point of uploading a gcode when you then need to go to the printer to validate it ? Worse even, the printer maintains the heatbed temperature forever, waiting for the operator to clear the mapping menu locally.
Proposal: make this menu optional in the printer settings, and/or add it to the PrusaLink interface so that it can be cleared remotely.

Mike-FUT, 2024-09-20:
I'm not understanding this request.
MK4S with MMU3 here with latest release firmware on MK4S and MMU3.

When I select the MK4S-MMU3 printer profile in PrusaSlicer the mentioned screen from this request to assign a filament never shows up.
In fact I sometimes wish it would show up so I could use SpoolJoin with MMU prints. But for me this screen only shows up for single-material prints (when I select the non-MMU3 profile in PrusaSlicer and only then I can use SpoolJoin).

lsr1, 2024-09-20:
>>When I select the MK4S-MMU3 printer profile in PrusaSlicer the mentioned screen from this request to assign a filament never shows up.
Plz see my comment from June 3. AFAICT the screen showing up depends on how the filament was loaded on the printer - with "load filament advanced screen" or "load filament" screen.

ljford7, 2024-11-23:
Just adding my two cents. This is horribly annoying and needs to be a higher priority in getting resolved. What is the point of being able to send files to the printer, if you have to physically interact with the printer to get it to start printing?

dhouckii, 2024-12-27:
This makes me want to throw away my MK4.  I can't tell you how many times I have started a print job moved on to other tasks only to come back to the heater disabled message because I didn't manually start the print job.  I understand how some people would want this feature, but why punish the rest of us who just want to use our printers the same way we have for 10 years.

seifer69er, 2025-10-15:
Keeping this issue alive as its still unresolved