prusa3d/Prusa-Firmware-Buddy #4015: [BUG] Phase Stepping Causing Skipped Layers 6.0.1

Trini8ed, 2024-05-27:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - Prusa XL

**Printer firmware version** - 6.0.1

**Original or Custom firmware** - Original

**Optional upgrades** - None

**USB drive or USB/Octoprint**
 Issue is persistent printing over USB and Octoprint

**Describe the bug**
 Phase stepping is currently broken/unreliable in 6.0.1 it causes extruder skips. We had our farm blocked for 1-2 days because of this firmware updating thinking it was a cause of new nozzles we've put on per our maintenance cycle. After disabling phase stepping the issue went away. I would expect for this to be caught on Prusa's end with a bank of XL's running production prints to ensure bugs like this don't carry through and cost me man hours and order delays, quite embarrassing.

We ran a test across 7 XL machines testing a variety of things, printing on USB vs OctoPrint, Older firmware, and Phase stepping off on 6.0.1

What we found is USB or OctoPrint didn't matter, all machines on 5.12 worked fine, along with all machines on 6.0.1 with Phase Stepping disabled.

**How to reproduce**
1) Flash 6.0.1 on an XL
3) Ensure phase stepping calibration has been reset and ran again
3) Start any print on the machine USB or OctoPrint

**Expected behavior**
Not to have skipped extruder lines...

**G-code**
 Will work with any GCODE happens on PLA and PETG, I assume it happens on all materials. Using 2.7.4 stock profile, I suggest printing a cylinder and stretching it tall and running test prints if you see anything like this.

**Crash dump file**
 No crash dump file

**Picture**
[image]

Trini8ed, 2024-06-28:
Been over a month why is there no updates here?

Trini8ed, 2024-08-04:
Over 2 months now

Prusa-Support, 2024-08-06:
Thanks for reporting and I'm afraid you are right when you expect Prusa to catch a similar issue. We can't reproduce it though and there haven't been similar reports, not that I know.

If this is reproducible on multiple Prusa XL machines of yours, this must very likely be something rather specific about your printing farm setup, about the used filament, or the print settings.

There are clearly differences in the motion system and parameters with and without phase stepping, in fact, crash detection is disabled for the phase stepping, but again, slight friction that phase stepping can't fight may be related to something specific with your printing conditions.

Please consider repeating the phase stepping in the complete absence of vibration: no operating machines around the printer being calibrated.
Possibly, give it a try with Prusa-approved filaments and filament profiles: any in the [Configuration Wizard](https://help.prusa3d.com/article/configuration-wizard_1754) but the ones we sell are our best tuned.
Let the [official support](https://help.prusa3d.com/article/customer-support_2287) work together with you on this problem so that our agents can learn more about the reproducibility of the problem or help you sort this out once and for all, from evaluating the belts to studying your filament and files.

Michele Moramarco
Prusa Research

Trini8ed, 2024-08-08:
> Thanks for reporting and I'm afraid you are right when you expect Prusa to catch a similar issue. We can't reproduce it though and there haven't been similar reports, not that I know.
>
> If this is [...]

We've tried 6.0.4 and re-ran the calibration and so far things have been fine across all our machines I would mark this as resolved at this point I might re-open if we have this happens again. We had it happen on 1 more machine, but realized it was still on 6.0.1 after seeing the skipping happen we flashed it to 6.0.4 and re-ran the phase stepping calibration and it went away on the same printer with the same gcode, nozzle, and everything.

I don't think it's related to the setup I think there was some configuration issue of the phase stepping calibration between 6.0.0 and 6.0.1 that is directly firmware related. We ran the phase stepping calibration in 6.0.0 then did an upgrade to 6.0.1 and ran the calibration again and that's when we saw the issue. We also ran the calibrations in isolation with no vibrations from any other machines.

Also FYI this was relayed to support and our B2B rep 2 months ago and we heard nothing but crickets and it took from then until now until we heard anything. I'm hesitant for reaching out to support at this point. We don't have the time to debug Prusa's firmware and their machines, we spent a premium for these machines and unfortunately we are the first to pay the price constantly dealing with poor quality control on Prusa's end when it comes to testing their products and putting out updates that breaks our farm.

closed 2024-10-15 by github-actions[bot]