prusa3d/Prusa-Firmware-Buddy #4019: Prusa XL Park Nozzle on Pause (M601)

pipcorona, 2024-05-28:
When printing with filaments that tend to ooze (PETG & TPU for example) and having a pause to insert objects into the print (magnets or nuts) the toolhead moves off the the side and starts to cool the nozzle. Unfortunately, material oozes out, so when the print starts back up, the filament isn't flowing smoothly and leaves gaps in the perimeter usually.

**Describe the solution you'd like**
Park the toolhead in its spot during a pause. This would prevent ooze due to the nozzle tip being blocked.

**Describe how it would work**
At the point where the pause happens, the XL parks the toolhead in its spot. Then when the user resumes, it grabs it and resumes printing.

**Describe alternatives you've considered**
On my MK4 I usually do a manual color change (M600) instead of a pause (M601) and reinsert the filament after I insert the parts so it is flowing smoothly. The XL has the ability to plug the nozzle though so it would be easier to handle it that way.

Here is an example of the first perimeter after the pause.
[image]

sammy123k, 2024-06-23:
I've just discovered this behavior on my first print with inserted magnets in my Prusa XL, and it was quite surprising to me. I had oozing on the bed while it was paused, and the X carriage itself was positioned in a way where I didn't have full access to the print either! 

It seems like a no brainer that on reaching a M601 pause in the GCODE, the printer should park the current tool as part of the pause, and pick the tool again up on resume. There are manual workarounds for this, but it feels like what is being asked of here is a much safer default behavior than the current behavior.

rubillos, 2024-08-03:
Try setting your pause print gcode to this: (the `G4 P100` commands are important, won't work properly without them)
```
G1 F{min(350.0, travel_speed)*60} ; set fast travel speed
P0 S1 ; park current tool
G4 P100
M601 ; perform printer pause
G4 P100
T{current_extruder} S1 L0 D0 ; pick current tool
G4 P100
```

hickengr, 2024-09-04:
I also have the same problem with filament oozing and causing problems after resuming the print.  rubillos' workaround is ok if you remember to do it.  The XL has the ability to easily part the tool on the dock to save us the trouble of entering extra gcode.  Please implement this feature.

pipcorona, 2024-09-11:
> Try setting your pause print gcode to this: (the `G4 P100` commands are important, won't work properly without them)
>
> ```
> G1 F{min(350.0, travel_speed)*60} ; set fast travel speed
> P0 S1 ; park [...]

Super delayed reply to say thank you and this works great. Although I just had a print where it didn't park, so I need to investigate. It definitely had the custom G-Code so I'm not sure why it reverted to the old pause style. I do still think it would be nice to have this be the default behavior on a pause.

Cloud-Tickler, 2025-04-22:
> Try setting your pause print gcode to this: (the `G4 P100` commands are important, won't work properly without them)
>
> ```
> G1 F{min(350.0, travel_speed)*60} ; set fast travel speed
> P0 S1 ; park [...]

Thank you for the code example rubbilos. Which slicer Version have you been using? 
I tried this, and it doesn't work for me. The placeholder 'current_extruder' is not getting replaced when slicing. So the printer parks the tool, but then continues without picking the tool again.

CZDanol, 2025-04-23:
Afaik no

Paliur, 2025-05-09:
> > Try setting your pause print gcode to this: (the `G4 P100` commands are important, won't work properly without them)
> > ```
> > G1 F{min(350.0, travel_speed)*60} ; set fast travel speed
> > P0 S1 ; park current tool
> > G4 P100
> > M601 ; perform printer pause
> > G4 P100
> > T{current_extruder} S1 L0 D0 ; pick current tool
> > G4 P100
> > ```
> 
> Thank you for the code example rubillos. Which slicer version have you been using? I tried this, and it doesn't work for me. The placeholder 'current_extruder' is not getting replaced when slicing. So the printer parks the tool, but then continues without picking the tool again. It also doesn't work when I try to send the command in Octoprint via terminal
> 
> Send: T[current_extruder] S1 L0 D0 Recv: echo:Unknown command: "T[current_extruder] S1 L0 D0" Recv: ok
> 
> Has there been a change on that command?

In Prusa Slicer, this works without a hitch. However, you mustn't add custom code to a certain layer. You need to modify the pause parameter directly in “Printers -> Custom G-Code - > Print pause G-Code”.

Then, when you slice your model, you need to add a standard pause to the layer you're interested in.
The code will then be interpreted correctly.

Sorry for my English ! Hope you understand what I mean !