prusa3d/Prusa-Firmware-Buddy #4020: [BUG] Mk4 with MMU3, filament already loaded cause filament break on extruder and mmu3

vittorio76, 2024-05-28:
**Printer type** - MK4, with MMU3

Printer firmware version - 6.0.1
Original Firmare

The situation is a bit complicated but I will try to explain it.
During the printing process with the mmu3 and a filament already loaded in the extruder, a modal appeared.
The modal asked if I wanted to change the filament (answer YES) or use the one already loaded in use (answer NO).

I decided to use the PLA filament that I had already loaded.
From that moment the extruder started to emit really loud and very alarming noises and at the same time the MMU3 was emitting other honestly unpleasant noises in the loading of the filament.
I thought that a timeout or the sensors would interrupt this worrying situation and instead I had to perform a reset if not others to stop the procedure in progress.

Obviously I tried various maneuvers to unlock the filament that was practically nailed inside the extruder and also unfortunately in the mmu3.

I opened the extruder, removed the gears and found a piece of filament completely eaten by the gear with a flattened part, the nozzle was empty.

I opened the MMU3 and found the filament completely eroded.

At this point I believe that choosing NO in the answer triggered an unexpected situation, it seems that the mmu3 pulled the filament to expel it while the extruder tried to push it.

This is what I was able to verify, sincerely I do not intend to repeat the operation since I had to disassemble and clean a few things and that from the noises it made something could certainly get ruined.

RyanJGray, 2024-06-04:
Just chipping in to say that I've witnessed this exact same outcome with our MK4 kit and MMU3 kit.

We're running the latest firmware (6.0.1).

The filament was flattened in the gears in the Nextruder, and I had to disassemble the planetary gearbox to release the filament. The printer was making some concerningly loud grinding noises.

I also had to release and flip back the top cover of the MMU3 and the filament was completely chewed away to powder.

I can't remember exactly what triggered it, I think it was when I told the printer to manually unload the filament from the MMU.

It's like the Nextruder started pulling the filament back towards the MMU when the MMU wasn't yet pulling, and the slack on the filament got trapped in the Nextruder gears.

It was not fun to clean that up.

codingcatgirl, 2024-06-09:
I'm unfortunately experiencing a similar issue. The filament also gets chewed up into the extruder gear. This happens on nearly any MMU3 print about 50-100 toolchanges in, sometimes earlier, sometimes later. I haven't managed to get any successful prints aside from the sheep once, and i think that was just lucky cause it doesn't have many tool changes.

Eventually, the MMU just complains that the filament can't be pullsed out, and that's correct, cause instead of it being pushed back into the MMU the extruder has flattened it and showed it back into itself.

There are some pictures. I'm in contact with prusa support for weeks hoping to find a solution to this issue, so far unsuccessfully. Chiming in here because i'm glad to see people facing similar issues.

Since i have to always disassemble the extruder to get the filament out i can almost do it with my eyes closed now, but that won't get me non-failed prints either. A lot of things have been tried, including firmware resets, so far no solution has emerged.

[image]
[image]

Krosta8, 2024-06-09:
I have the same issue. Could it be some incorrect tension on the Nextruder or something? 
[image]
[image]

codingcatgirl, 2024-06-09:
@Krosta8 i see you're also using an enclosure. Prusa support asked me to try removing the buffer. WHich, probably also straightened the ptfe tubes quite a bit for me. I just got my first successful print in a while. 

I'll keep trying it like this, if it works i know it is caused by the buffer or the filament paths. Maybe yuo might come to the same conclusion.

Krosta8, 2024-06-10:
I would say that this is also possible. I need to check that also. I am using auto-rewind spool.

codingcatgirl, 2024-06-26:
I have not encountered this issue again after getting rid of the buffer.

closed 2024-09-02 by github-actions[bot]