prusa3d/Prusa-Firmware-Buddy #4022: [FEATURE REQUEST] make optional the possibility of disable send to printer and print

otaolafranc, 2024-05-31:
**Printer type** - [MK4]

**Is your feature request related to an existing problem? Please describe.**
  I am one of the persons in charge of the printer of our lab. but several people use them, I would be awsome to have the possibility to allow people to upload their files to the printer via wifi/ethernet, but disable the possibility to "send and print". as people are not careful and will send without checking if the printer has anything in the build plate. this would give a middle ground where you don't need to change the USB each 2 min, but would prevent any headache. this could be really functional for printers that are common for several people such as a fablab for example.

**Describe the expected functionality**
  Having a toggle in the printer setting of prusa connect to 'allow remote printing' if disable when sending a file via wifi/internet, it will give an error if you choose 'send and print' in the prusa slicer, but will allow to send to the USB (without printing), similar to the behavior we have right now if we changed the filament for the printer (without accepting all the steps) and we try to send a file to it.

Defozo, 2024-05-31:
FYI Prusa Connect has a "Set Ready" feature. Perhaps this would help you.

CZDanol, 2024-06-11:
Hmm, I feel like this should be more of a PrusaSlicer thing, where you would disable this action in the printer profile settings. Would that make sense to you?

otaolafranc, 2024-06-11:
IMO, it should be in the printer itself, the situation is several people using the printer with different profiles, to same printer, that send the gcode, so the idea is that the printer does not allow to 'send and print' whatever the computer/user is, but allow to send the file, so one then goes to the printer and prints from there. it would be reasonable to set this in the printer itself.

CZDanol, 2024-06-11:
Okay, I've created an internal ticket BFW-5696 for this, we'll discuss the proposal in the team.