prusa3d/Prusa-Firmware-Buddy #4023: [BUG] M601 Double Pause

GetTheBalanceRight, 2024-06-01:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI, MK4, XL]
MK3.5

**Printer firmware version** - [e.g. 4.0.5, ...]
6.0.0

**Original or Custom firmware** - [Original/Custom]
Original

**Optional upgrades** - [e.g. Filament Runout Sensor]
E3D Revo, MMU3

**USB drive or USB/Octoprint**
USB / PrusaConnect

**Describe the bug**
M601 Pauses twice.

**How to reproduce**
When inserting a pause (M601) using PrusaSlicer, the printer pauses, and upon resuming using the touch screen or the knob, looks like it prints a little, then goes back to the parked position as if there was another M601 given.

**Expected behavior**
When I press "Resume," the printer should resume and not stop again

**G-code**
Attached
[Small Knob_0.6n_0.32mm_PLA_MK3.5MMU3_2h1m.bgcode.zip](https://github.com/user-attachments/files/15520024/Small.Knob_0.6n_0.32mm_PLA_MK3.5MMU3_2h1m.bgcode.zip)

CZDanol, 2024-06-11:
Internal ticket: BFW-5694

loudaslife, 2024-08-22:
I'm also experiencing this exact bug, consistantly reproducable on my MK4 with firmware 6.1.2.

I haven't watched it super closely to be certain, but it looks like what's happening is that the first pause happens correctly, but then after the user presses "continue" it re-prints the last few seconds of gcode, which includes the same M601, causing it to pause again. Maybe there's some kind of gcode buffer not getting properly cleared out?

The duplicated print moves have come close to ruining some of my prints, since the next layer doesn't want to print smoothly over the extra material, but so far I've gotten lucky and there were enough layers above to even it out.

Attached is the .bgcode for one of the many prints I've seen this issue on lately, as well as the PrusaSlicer file that generated it.
[double pause bug example.zip](https://github.com/user-attachments/files/16719200/double.pause.bug.example.zip)

michalrudolf, 2024-09-30:
Hi, I have trouble reproducing the bug. I have tried reflashing to 6.0.0 firmware on MK3.5 and executing attached Small Knob gcode, printer correctly paused and resumed without any second pause.
I have also reflashed to firmware 6.1.2 and tried double_pause_bug_example gcode with the same result. Are you able to reproduce it consistently or is that a rare occurance?

Since there was a substential refactoring of USB media prefetch (6.2.0), it is possible, that this bug was fixed, so if you are still having this issue, try updating the firmware to 6.2.0-alpha.

loudaslife, 2024-10-01:
Annoyingly, I can't seem to replicate it anymore on 6.1.2. I'm certain it happened on four consecutive prints (with different gcodes) back when I commented, but I just tried a bunch of small test prints and can't get it to occur again. Maybe it only happens on longer gcode files, or maybe it was just a weird coincidence that it was so consistant back then. I'll keep an eye on it and report back if it happens again.

Prusa-Support, 2024-12-20:
Hi everyone. I seem to understand that so far it wasn't possible to reproduce the problem on the most recent firmware pre-release 6.2.0-alpha1 or 6.2.0-alpha2. The last pre-release proved to be rather stable over many testing weeks so please consider giving it a go.

Michele Moramarco
Prusa Research

closed 2025-03-31 by github-actions[bot]