prusa3d/Prusa-Firmware-Buddy #4027: [BUG] M600 with MMU3 behaves unexpectedly

firetech, 2024-06-02:
**Printer type** - MK4, with MMU3

**Printer firmware version** - 6.0.1+14848

**Original or Custom firmware** - Original

**USB drive or USB/Octoprint** - USB drive

**Describe the bug**
M600 is handled weirdly on MK4 with MMU3. Instead of unloading the filament to MMU3 and asking which slot to load instead (which MK3S+ etc. seem to do since [this PR](https://github.com/prusa3d/Prusa-Firmware/pull/4216)), it ejects the current filament out of the front of the MMU and asks the user to change the filament in the currently used MMU slot, as if the filament in the slot ran out.

I know you can add color change based on layer while slicing for MMU3, but some use cases for printing with M600 instead are:
1. You want a quick color change by layer, but haven't got any guess regarding correct purge volumes for the filaments in use, so you'd rather do it manually.
2. You're printing with a nozzle size other than 0.4mm (at the time of writing, MMU3 profiles for MK4/3.9 only exist for 0.4mm nozzles).
3. You're printing G-code supplied by someone else.

**How to reproduce**
Slice something using a normal (non-MMU3) MK4 profile, and add an M600 color change before exporting the G-code to the printer. Print on a MK4 with MMU3. Probably also affects MK3.9, possibly also MK3.5.

**Expected behavior**
For the M600 color change G-code, unload filament to MMU, ask user what slot to load instead.

**G-code**
[M600 test_0.4n_0.2mm_PETG_MK4IS_1m.zip](https://github.com/user-attachments/files/15527025/M600.test_0.4n_0.2mm_PETG_MK4IS_1m.zip)

ojrik, 2024-06-07:
My problem is the same. I also prefer to use the M600 method for single extruder color printing. I was used to my MK3S+ MMU2S, after upgrading to MK3.5 MMU3, I'm a little disappointed that they didn't add the option to select a slot for the MMU.

CertainLach, 2024-06-15:
Furthermore, If you try to use buttons on MMU to choose the filament manually (which it allows you to do), it will get stuck in "disengaging idler", because it is confused, and doesn't know what to do next, completely failing to continue the print in result.

firetech, 2024-08-15:
This is only a bug if you compare it to the 8-bit firmware for older models. Regardless, the current behavior is a bit weird.

rcmaniac25, 2024-11-25:
Not stale

rcmaniac25, 2025-01-27:
Not stale

rcmaniac25, 2025-03-31:
Still not stale

CertainLach, 2025-06-02:
[Stalebot should die](https://nostalebots.xyz)

rcmaniac25, 2025-08-04:
I would be surprised if this issue was any different on the CORE One and Firmware 6.3.4. So very much still relevant.

3d-gussner, 2025-08-06:
I assigned myself to this issue to prevent the stale bot to close it.

Can you please verify that this issue still is valid with the newest firmware of the printer and MMU?

rcmaniac25, 2025-09-01:
@3d-gussner Sorry for the delay, correct: it doesn't give an option to pick another slot.

ulmeck, 2025-12-03:
Same experience on MK3.5, and watching.