prusa3d/Prusa-Firmware-Buddy #4030: [BUG] Wrong Z after Spooljoin

firetech, 2024-06-05:
**Printer type** - MK4, with MMU3

**Printer firmware version** - 6.0.1+14848

**Original or Custom firmware** - Original

**USB drive or USB/Octoprint** -  USB drive

**Describe the bug**
I've had it happen twice now (also the amount of times I've used Spooljoin on this printer) that the printer doesn't seem to return to the correct Z height after doing the Spooljoin filament switch for the rest of that layer. The next layer after it looks fine, though. The first time, the rest of the layer just looked slightly off, the second time, the filament barely stuck on for the rest of the layer. It looks like bridging, but it was supposed to be the last layer of solid infill...

[image]

Note that the first layer of grid infill looks fine next to the problem area, so the next layer was unaffected. Also, the grid infill that can be seen in the problem area is _on top_ of what's supposed to just be solid infill. It may look like it's just bridging over the grid infill, but that isn't the case.

**How to reproduce**
Start a print with Spooljoin, make the filament run out mid-layer.

**Expected behavior**
The rest of the layer after a Spooljoin filament switch should look the same as before the filament was switched.

danopernis, 2024-06-05:
Thanks for the report @firetech it is always great when bugs are reproducible :slightly_smiling_face: I have created internal ticket BFW-5647

jmillerfo, 2024-08-13:
I have the same problem.  I was hoping this would be fixed in version 6.0.4 but it was not or at least not reported in the change log.

firetech, 2025-01-05:
I retested this yesterday with 6.1.4+7908 on a MK4S, and the printer returned to the correct Z after an induced spool join during the first layer (0.2 mm). Thinking this issue was fixed, I started a print with spool join configured, and when the filament ran out at layer height 1.2 mm, the printer did a spool join and promptly returned to printing ...at layer height 1.45 mm. 😞

Skimming the code related to spool join, I can't find anything obviously wrong (although, I'm admittedly not exactly familiar with this code base).  Some thoughts:
* Could it be other threads modifying the `current_position` and/or `destination` variables?
  * Side note: Is the value set to `destination` [here](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/8978fe1b166806a4ef6ecc5bff210cbbf183fc06/lib/Marlin/Marlin/src/module/prusa/spool_join.cpp#L259) ever used with MMU spool join? `MMU2::mmu2.tool_change_full()` stores its own return position [here](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/8978fe1b166806a4ef6ecc5bff210cbbf183fc06/lib/Marlin/Marlin/src/feature/prusa/MMU2/mmu2_mk4.cpp#L665).
* Could this be affected at all by having the "Z height" footer enabled? (I'll probably test this myself later when my printer isn't busy printing.)

CZDanol, 2025-01-06:
Hello, this should be fixed now in 6.2.0-alpha2. Can you please retest on that version?

firetech, 2025-01-06:
Ah, I didn't see any mention of that in the changelog. I'll test it out!

CZDanol, 2025-01-10:
Great :) Closing the issue, will reopen if we find out there is still something to fix

FlowerDIY, 2025-01-27:
Hi Danol,

I have encountered this issue on 6.2.0 alpha 2. I have ruled out calibration issues because multi color prints come out flawlessly. I only see the wrong Z offset when spool join occurs at filament runout. 

Would you suggest re-loading the firmware to see if that fixes it?

FlowerDIY, 2025-01-28:
I have been getting this issue in 6.2.0 alpha 2.
Multi color prints come out fine, so I can rule out calibration.
When I do a spool join, the z offset is apparently zero from the previous layer causing the tool to be directly in contact with the previous layer and ooze filament out of the side.
The z is correct after that, and the rest of the print comes out fine. I can reproduce this with every print by cutting the filament and letting it run out.

CATSNACKz, 2025-02-09:
I'm on 6.2.0 and this is still an issue. After spool join, it prints a little too high, but will fix itself once it moves to a new layer or moves the bed so it can move to a new spot on the current layer. This is only an issue with spool join. This does not happen in multi color/material tool changes. In my photos you can see it printed good until it did spool join and started printing too high. Then when it moved to another spot on the bed, it went back to printing good. 

[image]
[image]
[image]

CZDanol, 2025-02-10:
Reopening and assigning a new ticket: BFW-6752 (formerly BFW-5647)

firetech, 2025-04-04:
Should this issue be fixed by 6.2.3, or is the fix quoted below a separate issue? 🤔 

> Fixed returning to wrong Z position after pause
> When unpausing, the printer was not applying the MBL correction to the unpark move.

FlowerDIY, 2025-04-25:
[image]

Tested in 6.2.4 RC1, and it still returns to the Z height after spool join.

The new firmware addresses the wrong Z position after a pause, but this does not appear to fix the spool join issue. 

The important part is that normal multicolor prints turn out fine. If there is any other troubleshooting I can do to more thoroughly test this issue, please advise.

CZDanol, 2025-04-28:
Hello, testing wasn't able to reproduce this issue on our 6.4 branch, so we are thinking that it got fixed somewhere along the way. Our current plan is to wait for the 6.4 release, and potentially revisit this issue if we get new reports.

FlowerDIY, 2025-09-13:
> Please verify if Firmware 6.4.0-alpha
> solves the issue.

It looks like it may nearly be solved, but something else seems to be
broken with spool join sequence.
The main head crashes into the dock when docking, and the print results in
a large layer shift in X. The Z height appears to be correct though.

Tools 4 is main and 5 is the join head.
Start print (36mm cube)
At about halfway through the print, run out of filament on tool 4.
Tool 4 move to dock and waits for temperature on tool 5.
Tool 5 reaches temperature
Tool 4 attempts to dock and results in erratic motion, violently jumping
and crashing into the dock, but ultimately docks with no error or fault.
Empty carriage does homing routine
Carriage picks tool 5 successfully and resumes print
Print continues with shift in X axis by 2-3mm mm. But Z is the correct
height.

This occurred repeatably in 3 attempts.

3d-gussner, 2025-09-15:
@FlowerDIY Thanks for testing 6.4.0-alpha. Glad to hear that the Z height is now correct.
There have been multiple reports that the alpha causes some issues on XL with the tools. There is another ticket you can comment on.

CATSNACKz, 2025-12-09:
I'm on FW 6.4.0 and I don't think this is fixed yet. Once spool join happened, it looks like the z height wasn't right because the line it printed came loose and drifted over to another object. This caused a layer separation on both objects. It printed just fine after this, presumably after z-hopping once. 

[image]

[image]

[image]

crasbe, 2026-01-15:
The aforementioned behavior is also documented in #5040.

FlowerDIY, 2026-09-28:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

This is still a problem. Can someone at Prusa test to see if it is fixed?

Deadhuntz, 2026-10-20:
Same issue for me on Prusa XL 5T with FW 7.0.0 

Z is a bit too low for me, printing on 0,2mm layer height and after spooljoin it's like 0,1mm until new extrude spot.