prusa3d/Prusa-Firmware-Buddy #4033: [FEATURE REQUEST] Delay pause until infill/toolchange to prevent external blemishes

murk-sy, 2024-06-07:
### Printer model

All

### Describe the feature

When pausing a print, you can easily end up with blemishes in external perimeters due to reduced nozzle pressure between pause and resume. I would generally consider "pause" to be "not an emergency", so continuing to print for a little bit longer should not be an issue - especially since pressing pause does not immediately stop it either.

Ideally, this would be done when starting printing infill (or after printing a single line of infill), so the issue is hidden. Alternatively, though this should be chosen by the user, an option to pause during toolchange would be good (for example to clean up any mess that might cause further issues with no tool in the way).

Functionality similar to this issue:
- https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2564

Related issues:
- https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4019
- https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3878

### Expected functionality

When pressing Pause, a menu appears with the following options:

- Cancel
- Pause now
- Pause on infill
- Pause on tool/filament change (XL, MMU)

Alternatively, if the default functionality should still be "just pause", this menu could appear if the button/screen is pressed for >2 seconds.
(How would "hold to open menu" be communicated to user? Only in the manual or on screen too? Is there a universal "hold to open menu" that could be used across the printers?)

CZDanol, 2024-06-11:
Hey, I like this idea! I've created an internal ticket BFW-5693 for it, let's try to let it get through the process :)

tsgautier, 2024-06-17:
I came here looking for a similar feature - I think there should be an option to pause after a layer has been completed

murk-sy, 2024-06-17:
That's a good idea - if you've got an unrecoverable failure and plan to glue parts together it's a good way to ensure you won't have a random gap from a missing outer perimeter on one of the islands.

tsgautier, 2024-06-17:
Yes, that would be one good use, my main reasoning is actually occasionally I have an imperfection I want to fix - which is pretty hard to do while the printer is running.  For example maybe a stringy bit has laid over the top of the current layer or a blob has formed - removal is easy - if the printer is not running.  Pausing mid-layer and resuming will almost certainly introduce more imperfections.

Prusa-Support, 2024-07-26:
Sounds good and I agree that it should be up to the user whether to print longer or pause right away.
I seem to understand the only difference between #2564 and #4033 is that in the first case, a filament outage is detected, while in the second case, the user intentionally prompts a pause.

I assume both feature requests should be implemented together at once and may share some options, whereas in the first case the user may want to consume as much filament as possible without interruptions on the external perimeters, and in the second case mostly spare external perimeters. Helping the printer distinguish between external perimeters and the rest may be a good start.  

Michele Moramarco
Prusa Research