prusa3d/Prusa-Firmware-Buddy #4034: [BUG] Layer Shift at 4 of 4 MK4 with MMU Color prints since update to 6.0.2

tlachmann, 2024-06-07:
### Printer model

MK4 with MMU3

### Firmware version

6.0.2

### Printing from...

Octoprintt

### Describe the bug

While printing a colored model with wipe tower the extruder dives too deep into the wipe tower or the model and looses its correct X/Y position. For Z, after another lift, it wseems it resumes at the correct layer height.

the nozzle impact can be seen in the pictures below (marekd by red lines).

while trying to document another isse (where a manual filament change happens without filamentchange) i had the chance to caught an impact and resulting layer change onto video:
https://we.tl/t-FvPeY64b72 (availavble till June, 24th 2024)

[image]
[image]
[image]

### How to reproduce

For me, I installed FW 6.0.2 and print any color print with wipe tower. with FW 6.0.1 I did only observed that behavior at 1 out of 10 , now 100%

### Expected behavior

After the head lifts and travels to wipetower and back, the extruder gets a wrong height position, this should not happen

danopernis, 2024-06-10:
Hi @tlachmann thanks for the report. Looking at the changelog between 6.0.1 and 6.0.2 I can't identify anything suspicious. There are fixes related to phase stepping and enclosure which are only relevant for XL, some translation fixes which are almost always harmless, fix of the memory leak and fix of the stealth mode. Can you please double check that the issue is not hardware-related, by downgrading the firmware?

codingcatgirl, 2024-06-30:
At this point, successful prints with the MMU3 on my MK4 are a rare occurence and something to celebrate.

Without the MMU3 my MK4 works fine and without issues. With the MMU3? The MMU3 works fine but the firmware bugs out in one fun way after another, be it the famous USB disconnects or – unfortunately – this issue.

Ailin-Lyons, 2024-07-08:
I am also getting overextrusion in my purge block causing every single MMU3 print to fail. However I am not sure if this is due to one prusa firmware of some other misconfiguration.

codingcatgirl, 2024-08-19:
Sad to say that issue still persists.

Ailin-Lyons, 2024-08-19:
Hey @codingcatgirl,

I was able to workaround this problem by changing away from default MMU3 settings. The culprit setting here is Printers -> Single extruder MM setup -> Extra loading distance

The default value causes a blob of plastic right after color change. You can put a larger number (e.g. -60 instead of the default -52). This will avoid creating the blob, but it will also require you to purge more total volume so be sure to increase your purging volume or purging volume multipliers accordingly if you experience color bleed.

codingcatgirl, 2024-08-20:
It was a nice idea, but it did not help.

codingcatgirl, 2024-08-21:
For reference, the solution to my problem was #13235.

WillGIam7, 2024-11-17:
Same here. Already reduced purge extra flow, line distance in purge tower and disabled purging in infill. reducing extra flow and disabling purging in infill seems to make the purge tower more stable and I was able to print my model, but next was print wasnt succsessful again...