prusa3d/Prusa-Firmware-Buddy #4035: [BUG] unable to load via Prusa connect Italian firmware 6.0.2 on mini

SNDR0, 2024-06-07:
### Printer model

Mini 

### Firmware version

6.0.2

### Upgrades and modifications

Original

### Printing from...

Send firmware file downloaded from GitHub and from prusa

### Describe the bug

Only Italian firmware is refused by Prusa connect. Other languages works perfectly.

### How to reproduce

Send the Italian version for mini 6.0.2 via Prusa connect and an error message appears

### Expected behavior

Little bug while compiling..

CZDanol, 2024-06-11:
Hey, what is the error message?

CZDanol, 2024-06-11:
Internal ticket: BFW-5697

SNDR0, 2024-06-11:
@CZDanol 
> Hey, what is the error message?

[image]
and if i scroll through the list of available firmware italian is missing
[image]

ClaGre, 2024-06-16:
That's right but I've upgraded the firmware in the usually way; download and place the firmware file into usb key.
Beware because before I've try to upload the file by Prusa Connect but without success because Prusa Connect say me that the file isn't correct.