prusa3d/Prusa-Firmware-Buddy #4036: [BUG] [MK3.5] Print Fan error after a couple of layers but all tests passed

Makeneer, 2024-06-09:
### Printer model

MK3.5

### Firmware version

6.0.1

### Printing from...

Octoprint on RasPi4 via USB cable 

### Describe the bug

I have upgraded my Prusa i3 MK3S+ with MMU2S to Prusa i3 MK3.5 with MMU3.
Before the upgrade I had no problems with my printer and I have printed all necessary parts for the upgrade with no problem at all.

After the upgrade everything looked fine, the self-test passed with everything OK and printing was no problem until a couple of layers where printed.
After a couple of layers my printer gave an error on the print fan. I have checked the wiring and connectors, checked the working of the fan with the available menu options and did the fan test again.
Everything seems to be OK.

But my printer is still giving the same print fan error after a couple of layers.
I have changed the slicer profile so the print fan is running from the first layer and made a new test print.
The fan is running OK from the start of the print but after a couple of layers the printer is giving me the same error again.

I have checked the working of the print fan over the complete X-axis range a couple of times and it is working OK every time and every test.

I have disabled the Fan Check option for now. I have made several prints and checked the fan for working from time to time. It was spinning all the time.

ghouscht, 2024-06-11:
I have exactly the same issue and thus disabled the fan check. I did at least a dozen prints since upgrading to 3.5 and checked the fan from time to time it was always working. All the tests pass as well, I ran them at least 5 times.

Makeneer, 2024-06-12:
To figure it out, I have hooked up my oscilloscope to the print fan connected to:
1st: the Rambo board of the MK3S+
2nd: the xBuddy board of the MK3.5

The print fan used for the MK3S+ and also for the MK3.5 is a 3-pin 5VDC fan. This fan is supposed to be powered with a continues voltage. The third pin gifs a RPM signal only when the fan has power.
Instead of a continues voltage, the fan is powered by a pulsating voltage which is switched on and of very quick (PWM signal). By changing the time the power is on and off (duty cycle), the speed of the fan is regulated. The problem is, that the RPM signal cannot be measured when the power is off.

On the Rambo board (Firmware 3.13.3) the PWM frequency on the fan is approx. 60Hz. The speed is regulated by changing the duty cycle of the PWM signal. Every 5.4 seconds the PWM signal is set to a duty cycle of 100% for approx. 0.2s, I guess to check the rotation of the fan. This 0.2s is long enough to have a readable RPM signal. It is only an indication that the fan is still rotating. The fan speeds up as soon as the duty cycle is set to 100%, so it is not giving the speed of the fan between these speed bursts.

On the xBuddy board (Firmware 6.0.2) the PWM frequency on the fan is approx. 20Hz. So the duration of the pulses is 3 times longer than the at the Rambo board. The speed is regulated by changing the duty cycle of the PWM signal.
The longer pulses (on-time of the PWM signal )could be long enough to make a RPM reading but:
•	On my original Prusa print fan, the speed of the fan is at lower speeds (<30%) to slow to produce a reliable RPM signal in the on-time of the PWM signal.
•	I have tried an other fan with a little higher speed and that one gave at lower speeds (<30%) enough pulses during in the on-time of the PWM signal to be reliable readable. With this fan hooked up I made a couple of test prints without error.

I understand the choice to change the way the RPM signal is read. On the xBuddy it is possible to get the more or less real RPM of the fan, but it does not work for all original fans.

I use my printer for 95% with PETG, so there is a low fan speed. During the speed ramps up after the first layer(s) the original fan is to slow to measure the RPM which results in the Error.

I hope I have give some insight in the cause of the difference between the print fan on the MK3S+ and the MK3.5. I had hoped Prusa had given this answer instead of letting me to figure it out.

CZDanol, 2024-06-27:
Internal ticket: BFW-5761

ghouscht, 2024-06-27:
> I have exactly the same issue and thus disabled the fan check. I did at least a dozen prints since upgrading to 3.5 and checked the fan from time to time it was always working. All the tests pass as well, I ran them at least 5 times.
> 
> Anyway I bought a replacement fan and I plan to install it, re-enable the fan check and then see if the issue is still present. I'll post an update here once I did that.

The issue is still the same even with the new fan. But now after reading the very detailed input from @Makeneer that is no longer a surprise. Thank you for the detailed analysis 🙏🏻

Makeneer, 2024-06-27:
I think this issue is easy to fix by just ignoring the printfan status below 30% speed, just like the MK3S+ on the Rambo board did.

Prusa-Support, 2024-07-06:
Thanks for reporting.

With the previous printer models MK3/MINI the print fan was never supposed to be set any lower than 20% in Prusa Slicer cooling settings - or 30% if ramping so it wouldn't spin below 15% for transitioning layers. This was also highlighted in issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2314.
The same currently applies for MK3.5 which actually still works with the same print fan used on MK3, and this apparently matches well what @Makeneer says.

I seem to understand that the newer models MK4/XL at times may operate with print fan set to 10%, however, if I'm not missing anything, our stock filament profiles for MK3.5 still use the print fan set to a minimum of 20%, or 15% as transitioning threshold.

I wouldn't expect problems with the cooling thresholds set by our stock profiles, except for degrading or dirty print fans, but I recognize that RPM detection is not extremely reliable at low speeds, so please let us know.
Our developers are currently looking into this.

In the meantime, please ensure a minimum of 20% print fan speed (or 15% while ramping up), and that the print fan is in optimal conditions. So please double-check the fan and, in case of further reports, consider including the cooling settings you had problems with (min speed, no-fan layers, full-fan layer).

Michele Moramarco
Prusa Research

Tomcus, 2024-07-08:
Hi, we have prepared a fix for next minor release (not patch). The print fan error shouldn't trigger on MK3.5 if the fan speed is bellow 30%.

danopernis, 2024-08-26:
@Makeneer Fixed in v6.1.2 via https://github.com/prusa3d/Prusa-Firmware-Buddy/commit/fe2900eb19c4cad9576645d801ead85a2fb9fdac can you re-test please?

ghouscht, 2024-09-03:
Hey @danopernis,
I updated my MK3.5 to v6.1.2 and did some test prints. I can confirm that this release seems to fix the print fan errors. The self test passes and during printing (after turning the fan check back on) I no longer get the fan errors.

ghouscht, 2024-09-10:
> Hey @danopernis, ~I updated my MK3.5 to v6.1.2 and did some test prints. I can confirm that this release seems to fix the print fan errors. The self test passes and during printing (after turning the fan check back on) I no longer get the fan errors.~
> 
> **EDIT 09.09.2024:** Just got the same fan error as seen previously with FW v6.1.2

 figured out why I first thought the fan error issue is solved. Some context: I mostly print PETG where the print fan is not running on the first few layers (using the Generic PETG profile in Prusa Slicer) and then starts to run at ~15% on the 4th layer or so. However since installing FW v6.1.2 I printed in PLA which in terms of fan speed is entirely different and I never saw a fan error. With the Generic PLA profile fan speeds are much higher (>80%) so the issue described here isn't relevant. Now that I started printing in PETG again the same fan error happened again. Sorry for the confusion.

danopernis, 2024-09-17:
That's OK @ghouscht mistakes happen. Now that we have a confirmation, I am closing this issue even though @Makeneer did not respond.

closed 2024-09-17 by danopernis