prusa3d/Prusa-Firmware-Buddy #4038: Filament skip/clog detection

Avamander, 2024-06-09:
### Printer model

MK4 Stock

### Firmware version

6.0.0

### Printing from...

USB drive

### Describe the bug

I've had multiple prints fail because of an undetected clog. I would really like support for having a sensor for detecting filament not moving.

Although avoiding both would be ideal in the first place, but I'm asking exactly because of these rare cases.

I've weirdly had two different (large) prints both fail at the same layer height twice. I'll likely create a thread about this on the forums, maybe PrusaSlicer generates some odd clog-gcode. (The pain of resuming at a failure layer height is an issue on its own.)

### How to reproduce

I guess one could tie the filament to something fixed and have the extruder struggle extruding.

### Expected behavior

Filament loss of grip and/or clogs are detected.

danopernis, 2024-06-10:
There is already a filament clog detection implemented since v6.0.0

It only works for detecting internal pressure in the melting zone. As for extruder skipping steps and external pressure, there is an issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3710 so this seems to be a duplicate

closed 2024-06-10 by danopernis

Avamander, 2024-06-10:
@danopernis The extruder is not skipping per se, so that issue is not a duplicate. Pressure detection does not seem to work in my case.

danopernis, 2024-06-10:
I am sorry if that is too obvious, but do you actually have the "stuck filament detection" enabled? Other than that, I am not sure what else can be done here. Maybe after the telemetry is finished and folks opt into it we might gather more data to improve the detection algorithm, but until then, I don't see anything we can do with your feature request.

m10d, 2025-01-27:
> There is already a filament clog detection implemented since v6.0.0
>
> It only works for detecting internal pressure in the melting zone. As for extruder skipping steps and external pressure, there is [...]

@danopernis - is that issue, or essentially identical discussion in this open issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4086#issuecomment-2614788453, actually about e-steps at all?

a skip seems likely but NOT guaranteed in the case of grinding the filament (eg a full tangle / knot).  Whereas the load cell would detect ~zero force compared to the normal force it experiences while extruding.  

as a sw/fw engr myself, I'm confused by what I've read on prusa's support pages here (my comment is linked ^) - I am assuming that the normal force imparted by the nozzle/meltzone against the e-motor drive is very noticeable, meaning "zero" force exerted during the tangle case would be a more clear signal than the varying "force too high" of a partial clog or filament variation?

Avamander, 2025-01-27:
@danopernis 

> I am sorry if that is too obvious, but do you actually have the "stuck filament detection" enabled?

I do.

> Edit: can you perhaps clarify if this is a bug report or feature request?

I'm now not sure if "stuck filament detection" is supposed to detect stuck filament or a clogged nozzle, because it has repeatedly not detected stuck filament for me.

I've avoided it for a while by changing the filament, cleaning the filament path with a weak solvent (in case there's some grease or something similar that's introducing a loss of friction). But I'd still like the machine itself to detect this. Even if it requires some extra hardware.