prusa3d/Prusa-Firmware-Buddy #4040: [BUG] After successfull print, Auto Home does not work properly

Racemade, 2024-06-11:
### Printer model

XL

### Firmware version

6.0.2 Official Release

### Printing from...

Prusa Connect

### Describe the bug

After a successful print, the printer does not work properly.
If you go back to the start menu und press Auto Home, the motors are louder then normal and (thats the most important thing) the printer does a very, very slow auto home.

### How to reproduce

Print something.
Go back to menu.
Do an Auto Home

### Expected behavior

If i click Auto Home the printer should do it as always - silent and fast.

Racemade, 2024-06-27:
Still exists in 6.0.3 and after a full factory reset.

danopernis, 2024-06-27:
I managed to reproduce this. Doesn't seem like a big issue w.r.t. homing, because the distance the nozzle needs to travel to get from the corner to the probing point isn't that large. However, it may indicate some other underlying issue so it is worth further investigation. I am opening internal ticket for that.

Racemade, 2024-07-03:
https://github.com/prusa3d/Prusa-Firmware-Buddy/assets/168211087/40d97ba6-50cd-40d1-a473-33eb3f0cc34b

Thats slow...auto home After Print. There is a bug or something else.

Racemade, 2024-07-19:
Any news here? Or anybody else with this kind of problem?

clefranc, 2024-07-30:
Same here with 6.0.4+14924

Racemade, 2024-09-08:
Still exists in 6.1.3

danopernis, 2024-09-17:
This should be fixed in 6.2.0-alpha1 according to the internal ticket. It somehow didn't make it to the release notes. Can you test it? Or wait a bit for stable 6.2.0 if you don't want to use alpha firmware (which has some regression w.r.t. to Wi-Fi)

Racemade, 2024-10-05:
I wait for the official release- thank you for comment!

Racemade, 2024-11-13:
So for now 6.2.0 alpha 2 resolved the issue. Waiting now for the official release and we will see.
Thanks!

closed 2024-11-13 by Racemade