prusa3d/Prusa-Firmware-Buddy #4041: [BUG] Zits on outer perimeter of small objects with Input Shaper

Bastel-Bodo, 2024-06-11:
### Printer model

MINI+

### Firmware version

6.0.2

### Upgrades and modifications

Z-axis stabilization brace; overhead light

### Printing from...

PrusaLink

### Describe the bug

I am trying to print a very small and lightweight object essentially consisting only of external perimeters, with each layer being slowed down to the minimum print speed due to the small size of the object. Seams are aligned so that no retractions need to happen (as far as possible), not even on layer changes.

When I slice the object using an InputShaper profile in PrusaSlicer 2.7.4 and run the print, then small zits appear infrequently on the outer perimeter, briefly after each jump to the next layer. The result looks like this (Z seam is in the corner of the fin on the left):

[image]

Observing the print, I noticed that the print head very briefly stutters in its movement along the outer perimeter where this happens. The interrruptions were non-deterministic - running the same G-code twice led to a different "pattern" of zits on the object. The stutter only appeared within a few millimeters after the layer change.

When I sliced the object using an old non-InputShaper profile and ran that on my printer, the object came out perfect, with no zits at all.

I'm not entirely sure whether this is an issue with the printer itself or a slicer issue. If it turns out to be the latter, I can post an issue to PrusaSlicer's bug tracker as well

### How to reproduce

The behavior can be observed running the corresponding G-codes on a Prusa MINI+ using Prusament PLA.

[Zits-on-outer-perimeter.zip](https://github.com/user-attachments/files/15794153/Zits-on-outer-perimeter.zip)

### Expected behavior

The outer perimeters should be clean, and no zits should appear.

MichaelJLew, 2024-06-22:
In my explorations it seems that the slicer will put in too many small G1 movements for the printer to keep up with leading to the stutters and the zits. Of course, the other way to look at the issue is that the printer hardware or firmware is unable to deal with enough small movements to keep up with the slicer...

Bastel-Bodo, 2024-06-22:
> In my explorations it seems that the slicer will put in too many small G1 movements for the printer to keep up with leading to the stutters and the zits. Of course, the other way to look at the [...]

@MichaelJLew Thanks for your assessment! In that case, it should be possible to resolve the issue by simply increasing the G-Code resolution in millimeters for MINI-targeted print setting in PrusaSlicer, right?

MichaelJLew, 2024-06-22:
That sounds like it is worth a try. However, my Prusa MK4 suffers from  stutters with gcode that runs without any problem on my Duet-equipped Voron running RepRap firmware and so it is a deficiency of the hardware and or firmware, in my opinion. That means that increasing the smallest segment is the slicer is a workaround rather than a fix.

My own initial interest in the problem was with G2 and G3 arc commands in my custom gcode that would sometimes lead to too many segments from the firmware. I'm not sure if that is still a problem as I long ago decided to do away with those arc commands. (I will note that the Prusa MK4 was initially unable to use the arc commands at all until about six months after I received mine.)

Bastel-Bodo, 2024-06-22:
Just gave it a try and compared the G-Code results with different settings for the G-Code resolution. However, there's no significant difference as far as I can see. The positions in subsequent G1 commands are about 0.5 mm apart, regardless of the setting - plus, due to the small object size, the print is slowed down to a crawl anyway, so the printer should have all the time it needs to process those commands.

Also, when I slice without Input Shaping, I'm using the same resolution setting, and the results are perfect.

So, it's quite likely that there is indeed a firmware issue in play here. For now, I can work around this by just using my old, non-InputShaper profiles. But it would still be great to have this resolved.

As for arc fitting, I also disabled this a while ago after a few attempts at using it which resulted in (very reproducible) zits that went away when I disabled it.
I did try enabling this to fix my issue here, of course, but that also didn't change anything about the results.

MichaelJLew, 2024-06-22:
Oh well. It was worth a try. Definitely a firmware issue then.

I'm pleased to hear that I am not the only one to find problems with arc commands. I think that my bug posts have probably timed-out by now and so have gone into the ether. There was never any acknowledgement of the problem if I recall correctly. A shame.

closed 2024-08-29 by github-actions[bot]