prusa3d/Prusa-Firmware-Buddy #4042: [ENHANCEMENT] Don't remove trailing zeroes of the Footer's axis value for static footer appearance

puhitaku, 2024-06-13:
### Printer model

All

### Describe the enhancement

Hi, thanks for the cool firmware!

The current implementation of footer items of [axis position](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/3e63d0e8ec1dcefeac7e128c198538586d83f01a/src/gui/footer/footer_item_axis.hpp#L36-L38), [gcode position](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/3e63d0e8ec1dcefeac7e128c198538586d83f01a/src/gui/footer/footer_item_axis.hpp#L72-L74), and [live z](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/3e63d0e8ec1dcefeac7e128c198538586d83f01a/src/gui/footer/footer_item_live_z.cpp#L29-L32) removes trailing zero. This makes sense, for aesthetic reasons, but in reality this makes the footer items wobble around when the value is like `2.50` that will actually be `2.5` on the screen.

I personally prefer NOT removing zero since not removing it will result in constant displayed width of footer items and constant positioning. I'm ready to contribute when this idea is accepted. Thank you!

danopernis, 2024-06-14:
Hi @puhitaku thanks for the generous offer. I talked to the product department and they like your proposal.

Before implementing it, please wait for the next minor release. The footer was recently refactored in our private fork so integrating your changes if you start developing now would involve solving a lot of git conflicts.

puhitaku, 2024-06-14:
Thanks @danopernis. Then, I'll begin watching this repository for the upcoming release. Please ping me if I seem to forget here, just in case.

puhitaku, 2024-06-25:
@danopernis Apologies for taking your time, but if I can know the estimated release date of the next minor release, I'd be happy

danopernis, 2024-06-25:
@puhitaku I am expecting several weeks, but don't quote me on that :slightly_smiling_face: it wouldn't be the first delay I've seen in my software engineering career.

Perhaps the merge conflict wouldn't be as problematic as I was originally imagining it, so if you want to, go ahead :+1:

puhitaku, 2024-08-08:
I've filed a PR for this issue (and I've invalidated the warranty of my MK4 to see if this change works haha 😁)
Looking forward to seeing the upcoming review and merger!

danopernis, 2024-08-14:
Thank you for the contribution. I merged your fix. Considering this is not a bugfix, this will not make it into 6.1.x, but it will be eventually released in some future release. I am keeping the issue open to track progress.

bkerler, 2025-04-13:
Can this issue be closed, as PR has been merged ?

CZDanol, 2025-04-14:
I suppose so.

closed 2025-04-14 by CZDanol