prusa3d/Prusa-Firmware-Buddy #4043: [BUG] MMU3 footer bug - Filament type missing

seifer69er, 2024-06-14:
### Printer model

Mk4 with MMu3

### Firmware version

6.0.2

### Upgrades and modifications

MMU3

### Printing from...

USB Drive

### Describe the bug

Mk4 display footer does not display the filament type when MMu3 is used
Only displays "- - -" 

### How to reproduce

default settings, will always display when printing

### Expected behavior

display the Filament type with MMu3 installed

firetech, 2024-06-14:
I noticed yesterday that the Filament footer only ever seems to list what's in the first slot (if specified using the Advanced Preload), regardless what slot is currently in use. Quite useless...

unclej84, 2024-06-18:
Maybe related to #3936 ?

seifer69er, 2024-06-19:
> Maybe related to #3936 ?

I don't believe it's a directly related, this issue is missing filament types ,where as 3936 is incorrect types

closed 2024-10-28 by github-actions[bot]