prusa3d/Prusa-Firmware-Buddy #4044: [BFW-5710] [FEATURE REQUEST] Expose Stealth Mode controls via API

aderusha, 2024-06-16:
### Printer model

All network-capable printers

### Describe the feature

Stealth mode allows a modern Prusa printer to run nearly silent at the expense of a slight increase in overall print time.  Surfacing external control over stealth mode could enable a wide range of use cases by silencing the printer in response to events in the outside world.

For example, if Home Assistant is deployed one could enable any of the following use cases:

* Turn on Stealth Mode after 11:00pm so people can sleep
* Turn on Stealth Mode when the lights in the room turn on or a motion sensor detects presence (be quiet when I am in the room)
* Turn on Stealth Mode when my calendar shows that I am in a meeting
* Turn on Stealth Mode when my computer microphone is turned on
* Turn on Stealth Mode when my cell phone indicates that I am on a call
* Turn on Stealth Mode when my TV or stereo are turned on

Tagging @CZDanol per request.

CZDanol, 2024-06-16:
I see you've already added the  BFW-5710 tag as per our standard, great! :D

jamkas, 2025-03-26:
I'm afraid I'm not good enough at software development to tackle this feature request myself but I would really love to have an API for stealth mode as well!

3gfisch, 2025-04-21:
Would like this too.

GarethBlain, 2025-06-01:
This would be VERY useful for when I tell my house I'm going to sleep. It turns all my lights into sleep mode etc but right now I have to get up and go and turn the printers into stealth mode manually. Would LOVE to have my house do this via Home Assistant!...

mattlennon3, 2026-06-07:
Really nice to see an issue + MR open for this! Just built my core one, would like to use silent mode pretty much as default, so this would be helpful to access.

ludovitmasek, 2026-09-17:
The APi would be a great option, as well as possible option to trigger this functionality threw hacker board pin, or setup schedule on the printer directly.