prusa3d/Prusa-Firmware-Buddy #4049: [BUG] "Refused" when trying to enable Prusa Connect / when restarting the printer

murk-sy, 2024-06-22:
### Printer model

XL, Mini (likely all)

### Firmware version

6.0.2 (and some previous versions)

### Printing from...

Prusa Connect

### Describe the bug

After restarting the printer, Prusa Connect fails to reconnect for an extended period of time, until suddenly connecting perhaps a day or so later - seemingly randomly.

### How to reproduce

It's a difficult issue so I'll give some background:

### Network setup
- mostly Ubiquiti equipment (everything that isn't a dumb switch)
- has 5 DNS servers set up
- primary DNS server is a local one that only has a handful of domains covered
- others are 2 cloudflare and 2 google
- all printers are connected via ethernet via the same switch
- printer IPs are assigned by DHCP (they are preset/static)

### Minis
There are 4 minis on the same network, one of which currently has the "refused" issue (happened on restart after upgrading to 6.0.2 from 6.0.1) - so far it has only happened on the XL.
None of them have wifi cards installed. Since Connect has become "stable" some versions ago (couldn't tell you a concrete version or date), I had no weird connection related issues.

On one where Connect is working, I turned it off, waited 15 second, restarted it, and it did the following:
- Acquire IP immediately
- Connect throws TLS error a few times
- Connect is online

Disabling Connect and re-enabling it causes it to be in "Connecting" for about 8 seconds, and then Online immediately after.

### XL
Currently runs into this issue every time after restart, which wasn't very helpful due to a few crashes happening in recent version.
Even if Connect was working perfectly fine

I honestly couldn't tell you at what version this has started but I'm fairly confident it hasn't always been the case.

Restarting the printer and capturing traffic based on the MAC address only shows DHCP and ARP requests.
[image]
(67 being the printer, 123 being the router)

### Files

I don't have anything concrete at this point, but I'm willing to do network captures/crash dumps/various tests on dev requests.

murk-sy, 2024-06-23:
Add-on info:

XL (currently online on Connect and normally accessible)
[image]

Mini (currently offline on Connect, not accessible via Link or ping)
[image]

danopernis, 2024-06-24:
Hi @murk-sy thanks for the report. Since you seem quite technologically capable, can you please try to obtain printer logs? Those are printed to serial line on baudrate 57600.

murk-sy, 2024-06-24:
Didn't know there's different things on different baudrates - that's neat. Had to use vscode serial monitor for it, since pronterface didn't like it.

I've actually managed to resolve it, at least for XL - within this log i've recreated a blank wifi config and loaded it. Even restarting the printer, it reconnects perfectly fine immediately.

When it was broken, the situation was as follows:
- Eth was chosen as default interface 
- Wlan had credentials entered, but did not have an ip (at least not displayed, I don't recall seeing it in Unifi)
- At some point before the XL was connected via Wlan (possibly on same ip, I don't recall), but ended up using Eth due to connectivity issues at the time.

So my gut feeling for the issue on the XL:
- Eth was selected as default
- Wlan had correct credentials and was (partially) connected
- Connect attempted to use Wlan despite not being selected as default

Removed some fluff and anonymised SSID.
```
....
641.525s [INFO  - GUI:4] Distributing event clicked
641.526s [INFO  - GUI:4] Distributing event child click
641.627s [ERROR - Network:12] mdns_resp_remove_netif: Not an active netif
641.628s [INFO  - ESPIF:12] Link went down
641.755s [WARN  - ESPIF:12] Recovered from UART error
642.279s [INFO  - ESPIF:12] MAC: 0c:b8:15:22:a1:4c
642.280s [INFO  - ESPIF:12] Joining AP ###ACCESS-POINT-NAME###:*(11)
642.484s [INFO  - ESPIF:12] MAC: 0c:b8:15:22:a1:4c
643.282s [INFO  - ESPIF:12] Link went up
644.562s [INFO  - GUI:4] Distributing event button down
644.673s [INFO  - GUI:4] Distributing event button up
644.674s [INFO  - GUI:4] Distributing event clicked
644.675s [INFO  - GUI:4] Distributing event child click
644.743s [ERROR - Network:12] mdns_resp_remove_netif: Not an active netif
644.777s [ERROR - Network:12] mdns_resp_remove_netif: Not an active netif
644.778s [INFO  - ESPIF:12] Link went down
644.905s [WARN  - ESPIF:12] Recovered from UART error
645.429s [INFO  - ESPIF:12] MAC: 0c:b8:15:22:a1:4c
645.430s [INFO  - ESPIF:12] Joining AP ###ACCESS-POINT-NAME###:*(11)
645.562s [INFO  - ESPIF:12] MAC: 0c:b8:15:22:a1:4c
646.362s [INFO  - ESPIF:12] Link went up
656.128s [INFO  - connect:14] Creating of connection failed
661.961s [INFO  - GUI:4] Distributing event value/index changed
....
675.345s [INFO  - GUI:4] Distributing event clicked
675.346s [INFO  - Marlin:8] enqueueing "M1587 I"
675.349s [INFO  - USBHost:16] MSC readahead stat speculative reads=117, hit=93, missed=106, blocking=106
675.349s [INFO  - GUI:4] Distributing event focus set
675.381s [INFO  - GUI:4] Distributing event capture set
677.347s [INFO  - Marlin:8] busy: processing
677.536s [INFO  - GUI:4] Distributing event value/index changed
....
679.640s [INFO  - GUI:4] Distributing event clicked
679.718s [INFO  - Network:8] ESP credentials generated to /usb/prusa_printer_settings.ini
680.824s [INFO  - GUI:4] Distributing event button down
....
682.355s [INFO  - GUI:4] Distributing event clicked
682.356s [INFO  - Marlin:8] enqueueing "M1587"
682.358s [INFO  - USBHost:16] MSC readahead stat speculative reads=152, hit=124, missed=139, blocking=139
682.399s [INFO  - GUI:4] Distributing event focus set
682.399s [INFO  - GUI:4] Distributing event capture set
684.357s [INFO  - Marlin:8] busy: processing
684.967s [ERROR - Network:12] mdns_resp_remove_netif: Not an active netif
684.967s [INFO  - ESPIF:12] Link went down
685.068s [ERROR - Network:12] netif is not up, old style port?
685.478s [WARN  - ESPIF:12] Recovered from UART error
685.618s [INFO  - ESPIF:12] MAC: 0c:b8:15:22:a1:4c
685.619s [INFO  - ESPIF:12] Joining AP :*(0)
685.841s [INFO  - ESPIF:12] MAC: 0c:b8:15:22:a1:4c
686.357s [INFO  - Marlin:8] busy: processing
688.357s [INFO  - Marlin:8] busy: processing
690.357s [INFO  - Marlin:8] busy: processing
692.357s [INFO  - Marlin:8] busy: processing
[... 53 lines cut ...]
13.573s [INFO  - Dwarf_3:10] 3.566s [INFO  - ModbusControl:0] Set Heatbreak requested temperature: 36
13.582s [INFO  - Dwarf_4:10] 3.573s [INFO  - ModbusControl:0] Set Heatbreak requested temperature: 36
13.592s [INFO  - Dwarf_5:10] 3.580s [INFO  - ModbusControl:0] Set Heatbreak requested temperature: 36
13.646s [INFO  - Dwarf_2:10] 3.557s [INFO  - ModbusControl:0] Set print fan PWM:: 0
13.652s [INFO  - Dwarf_3:10] 3.566s [INFO  - ModbusControl:0] Set print fan PWM:: 0
13.658s [INFO  - Dwarf_4:10] 3.573s [INFO  - ModbusControl:0] Set print fan PWM:: 0
13.664s [INFO  - Dwarf_5:10] 3.580s [INFO  - ModbusControl:0] Set print fan PWM:: 0
13.967s [INFO  - GUI:4] Distributing event child changed
14.084s [INFO  - ESPIF:12] MAC: 0c:b8:15:22:a1:4c
14.085s [INFO  - ESPIF:12] Joining AP :*(0)
14.209s [INFO  - ESPIF:12] MAC: 0c:b8:15:22:a1:4c
20.376s [INFO  - GUI:4] Distributing event value/index changed
20.377s [INFO  - GUI:4] Distributing event encoder plus
27.882s [INFO  - GUI:4] Distributing event capture set
27.883s [INFO  - GUI:4] Distributing event focus set
28.266s [INFO  - USBHost:16] MSC readahead stat speculative reads=117, hit=93, missed=106, blocking=106
28.353s [INFO  - connect:14] No transfer to recover
29.178s [INFO  - GUI:4] Distributing event value/index changed
29.179s [INFO  - GUI:4] Distributing event encoder plus
....
61.757s [INFO  - GUI:4] Distributing event focus lost
---- Closed the serial port COM5 ----
```

I will try recreating the issue. Though still doesn't explain why one of the Minis is having problems now. Might also be a broken config frankly but I'm running a print on it right now.

murk-sy, 2024-06-24:
Managed to reproduce, I believe reliably.

- Connect printer via eth (using assigned DHCP?)
- Create credentials file with valid wlan settings (using DHCP)
- Change wlan to use static ip
- Restart printer
- Connect will error/refuse

To undo:
- Change wlan to use DHCP (in menu it seems DHCP is already selected (with ip 0.0.0.0)? Change to static and back to DHCP)
- Connect will go online normally

Still not 100% sure what the issue is on one of the minis, but I've had "set up wlan" popups before, so the printer might *think* it has a wlan card installed, causing the same issue? I'll check once the print is done, likely tomorrow.

murk-sy, 2024-06-25:
### Mini (may be a slightly different issue)

I've sent a zip archive with the following files:
- 1-prusa_printer_settings-from-prusaconnect.ini
- 2-mini2-log-before-loading-clearstatic.txt
- 3-mini2_dump.bin
- 4-prusa_printer_settings_clear_static.ini
- 5-mini2-log-after-loading-clearstatic.txt
- 6-mini2_dump_after-loading-clearstatic.bin
- 7-prusa_printer_settings_set_static_to_0.ini
- 8-mini2-log-after-loading-set-to-0.txt
- 9-mini2_dump_after-set-static-to-0.bin

I've noticed that the mini with the issue (referred to as mini2) has had a static IP in its config for a while. Likely because I've had issues with DHCP.
Loading just the default settings from Connect didn't do much for that setting, so I tried importing clearstatic ini, which had blank entries. That didn't work, so I set the IPs to 0, which finally cleared it (should be visible in dump 9 too).

However, the printer simply isn't pulling an IP via DHCP. It's trying, but it's not being assigned it seems.
(61 is mini2, 123 is router/dhcp server)
[image]

I haven't tried a factory reset yet, to make sure it's reproducible, since I'm fairly sure it's a weird config issue. The board itself is from february 2022 (replaced after, funnily enough, some weird network issues), and it's *probably* had a factory reset since then, but I can't say for sure.

ucirello, 2024-12-07:
I can confirm that MK4S server seems to have similar issues. Static IP seems to increase the odds of things working, but overall it seems sensitive to the same issues when running off of DHCP. Behaves the same in Ethernet and WiFi from I can gather here.

dzuelke, 2025-01-08:
Is this somehow related to #4303 maybe?

tonysurma, 2025-02-18:
I have several MK4S/MK35/XL Printers and this happens on only one of them and that is the only one that previously had wifi working and I recently connected it via ethernet and switched that to primary and it says DNS refused on connect.

If I switch it back to wifi with ethernet removed it works immediately even without restarting.

Every other printer on same switch and settings that is ethernet only (never wifi) works great.  And every other printer on wifi works great as well.

So for me it appears to be having a working wifi and then switching to ethernet.  For now my solution is to just leave that one on wifi even though it is sitting right next to switch.

ucirello, 2025-02-18:
Another data point from Feb 2025 - I have a RPi running a ping in a GNU screen, pinging the printer's static IP, and ever since, it never has problems. I am not sure, but it seems ping keeps something alive.

kacperkor, 2025-08-12:
Hi, maybe this is irrelevant, maybe it's a simple fix. I had a similar problem but when registering my printer on prusa connect. I had error 12401, it sent me to [this page](https://help.prusa3d.com/article/connect-registration-failed-12401-mini_658941), which links [here](https://help.prusa3d.com/article/connect-registration-failed_654926). Nothing worked there but I resolved the issue myself. I use ethernet and my active interface in settings->network was set to Wifi. I switched it and now it works. Sorry it it's not relevant to this issue but maybe it should be added on [this page](https://help.prusa3d.com/article/connect-registration-failed_654926) as a troubleshooting step.